esp-rs/esp-idf-sys #4: Failure to build with the Espressif-built compiler

andrewshadura, 2021-08-01:
I’m trying to build rust-esp32-std-hello using the pre-compiled compiler I installed according to [this guide](https://github.com/espressif/rust-esp32-example/blob/main/docs/rust-on-xtensa-installation-x86_64-unknown-linux-gnu.md). (I have previously built the compiler myself, but I wanted to cut the installation time by a couple of hours this time.)

I’m using rustup to manage toolchains. Having set the local override to `esp`, I’m getting this output:
```
$ rustc --version
rustc 1.53.0-dev
$ rustc --print target-list | grep xtensa
xtensa-esp32-espidf
xtensa-esp32-none-elf
xtensa-esp32s2-espidf
xtensa-esp32s2-none-elf
xtensa-esp8266-none-elf
xtensa-none-elf
```
However, during the build, I’m getting this error:
```
$ cargo build
   Compiling esp-idf-sys v0.16.3
error: failed to run custom build command for `esp-idf-sys v0.16.3`

Caused by:
  process didn't exit successfully: `/home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/debug/build/esp-idf-sys-e657db78e305d488/build-script-build` (exit status: 101)
  --- stdout
  Found compatible PlatformIO Core 5.1.1 -> /home/andrewsh/.platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/master_missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/esp32c3_atomics_emul.diff
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/master_restore_link_fragments_for_pio.diff
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32cam; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
  PLATFORM: Espressif 32 (3.3.1) > AI Thinker ESP32-CAM
  HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
  DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
  PACKAGES: 
   - framework-espidf 3.40300.0 (4.3.0) 
   - tool-cmake 3.16.4 
   - tool-esptoolpy 1.30100.210531 (3.1.0) 
   - tool-ninja 1.7.1 
   - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
   - toolchain-xtensa32 2.80400.210211 (8.4.0)
  Reading CMake configuration...
  LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
  LDF Modes: Finder ~ chain, Compatibility ~ soft
  Found 0 compatible libraries
  Scanning dependencies...
  No dependencies
  Building in debug mode
  Retrieving maximum program size .pio/build/debug/firmware.elf
  Checking size .pio/build/debug/firmware.elf
  Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
  RAM:   [          ]   3.8% (used 12596 bytes from 327680 bytes)
  Flash: [==        ]  15.4% (used 161793 bytes from 1048576 bytes)
  ========================= [SUCCESS] Took 11.51 seconds =========================

  Environment    Status    Duration
  -------------  --------  ------------
  debug          SUCCESS   00:00:11.507
  ========================= 1 succeeded in 00:00:11.507 =========================
  cargo:CARGO_PIO_LINK_ARGS=--cargo-pio-link-linker=/home/andrewsh/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc --cargo-pio-link-remove-duplicate-libs -L/home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf -L/home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug -L/home/andrewsh/.platformio/packages/framework-espidf/components/esp_wifi/lib/esp32 -L/home/andrewsh/.platformio/packages/framework-espidf/components/esp_rom/esp32/ld -L/home/andrewsh/.platformio/packages/framework-espidf/components/xtensa/esp32 -L/home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/esp32 -L/home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/esp32/ld -L/home/andrewsh/.platformio/packages/framework-espidf/components/esp32/ld -Wl,--start-group /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/esp_pm/libesp_pm.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/asio/libasio.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/cbor/libcbor.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/unity/libunity.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/cmock/libcmock.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/coap/libcoap.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/console/libconsole.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/esp_adc_cal/libesp_adc_cal.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/esp_hid/libesp_hid.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/protobuf-c/libprotobuf-c.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/protocomm/libprotocomm.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/mdns/libmdns.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/esp_local_ctrl/libesp_local_ctrl.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/esp_websocket_client/libesp_websocket_client.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/expat/libexpat.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/wear_levelling/libwear_levelling.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/fatfs/libfatfs.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/freemodbus/libfreemodbus.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/jsmn/libjsmn.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/json/libjson.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/libsodium/liblibsodium.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-idf/mqtt/libmqtt.a /home/andrewsh/projects/rust-esp/rust-esp32-std-hello/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-c914de5a2aae2661/out/esp-idf/.pio/build/debug/esp-i [... cut ...]
  error: unknown target triple 'xtensa-esp32-espidf', please use -triple or -arch
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /home/andrewsh/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.57.0/src/ir/context.rs:531:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

ivmarkov, 2021-08-01:
Also posted in the chat:

One more idea: it might be clang itself. Bindgen is just a library crate which is a build dependency of esp-idf-sys. However it does use and does try to find the Clang .so library at runtime. So if somehow you have an oldish clang on your path, it might be picking up an oldish clang .so library, which - of course - knows nothing about the xtensa architecture

I had a case in the past with my own clang first on the path, and then an ancient distro clang further down in my path, and it was - for whatever reason - picking up the clang .so library from the distro's clang
I had to actually uninstall the distro clang to fix this issue (ubuntu 18.04)

In fact, @xymist had the same case, if I'm not mistaken.

Xymist, 2021-08-01:
I did. I had to remove Clang completely from my system and then build from scratch.

ivmarkov, 2021-08-01:
It is indeed another clang instance from where the clang .so library is (wrongly) picked up.

Seems that this fixes it:
`LIBCLANG_PATH=$HOME/xtensa-esp32-elf-clang/lib cargo`

closed 2021-08-01 by ivmarkov