esp-rs/esp-idf-sys #6: Error building esp-idf-sys 0.16.4

DavidGoodenough, 2021-08-09:
When I try to build a project with esp-idf-sys 0.16.4 I get the following error:

   Compiling esp-idf-sys v0.16.4
error: failed to run custom build command for `esp-idf-sys v0.16.4`

Caused by:
  process didn't exit successfully: `/home/david/src/ESP32/AcrRust/target/debug/build/esp-idf-sys-2326459d2b7cd194/build-script-build` (exit status: 1)
  --- stdout
  Found compatible PlatformIO Core 5.1.1 -> /home/david/.platformio/penv/bin/platformio

  --- stderr
  Error: Cannot derive default PIO platform, MCU and frameworks for target 'x86_64-unknown-linux-gnu'

This is on a linux (Debian) system, amd64

ivmarkov, 2021-08-10:
> --- stderr
> Error: Cannot derive default PIO platform, MCU and frameworks for target **'x86_64-unknown-linux-gnu'**

Noticed the above ^^^ (emphasis mine)?

You are building the `esp-idf-sys` for the wrong target. There is no way to build `esp-idf-sys` for Linux. Only for Espressif chips.

Which brings the question: are you building the `rust-esp32-std-hello` binary crate, and following the instructions? Because the `rust-esp32-std-hello` crate *does* have  a `.cargo/config.toml` file, where the correct Rust target is already specified?

Or are you trying to just build `esp-idf-sys` separately from `rust-esp32-std-hello`? That's possible, but you have to specify a bunch of command line parameters, like a correct `--target` as well as `-Zbuild-std=...`. If you are doing this, I don't see the purpose? As discussed - please try to work incrementally - build the `rust-esp32-satd-hello` crate first. It will also build `esp-idf-sys`.

DavidGoodenough, 2021-08-10:
On Tuesday, 10 August 2021 07:04:40 BST Ivan Markov wrote:
> Noticed the above ^^^ (emphasis mine)?
>
> You are building the `esp-idf-sys` for the wrong target. There is no way to
> build `esp-idf-sys` for Linux. Only for Espressif chips.
>
> Which brings the [...]
I was building rust-esp32-std-hello, this was just one of the stages is went through.

All I did was a "cargo build" in the rust-esp32-std-hello directory, no other parameters.  
The only thing I had configured was to tell rustup that the default toolchain was the esp 
one.

ivmarkov, 2021-08-10:
Did you switch to cargo nightly first, as described in the rust-esp32-std-hello README.md file?
Did it say something during the build process? Like 'invalid target'?

DavidGoodenough, 2021-08-10:
I had an old PlatformIO setup (now deleted) which created a bad target device (esp32-c3-devkitm-1) which the build process did not like.  So I removed the .platformio directory, and did a cargo clean.
Running cargo build ran through a lot of stuff successfully, but then failed - generating a log of 130KB, so I reran cargo build so as just to get the error step log, and it says:-
 cargo build
   Compiling esp-idf-sys v0.16.4
error: failed to run custom build command for `esp-idf-sys v0.16.4`

Caused by:
  process didn't exit successfully: `/home/david/git/rust-esp32-std-hello/target/debug/build/esp-idf-sys-54b8f62162d113ff/build-script-build` (exit status: 1)
  --- stdout
  Found compatible PlatformIO Core 5.1.1 -> /home/david/.platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/master_missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/esp32c3_atomics_emul.diff
  cargo:rerun-if-changed=./patches/master_restore_link_fragments_for_pio.diff
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32-c3-devkitm-1; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
  PLATFORM: Espressif 32 (3.3.1) > Espressif ESP32-C3-DevKitM-1
  HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
  DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
  PACKAGES: 
   - framework-espidf 3.40300.0 (4.3.0) 
   - tool-cmake 3.16.4 
   - tool-esptoolpy 1.30100.210531 (3.1.0) 
   - tool-ninja 1.7.1 
   - toolchain-riscv-esp 1.80400.0 (8.4.0)
  Reading CMake configuration...
  LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
  LDF Modes: Finder ~ chain, Compatibility ~ soft
  Found 0 compatible libraries
  Scanning dependencies...
  No dependencies
  Building in debug mode
  Retrieving maximum program size .pio/build/debug/firmware.elf
  Checking size .pio/build/debug/firmware.elf
  Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
  RAM:   [          ]   3.2% (used 10552 bytes from 327680 bytes)
  Flash: [=         ]  15.0% (used 156798 bytes from 1048576 bytes)
  ========================= [SUCCESS] Took 6.00 seconds =========================

  Environment    Status    Duration
  -------------  --------  ------------
  debug          SUCCESS   00:00:06.001
  ========================= 1 succeeded in 00:00:06.001 =========================
  cargo:CARGO_PIO_LINK_ARGS=--cargo-pio-link-linker=/home/david/.platformio/packages/toolchain-riscv-esp/bin/riscv32-esp-elf-gcc --cargo-pio-link-remove-duplicate-libs -L/home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf -L/home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug -L/home/david/.platformio/packages/framework-espidf/components/esp_wifi/lib/esp32c3 -L/home/david/.platformio/packages/framework-espidf/components/esp_rom/esp32c3/ld -L/home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/esp32c3 -L/home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/esp32c3/ld -L/home/david/.platformio/packages/framework-espidf/components/esp32c3/ld -Wl,--start-group /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/esp_pm/libesp_pm.a /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/asio/libasio.a /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/cbor/libcbor.a /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/unity/libunity.a /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/cmock/libcmock.a /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/coap/libcoap.a /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/console/libconsole.a /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/esp_adc_cal/libesp_adc_cal.a /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/esp_hid/libesp_hid.a /home/david/git/rust-esp32-std-hello/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d8190040417f69f/out/esp-idf/.pio/build/debug/esp-idf/protobuf-c/libprotobuf-c.a /home/david/git/rust-esp32-std-hell [... cut ...] ome/david/.platformio/packages/framework-espidf/components/wifi_provisioning/include", "-target", "riscv32"]
  /home/david/.platformio/packages/framework-espidf/components/esp_system/include/esp_system.h:19:10: fatal error: 'stdbool.h' file not found
  /home/david/.platformio/packages/framework-espidf/components/esp_system/include/esp_system.h:19:10: fatal error: 'stdbool.h' file not found, err: true
  Error: Failed to generate bindings

I have the full 130KB log of the complete build if needed.

ivmarkov, 2021-08-10:
> When I try to build a project with esp-idf-sys 0.16.4 I get the following error:
>
> Compiling esp-idf-sys v0.16.4
> error: failed to run custom build command for `esp-idf-sys v0.16.4`
>
> Caused by: [...]

How did you solve this??

DavidGoodenough, 2021-08-10:
I deleted .platformio from my home directory, did a cargo clean, and then a cargo build.  I had an old PlatformIO configuration lying around after deleting PlatformIO from my system as it clashed with other extensions of VSCode.

ivmarkov, 2021-08-10:
Can you uninstall any clang you might have on your system which is installed with e.g. a package manager and then make sure that you have the Espressif provided clang on your path?

DavidGoodenough, 2021-08-10:
By cland do you mean gcc or llvm?

ivmarkov, 2021-08-11:
By clang I mean clang - the c/c++ compiler toolchain based on llvn.

ivmarkov, 2021-08-11:
clang is a command line executable, as well as a library (libclang.so) which can be produced when you are compiling the LLVM toolchain. The thing is, bindgen is using the libclang.so to generate the unsafe bindings to ESP-IDF APIs as part of compiling the `esp-idf-sys` crate.

Now, in theory, bindgen should just take the first libclang.so it finds on the $PATH (it has a complicated algorithm to detect the library which comes from another crate that it is using). Unfortunately we noticed that even if you put the xclang compiled from Espressif as first on your `$PATH`, bindgen still has the habit of finding "other" instance of the clang library, especially if you have the clang installed using the OS system packager. So in the worst case - yes - you might have to uninstall the system clang. Now, if the problem comes from the LLVM that your Scala environment is using (no idea how that is setup though - if LLVM is just statically linked into scalac, you should not have that issue!), then yes - you might have to even temporarily remove the native scala environment or at least make sure it is not on your `$PATH`.

What we found as a workaround that might not require uninstalling other clangs is this command line:
`LIBCLANG_PATH=$HOME/xtensa-esp32-elf-clang/lib cargo build`. You might try that first.

DavidGoodenough, 2021-08-11:
OK I fixed it.

In https://github.com/espressif/rust-esp32-example/blob/main/docs/rust-on-xtensa-installation-x86_64-unknown-linux-gnu.md it does not which directory you should run the commands in.  I like to download TARs into a Download/TARs file, so I moved there.  Unfortunately the extraction put three directories in the wrong place.  The commands (at least the tar xvf ... ones) MUST be run in your HOME directory.  I had assumed that the unpacked files were installed by the subsequent commands, but that would appear not to be the case.

However even after doing this, I still get the stdbool.h problem, but now that the above is fixed using:-

LIBCLANG_PATH=$HOME/xtensa-esp32-elf-clang/lib cargo build

fixes that and it builds cleanly - Hooray!

closed 2021-08-22 by ivmarkov