esp-rs/esp-idf-sys #8: LoadProhibited exception when using `EspMutex`

N3xed, 2021-08-22:
The implementation for `EspMutex` was recently changed from using a `pthread_mutex_t` to a `*mut pthread_mutex_t`. This is incorrect as when locking, the [`pthread_mutex_lock`](https://github.com/espressif/esp-idf/blob/1ca2afd982230e4cbf021c58137f3141c1870572/components/pthread/pthread.c#L654) function calls [`pthread_mutex_init_if_static`](https://github.com/espressif/esp-idf/blob/1ca2afd982230e4cbf021c58137f3141c1870572/components/pthread/pthread.c#L641) which dereferences the mutex pointer. Because this pointer is initially set to `PTHREAD_MUTEX_INITIALIZER` (`0xFFFFFFFF`), this causes a load to this address which is invalid and causes a `LoadProhibited` exception.

ivmarkov, 2021-08-22:
Sorry for this stupid mistake. Hopefully fixed now.

N3xed, 2021-08-22:
Yup, fixed it.

closed 2021-08-22 by N3xed