esp-rs/esp-idf-sys #11: Decide tools installation directory

N3xed, 2021-09-10:


N3xed, 2021-10-17:
Okay so I think what we decided on was:

For the native build:
- Global install dir: `~/.espressif`
- Local install dir (used by default): `<workspace-dir>/.embuild/espressif`

For the pio build:
- Continue to use the global install dir `~/.platformio`, which is the pio default.

N3xed, 2021-10-17:
Well, installing the tools into `~/.espressif` shouldn't cause issues, I see that (because this uses the exact same mechanism to install the tools as the esp-idf itself, as far as I know). But the problem was with installing the esp-idf source. Currently, both the source and tools are always installed in the same directory. And when installing the source in the global location and patching it, that could cause problems, should the user installation of the esp-idf source and our installation coincide.

With #19, this problem still exists, as again, both the source and tools are always installed in the same location. Changing this would make it more complicated. 
The default is to install both things locally, but the user can opt into the global install. Also, I think the goal is to eventually not have to patch the esp-idf. Additionally, unless a specific hash is installed, the esp-idf source folder always gets the tag or branch 
name suffixed so this should also prevent collisions in most cases.

So with that, I think that this shouldn't really be an issue. Please tell me if you think otherwise.

> For IDF itself, it is typically not installed into ~/.espressif

That's good to know.

ivmarkov, 2021-12-04:
The PlatformIO aspect of this should now be fixed by https://github.com/esp-rs/esp-idf-sys/commit/9022587a6fdd9defb7f1e2eec8c4100705407b16

I'll publish a new release now, with these changes as well as the previous ones from @N3xed that unified the handling of `sdkconfig*` files and glob file copies.

closed 2021-12-04 by ivmarkov