esp-rs/esp-idf-sys #14: Error when building on Linux aarch64

skerkour, 2021-09-21:
Hello,

When trying to build on Linux aarch64 (Raspberry pi 4) I got the following error:

```
error: failed to run custom build command for `esp-idf-sys v0.20.0`

Caused by:
  process didn't exit successfully: `/home/ubuntu/rust-esp32-std-hello/target/debug/build/esp-idf-sys-791a8c4daff86dfa/build-script-build` (exit status: 1)
  --- stdout
  Found compatible PlatformIO Core 5.2.0 -> /home/ubuntu/.platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/ping_setsockopt_fix.diff
  cargo:rerun-if-changed=./patches/master_missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=/home/ubuntu/rust-esp32-std-hello/./sdkconfig.defaults
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32cam; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Tool Manager: Installing platformio/toolchain-xtensa32 @ ~2.80400.0

  --- stderr
  Error: Could not find the package with 'platformio/toolchain-xtensa32 @ ~2.80400.0' requirements for your system 'linux_aarch64'
  Error: No such file or directory (os error 2)
```

Then:
```shell
$ pio platform install espressif32
Platform Manager: espressif32 @ 3.3.2 is already installed
Tool Manager: toolchain-xtensa32 @ 2.50200.97 is already installed
Tool Manager: tool-esptoolpy @ 1.30100.210531 is already installed
The platform 'espressif32' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
```

So I successfully hacked my way by manually modifying `platforms/espressif32/platform.py` to require `toolchain-xtensa32 @ 2.50200.97` instead of `toolchain-xtensa32 @ 2.80400.0`

But then I got the following error:

```
error: failed to run custom build command for `esp-idf-sys v0.20.0`

Caused by:
  process didn't exit successfully: `/home/ubuntu/rust-esp32-std-hello/target/debug/build/esp-idf-sys-791a8c4daff86dfa/build-script-build` (exit status: 1)
  --- stdout
  Found compatible PlatformIO Core 5.2.0 -> /home/ubuntu/.platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/ping_setsockopt_fix.diff
  cargo:rerun-if-changed=./patches/master_missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=/home/ubuntu/rust-esp32-std-hello/./sdkconfig.defaults
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32cam; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0

  --- stderr
  Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
  Error: No such file or directory (os error 2)
```

```shell
$ pio platform install toolchain-esp32ulp
Platform Manager: Installing toolchain-esp32ulp
Error: Could not find the package with 'toolchain-esp32ulp' requirements for your system 'linux_aarch64'
```

And can't find a solution for this error. 

Any idea?

georgik, 2021-09-21:
Sure. It makes sense to keep it as open. I'll add it as discussion point for next community meeting: https://github.com/esp-rs/rust/discussions/80

georgik, 2021-11-16:
@JurajSadel Steps:
- build Python Wheels for aarch64 and publish to dl.espressif.com/pypi
- verify status of publishing aarch64 toolchains from ESP-IDF (tools/tools.json)
- verify RISC-V and aarch64 for ESP32-C3 - compile https://github.com/espressif/rust-esp32-example
- build Rust Toolchain for aarch64 Linux - esp-rs/rust-build
- update installation scripts for Xtensa Rust toolchain

closed 2022-02-07 by ivmarkov

RepRapid, 2022-03-14:
I installed the Rust tool-chain last night, as I saw their was an update. It was a fresh install on windows 11 running in a ubuntu docker container, as per the instructions. I understand the test-rust-toolchain.sh is not yet working for the version 4.3 as I am getting an error there. Also I haven't plugged the board in yet, but I noticed that file doesn't have the S3 target in there either. I'm pretty sure the version of IDF I have installed is 4.4 but its calling 4.3 I didn't notice that mentioned in the .sh file. Should I just disregard the test-rust-toolchain.sh? And just try build a project I have for arm64, with the esp32-s3 target? the ulp_gcc is working now, or is it still under development?

JurajSadel, 2022-03-14:
Hello @RepRapid! 
ESP-IDF4.3 doesn't have needed patches yet. Currently, `esp-idf-sys` is always downloading new ESP-IDF even if you have installed one on your machine - [here](https://github.com/ivmarkov/embuild/pull/29) is opened PR for this.

The [default ESP-IDF version](https://github.com/esp-rs/esp-idf-sys/blob/master/build/native.rs#L32) that `esp-idf-sys` builds is `4.3.2`. As a workaround, you can override the default version with the `ESP_IDF_VERSION` environment variable.
`export ESP_IDF_VERSION=release/v4.4` and then clean build should work on aarch64 Linux.

At this moment, `ULP coprocessor` toolchain is not ready for aarch64 Linux, as mentioned [here](https://github.com/espressif/esp-idf/issues/6432#issuecomment-1066177007)

PR #62 (2022-02-07): build: remove ulp_gcc_toolchain for aarch64 linux
Together with the [add cmake for aarch64 linux PR](https://github.com/ivmarkov/embuild/pull/41) it should be able to build [std-demo](https://github.com/ivmarkov/rust-esp32-std-demo) on RPi (aarch64 linux).
Closes #14