esp-rs/esp-idf-sys #15: Docs do not build on docs.rs

lily-mara, 2021-09-24:
The docs do not build on docs.rs

[build log](https://docs.rs/crate/esp-idf-svc/0.20.1/builds/437848)

```
# rustc version
rustc 1.57.0-nightly (5ecc8ad84 2021-09-19)
# docs.rs version
docsrs 0.6.0 (432e30a 2021-09-12)

# build log
[INFO] running `Command { std: "docker" "create" "-v" "/home/cratesfyi/workspace/builds/esp-idf-svc-0.20.1/target:/opt/rustwide/target:rw,Z" "-v" "/home/cratesfyi/workspace/builds/esp-idf-svc-0.20.1/source:/opt/rustwide/workdir:ro,Z" "-v" "/home/cratesfyi/workspace/cargo-home:/opt/rustwide/cargo-home:ro,Z" "-v" "/home/cratesfyi/workspace/rustup-home:/opt/rustwide/rustup-home:ro,Z" "-e" "SOURCE_DIR=/opt/rustwide/workdir" "-e" "CARGO_TARGET_DIR=/opt/rustwide/target" "-e" "DOCS_RS=1" "-e" "CARGO_HOME=/opt/rustwide/cargo-home" "-e" "RUSTUP_HOME=/opt/rustwide/rustup-home" "-w" "/opt/rustwide/workdir" "-m" "3221225472" "--cpus" "2" "--user" "1001:1001" "--network" "none" "ghcr.io/rust-lang/crates-build-env/linux@sha256:a89201d6b5b9fd45c15d5952ece0a0129e7e80cb26ec023fd59cf09bc26f1604" "/opt/rustwide/cargo-home/bin/cargo" "+nightly" "rustdoc" "--lib" "-Zrustdoc-map" "-Zunstable-options" "--config=doc.extern-map.registries.crates-io=\"https://docs.rs/{pkg_name}/{version}/x86_64-unknown-linux-gnu\"" "-j2" "--" "-Z" "unstable-options" "--emit=invocation-specific" "--resource-suffix" "-20210919-1.57.0-nightly-5ecc8ad84" "--static-root-path" "/" "--cap-lints" "warn" "--disable-per-crate-search", kill_on_drop: false }`
[INFO] [stdout] f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387
[INFO] [stderr] WARNING: Your kernel does not support swap limit capabilities or the cgroup is not mounted. Memory limited without swap.
[INFO] running `Command { std: "docker" "start" "-a" "f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387", kill_on_drop: false }`
[INFO] [stderr]    Compiling esp-idf-sys v0.20.0
[INFO] [stderr] error: failed to run custom build command for `esp-idf-sys v0.20.0`
[INFO] [stderr] 
[INFO] [stderr] Caused by:
[INFO] [stderr]   process didn't exit successfully: `/opt/rustwide/target/debug/build/esp-idf-sys-a2bf8941e6b245cc/build-script-build` (exit status: 1)
[INFO] [stderr]   --- stdout
[INFO] [stderr]   Installer version: 1.0.2
[INFO] [stderr]   Platform: Linux-5.4.0-1037-aws-x86_64-with-glibc2.29
[INFO] [stderr]   Python version: 3.8.10 (default, Jun  2 2021, 10:49:15) 
[INFO] [stderr]   [GCC 9.4.0]
[INFO] [stderr]   Python path: /usr/bin/python3
[INFO] [stderr]   Creating a virtual environment at /.platformio/penv
[INFO] [stderr]   /usr/bin/python3
[INFO] [stderr] 
[INFO] [stderr]   --- stderr
[INFO] [stderr]   Error: Compatible PlatformIO Core not found.
[INFO] [stderr]   Reason: PlatformIO executable not found in `/.platformio/penv/bin`
[INFO] [stderr]   Traceback (most recent call last):
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connection.py", line 169, in _new_conn
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/util/connection.py", line 73, in create_connection
[INFO] [stderr]     File "/usr/lib/python3.8/socket.py", line 918, in getaddrinfo
[INFO] [stderr]       for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
[INFO] [stderr]   socket.gaierror: [Errno -3] Temporary failure in name resolution
[INFO] [stderr] 
[INFO] [stderr]   During handling of the above exception, another exception occurred:
[INFO] [stderr] 
[INFO] [stderr]   Traceback (most recent call last):
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connectionpool.py", line 699, in urlopen
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connectionpool.py", line 382, in _make_request
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connectionpool.py", line 1010, in _validate_conn
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connection.py", line 353, in connect
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connection.py", line 181, in _new_conn
[INFO] [stderr]   urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7f96e5f5b340>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution
[INFO] [stderr] 
[INFO] [stderr]   During handling of the above exception, another exception occurred:
[INFO] [stderr] 
[INFO] [stderr]   Traceback (most recent call last):
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/adapters.py", line 439, in send
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/connectionpool.py", line 755, in urlopen
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/urllib3/util/retry.py", line 574, in increment
[INFO] [stderr]   urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/python-portable (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f96e5f5b340>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
[INFO] [stderr] 
[INFO] [stderr]   During handling of the above exception, another exception occurred:
[INFO] [stderr] 
[INFO] [stderr]   Traceback (most recent call last):
[INFO] [stderr]     File "/tmp/.tmpJjdIul", line 68, in <module>
[INFO] [stderr]       main()
[INFO] [stderr]     File "/tmp/.tmpJjdIul", line 60, in main
[INFO] [stderr]       bootstrap()
[INFO] [stderr]     File "/tmp/.tmpJjdIul", line 46, in bootstrap
[INFO] [stderr]       pioinstaller.__main__.main()
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/pioinstaller/__main__.py", line 131, in main
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/click/core.py", line 829, in __call__
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/click/core.py", line 782, in main
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/click/core.py", line 1236, in invoke
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/click/core.py", line 1066, in invoke
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/click/core.py", line 610, in invoke
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/click/decorators.py", line 21, in new_func
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/pioinstaller/core.py", line 67, in install_platformio_core
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/pioinstaller/core.py", line 95, in _install_platformio_core
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/pioinstaller/penv.py", line 58, in create_core_penv
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/pioinstaller/python.py", line 70, in fetch_portable_python
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/pioinstaller/python.py", line 97, in get_portable_python_url
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/api.py", line 76, in get
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/api.py", line 61, in request
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/sessions.py", line 542, in request
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/sessions.py", line 655, in send
[INFO] [stderr]     File "/tmp/.piocore-installer-pq16kjv6/tmphidq62b1/pioinstaller.zip/requests/adapters.py", line 516, in send
[INFO] [stderr]   requests.exceptions.ConnectionError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/python-portable (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at [... cut ...]
[INFO] running `Command { std: "docker" "inspect" "f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387", kill_on_drop: false }`
[INFO] running `Command { std: "docker" "rm" "-f" "f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387", kill_on_drop: false }`
[INFO] [stdout] f9c12e04431be89cc02118eae1b9c9eee59977444a9b61c055ef566694544387
```

It seems that the docs.rs build environment does not have a networking stack available, so the platformio installation script fails. Is there any way to provide fallback information in the build.rs script so that the docs can be built without networking?

lily-mara, 2021-09-24:
So I'm not very knowledgable about cross-compiling or embedded Rust, but I can try to take a crack at this. I do think it's worthwhile to investigate this. Even though the textual documentation on this crate and esp-idf-svc are minimal, I constantly use rustdoc generated docs to see how different types slot together. It's probably also easier to justify writing good docs if we know that those docs can actually be read by someone.

Documentation on docs.rs confirms that networking is not available for documentation builds (https://docs.rs/about/builds#hitting-resource-limits). I think to get around this, we would need to vendor the esp-idf C library as a git submodule or something.

Another alternative is to build the docs ourselves and publish them to github pages linked in the readme. This is less work today, but it does require more effort each time a new version is published.

I am able to build the docs locally with the esp compiler fork and the appropriate target, FWIW. @ivmarkov do you have any thoughts on which method we should take to resolving?

ivmarkov, 2021-09-24:
It is not just the ESP-IDF C library, right, it is also the whole toolchain that is necessary to compile the thing (as in the GCC toolchain for that target).

I'm thinking of an alternative: perhaps I can implement a build option (using a feature or something - but really have to see how much flexibility is there in the `docs.rs` build system) - where instead of compiling the ESP-IDF toolchain, we are actually using a pre-generated `bindings.rs` file. Since none of the crates in question (`esp-idf-svc`, `esp-idf-hal`, and most importantly - `esp-idf-sys` - from where the trouble comes) really **link** against the C library (they just use the autogenerated unsafe APIs from the `bindings.rs` file to build higher level abstractions) - that approach might work and circumvent the whole vendoring thing. Now, how exactly this pre-generated `bindings.rs` file will be generated and maintained up to date - is an interesting question by itself. Perhaps a GitHub action would do, as these are capable of doing networking.

N3xed, 2021-09-25:
One problem is also that the generated bindings depend on the generated `sdkconfig.h` by the esp-idf and on the selected mcu. So it isn't really clear which bindings should be displayed on `docs.rs`.

Maybe one approach would be to make the bindings for every mcu available in different modules, and just set sane defaults for the `sdkconfig`. Or, like @lily-mara already mentioned, host the docs ourselves, but separate docs for every mcu.

Also, that's the reason we can't/shouldn't vendor the bindings in the repository, again because they only correspond to one mcu and `sdkconfig`.

ivmarkov, 2021-09-25:
> Maybe one approach would be to make the bindings for every mcu available in different modules, and just set sane defaults for the `sdkconfig`. Or, like @lily-mara already mentioned, host the docs ourselves, but separate docs for every mcu.

Having bindings per-MCU is not really solving the problem completely but might be part of the solution. While indeed there are differences from MCU to MCU (pin number and layout as well as number of SPI devices in `esp-idf-hal` immediately come to mind), the problem is exacerbated by the gazillion of config options in the `kconfig` menu system of the ESP-IDF. In other words, if you don't explicitly enable the BT via menuconfig, you don't get unsafe BT bindings in `bindings.rs` and - by extension - you don't get *safe* abstraction on top of these in `esp-idf-svc` (not that we have safe abstractions for BT yet). Ditto for HTTPS, NAPT, and the list will keep growing.

So what we probably need is an auto-generated `bindings.rs` with **all** config options enabled (somehow), which is - optionally - per MCU.

> Also, that's the reason we can't/shouldn't vendor the bindings in the repository, again because they only correspond to one mcu and `sdkconfig`.

Indeed. The `bindings.rs` file used to be generated manually in the past and committed to the repo, but that was only usable for toy projects and just for one MCU.

Yet, (auto-)generating `bindings.rs` (per MCU?) with all `kconfig` options enabled, committing it to a special place in the repo and then using it *only* when compiling the crates in a "please generate docs" mode (if that's even possible) seems like the most plausible solution to me. With the benefit that we get the docs **in** crates.io, where they belong.

N3xed, 2021-09-25:
> Yet, (auto-)generating `bindings.rs` (per MCU?) with all `kconfig` options enabled, committing it to a special place in the repo and then using it _only_ when compiling the crates in a "please [...]

I don't even think we need to commit these bindings to the repository, instead, when publishing the crate we have to just publish the vendored bindings with it (and maybe automate this with a github action). But maybe we want to commit them so that someone that clones the repository is able to build the documentation without having LLVM installed (and without building the whole `esp-idf`)?

At any case, I think the way forward with the `docs.rs` approach would be to add modules for the bindings of every mcu, and then export the bindings of `esp32c3` (ex. `pub use bindings_esp32c3::*`) as that's the only one supported with upstream rust. But only do this in the special case when building documentation. In the default case, it should use the locally generated bindings as we do currently (with `include!(...)`).

For this, we also need to provide `sdkconfig`s for every mcu that enable all settings (or something to that effect) which are used to generate the bindings.

ivmarkov, 2021-09-25:
> I don't even think we need to commit these bindings to the repository, instead, when publishing the crate we have to just publish the vendored bindings with it (and maybe automate this with a github [...]

Yes, that reason and also if someone wants to browse the `esp-idf-sys/../bindings.rs` source code in GitHub.
Not strictly necessary, nice to have maybe.

> 
> At any case, I think the way forward with the `docs.rs` approach would be to add modules for the bindings of every mcu, and then export the bindings of `esp32c3` (ex. `pub use bindings_esp32c3::*`) as that's the only one supported with upstream rust atm. But only do this in the special case when building documentation. In the default case, it should use the locally generated bindings as we do currently (with `include!(...)`).
> 
> For this, we also need to provide `sdkconfig`s for every mcu that enable all settings (or something to that effect) which are used to generate the bindings.

Agreed on all points.

lily-mara, 2021-09-28:
Unfortunately I don't think I know enough about embuild or cross-compiling to create the per-chip setup that you're talking about. I think I'd need to punt this to someone else.

MabezDev, 2022-02-01:
Can this be closed now? Whilst it still isn't being built on docs.rs, we have links from docs.rs & crates.io to our hosted documentation here.

lily-mara, 2022-02-01:
Yes, thanks for adding the self-hosted docs!

closed 2022-02-01 by lily-mara