esp-rs/esp-idf-sys #25: Error while building with the pio feature

1c3t3a, 2021-11-01:
Hey there! While building the library I stumbled upon an error:

```
cargo b
   Compiling esp-idf-sys v0.24.1
error: failed to run custom build command for `esp-idf-sys v0.24.1`

Caused by:
  process didn't exit successfully: `.../code/esp/rust-esp32-std-mini/target/debug/build/esp-idf-sys-17ca833ec6c37798/build-script-build` (exit status: 1)
  --- stdout
  Found compatible PlatformIO Core 5.2.2 -> /.../.platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/ping_setsockopt_fix.diff
  cargo:rerun-if-changed=/.../rust-esp32-std-mini/./sdkconfig.debug
  cargo:rerun-if-changed=/.../esp/rust-esp32-std-mini/./sdkconfig.defaults
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32-c3-devkitm-1; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
  PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32-C3-DevKitM-1
  HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
  DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
  PACKAGES: 
   - framework-espidf 3.40300.0 (4.3.0) 
   - tool-cmake 3.16.4 
   - tool-esptoolpy 1.30100.210531 (3.1.0) 
   - tool-ninja 1.9.0 
   - toolchain-riscv-esp 1.80400.0 (8.4.0)
  Reading CMake configuration...
  Generating assembly for certificate bundle...
  LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
  LDF Modes: Finder ~ chain, Compatibility ~ soft
  Found 0 compatible libraries
  Scanning dependencies...
  No dependencies
  Building in debug mode

  --- stderr
  CMake Error at /../.platformio/packages/framework-espidf/tools/cmake/scripts/data_file_embed_asm.cmake:26 (file):
    file failed to open for reading (No such file or directory):

      /../rust-esp32-std-mini/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-ee61aad2301f479b/out/esp-idf/.pio/build/debug/x509_crt_bundle

  *** Error 1
  *** [.pio/build/debug/.pio/build/debug/x509_crt_bundle.o] Source `.pio/build/debug/x509_crt_bundle.S' not found, needed by target `.pio/build/debug/.pio/build/debug/x509_crt_bundle.o'.
  ========================== [FAILED] Took 3.09 seconds ==========================
  Environment    Status    Duration
  -------------  --------  ------------
  debug          FAILED    00:00:03.087
  ==================== 1 failed, 0 succeeded in 00:00:03.087 ====================
  Error: No such file or directory (os error 2)

  Stack backtrace:
     0: std::backtrace_rs::backtrace::libunwind::trace
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
     1: std::backtrace_rs::backtrace::trace_unsynchronized
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
     2: std::backtrace::Backtrace::create
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/backtrace.rs:327:13
     3: anyhow::error::<impl core::convert::From<E> for anyhow::Error>::from
               at /Users/basti/.cargo/registry/src/github.com-1ecc6299db9ec823/anyhow-1.0.44/src/error.rs:530:25
     4: <core::result::Result<T,F> as core::ops::try_trait::FromResidual<core::result::Result<core::convert::Infallible,E>>>::from_residual
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/core/src/result.rs:1915:27
     5: embuild::pio::project::SconsVariables::from_dump
               at /Users/basti/.cargo/registry/src/github.com-1ecc6299db9ec823/embuild-0.25.1/src/pio/project.rs:85:36
     6: build_script_build::build_driver::build
               at ./build/pio.rs:54:30
     7: build_script_build::main
               at ./build/build.rs:25:24
     8: core::ops::function::FnOnce::call_once
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/core/src/ops/function.rs:227:5
     9: std::sys_common::backtrace::__rust_begin_short_backtrace
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/sys_common/backtrace.rs:123:18
    10: std::rt::lang_start::{{closure}}
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/rt.rs:146:18
    11: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/core/src/ops/function.rs:259:13
    12: std::panicking::try::do_call
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/panicking.rs:403:40
    13: std::panicking::try
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/panicking.rs:367:19
    14: std::panic::catch_unwind
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/panic.rs:133:14
    15: std::rt::lang_start_internal::{{closure}}
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/rt.rs:128:48
    16: std::panicking::try::do_call
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/panicking.rs:403:40
    17: std::panicking::try
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/panicking.rs:367:19
    18: std::panic::catch_unwind
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/panic.rs:133:14
    19: std::rt::lang_start_internal
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/rt.rs:128:20
    20: std::rt::lang_start
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/rt.rs:145:17
    21: <unknown>
               at ./build/build.rs:52:6
```

I am building this library via the pio feature and also have the `binstart` feature enabled. This log is from building the newest version (`v0.24.1`) but I experienced the error with every previous version I tried as well.

1c3t3a, 2021-11-01:
By the way: Building with the feature `native` runs through, but results in a malformed binary on the esp32c3:
```
entry 0x403ce000
I (35) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (35) boot: compile time 12:10:40
I (35) boot: chip revision: 3
I (38) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (45) boot.esp32c3: SPI Speed      : 80MHz
I (50) boot.esp32c3: SPI Mode       : DIO
I (55) boot.esp32c3: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 003f0000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 3, min. application chip revision: 0
I (110) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=178d8h ( 96472) map
I (133) esp_image: segment 1: paddr=00027900 vaddr=3fc8b000 size=01cd8h (  7384) load
I (134) esp_image: segment 2: paddr=000295e0 vaddr=40380000 size=06a38h ( 27192) load
I (143) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=8bd44h (572740) map
I (233) esp_image: segment 4: paddr=000bbd6c vaddr=40386a38 size=044f4h ( 17652) load
I (236) esp_image: segment 5: paddr=000c0268 vaddr=50000000 size=00010h (    16) load
E (239) esp_image: invalid segment length 0xbada
E (244) boot: Factory app partition is not bootable
E (249) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d153e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
load:0x403d0000,len:0x2ce0
```

MabezDev, 2021-11-02:
Thanks for the report! I'll take a look at the native build issues.

For PIO you, please see https://github.com/espressif/rust-esp32-example/issues/42#issuecomment-955000615

1c3t3a, 2021-11-02:
Ah great! The pio fix is working! If you need me to test something with the native build just ping me here ;)

ivmarkov, 2021-11-05:
Another way to fix that would be to place [this sdkconfig.defaults file in the root of your binary crate](https://github.com/ivmarkov/esp-idf-template/blob/master/cargo/sdkconfig.defaults).

I would strongly advise generating your project using the [esp-idf-template project](https://github.com/ivmarkov/esp-idf-template) now.

closed 2021-11-08 by ivmarkov