esp-rs/esp-idf-sys #27: Slow rebuilds with `native` feature

MabezDev, 2021-11-04:
I think there was another issue floating around about slow builds, but I can't seem to find it now.

It seems making any change to the build when using the `native` feature forces an entire rebuild (possible recloning too?). Perhaps the file tracking for rebuilds on native build is tracking something it shouldn't.

imxood, 2021-11-04:
in .cargo/config.toml

try to add code:

[env]
ESP_IDF_GLOBAL_INSTALL = { value = "y" }

MabezDev, 2021-11-05:
The simplest way to reproduce is to:

* build `esp-idf-sys`.
* change one line anywhere, e.g comment out something in `bindings.h`
* run the build with `-vvv`

The output looks like this:

```rust
Running `/Users/mabez/Development/rust/embedded/util/esp-idf-sys/target/debug/build/esp-idf-sys-43f95c9f1a659274/build-script-build`
[esp-idf-sys 0.24.3] cargo:rerun-if-env-changed=ESP_IDF_INSTALL_DIR
[esp-idf-sys 0.24.3] cargo:rerun-if-env-changed=ESP_IDF_GLOBAL_INSTALL
[esp-idf-sys 0.24.3] cargo:rerun-if-env-changed=ESP_IDF_VERSION
[esp-idf-sys 0.24.3] cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
[esp-idf-sys 0.24.3] cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
[esp-idf-sys 0.24.3] cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
[esp-idf-sys 0.24.3] cargo:rerun-if-env-changed=ESP_IDF_EXTRA_TOOLS
[esp-idf-sys 0.24.3] cargo:rerun-if-env-changed=MCU
[esp-idf-sys 0.24.3] Cloning into '/Users/mabez/Development/rust/embedded/util/esp-idf-sys/.embuild/espressif/esp-idf-master'...
^C  Building [=======================> ] 210/212: esp-idf-sys(build)
```

As you can see it starts the clone again. Maybe this is because I am setting `ESP_IDF_VERSION="branch:master"`?

N3xed, 2021-11-05:
Yup, this should not be happening.

> Maybe this is because I am setting `ESP_IDF_VERSION="branch:master"`?

I guess it has to do with the build script not recognizing which rev that is checked out.

Let me investigate this issue.

N3xed, 2021-11-07:
@MabezDev Please try it again with the newest `embuild` version.

If it still happens:
Maybe try again after `cargo clean`, otherwise please give me the complete `esp-idf-sys` build log and the output of `git describe --all --exact-match` in the esp-idf repo, so that we can debug this further.

N3xed, 2021-12-05:
I'll close this for now.
Please reopen if this issue still persists.

closed 2021-12-05 by N3xed

TheLexoPlexx, 2023-10-06:
I might be stumbling upon the same error:

`cargo generate esp-rs/esp-idf-template cargo`

Done as requested: `cargo clean` and then `cargo run -vvv`:

```log
   Compiling embuild v0.31.3
     Running `CARGO=/home/alex/.rustup/toolchains/esp/bin/cargo CARGO_CRATE_NAME=embuild CARGO_MANIFEST_DIR=/home/alex/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embuild-0.31.3 CARGO_PKG_AUTHORS='Ivan Markov <ivan.markov@gmail.com>:Dominik Gschwind <dominik.gschwind99@gmail.com>' CARGO_PKG_DESCRIPTION='A build support library for embedded Rust' CARGO_PKG_HOMEPAGE='' CARGO_PKG_LICENSE='MIT OR Apache-2.0' CARGO_PKG_LICENSE_FILE='' CARGO_PKG_NAME=embuild CARGO_PKG_README=README.md CARGO_PKG_REPOSITORY='https://github.com/ivmarkov/embuild' CARGO_PKG_RUST_VERSION=1.58 CARGO_PKG_VERSION=0.31.3 CARGO_PKG_VERSION_MAJOR=0 CARGO_PKG_VERSION_MINOR=31 CARGO_PKG_VERSION_PATCH=3 CARGO_PKG_VERSION_PRE='' ESP_IDF_VERSION=v5.1 LD_LIBRARY_PATH='/home/alex/Projekte/rust-esp32-can/target/debug/deps:/home/alex/.rustup/toolchains/esp/lib:/home/alex/.rustup/toolchains/esp/lib' /home/alex/.rustup/toolchains/esp/bin/rustc --crate-name embuild --edition=2021 /home/alex/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embuild-0.31.3/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --diagnostic-width=282 --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="bindgen"' --cfg 'feature="cmake"' --cfg 'feature="default"' --cfg 'feature="dep-cmake"' --cfg 'feature="espidf"' --cfg 'feature="git"' --cfg 'feature="glob"' --cfg 'feature="globwalk"' --cfg 'feature="home"' --cfg 'feature="kconfig"' --cfg 'feature="remove_dir_all"' --cfg 'feature="serde"' --cfg 'feature="serde_json"' --cfg 'feature="strum"' --cfg 'feature="tempfile"' --cfg 'feature="which"' -C metadata=4f41d01e6eaeecfa -C extra-filename=-4f41d01e6eaeecfa --out-dir /home/alex/Projekte/rust-esp32-can/target/debug/deps -L dependency=/home/alex/Projekte/rust-esp32-can/target/debug/deps --extern anyhow=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libanyhow-4c2359ea5232b878.rmeta --extern bindgen=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libbindgen-d56a4e5eaf815806.rmeta --extern bitflags=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libbitflags-1d31dfe9f72fab4b.rmeta --extern dep_cmake=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libcmake-f32389259794c0a9.rmeta --extern filetime=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libfiletime-9098541a125e3039.rmeta --extern globwalk=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libglobwalk-72307c9d8b555073.rmeta --extern home=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libhome-8e42d57892ccda2a.rmeta --extern log=/home/alex/Projekte/rust-esp32-can/target/debug/deps/liblog-9a3b409b1bbdebf2.rmeta --extern remove_dir_all=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libremove_dir_all-309290c7504b1a6b.rmeta --extern serde=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libserde-deddf976a4504fed.rmeta --extern serde_json=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libserde_json-843f7166d498ecb9.rmeta --extern shlex=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libshlex-38c491047c11ee6d.rmeta --extern strum=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libstrum-5cd27db5eb9b8daf.rmeta --extern tempfile=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libtempfile-e0164d78aee18eeb.rmeta --extern thiserror=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libthiserror-0d6b4b4fb443787c.rmeta --extern which=/home/alex/Projekte/rust-esp32-can/target/debug/deps/libwhich-9ab93572e136d233.rmeta --cap-lints warn`
       Dirty esp-idf-sys v0.33.2: dependency info changed
   Compiling esp-idf-sys v0.33.2
       Dirty rust-esp32-can v0.1.0 (/home/alex/Projekte/rust-esp32-can): dependency info changed
   Compiling rust-esp32-can v0.1.0 (/home/alex/Projekte/rust-esp32-can)
     Running `CARGO=/home/alex/.rustup/toolchains/esp/bin/cargo CARGO_CRATE_NAME=build_script_build CARGO_MANIFEST_DIR=/home/alex/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.2 CARGO_PKG_AUTHORS='Alexey Arbuzov <aarbuzov@termt.com>:sapir <yasapir@gmail.com>:Ivan Markov <ivan.markov@gmail.com>:Dominik Gschwind <dominik.gschwind99@gmail.com>' CARGO_PKG_DESCRIPTION='Bindings for ESP-IDF (Espressif'\''s IoT Development Framework)' CARGO_PKG_HOMEPAGE='' CARGO_PKG_LICENSE='MIT OR Apache-2.0' CARGO_PKG_LICENSE_FILE='' CARGO_PKG_NAME=esp-idf-sys CARGO_PKG_README=README.md CARGO_PKG_REPOSITORY='https://github.com/esp-rs/esp-idf-sys' CARGO_PKG_RUST_VERSION=1.66 CARGO_PKG_VERSION=0.33.2 CARGO_PKG_VERSION_MAJOR=0 CARGO_PKG_VERSION_MINOR=33 CARGO_PKG_VERSION_PATCH=2 CARGO_PKG_VERSION_PRE='' ESP_IDF_VERSION=v5.1 LD_LIBRARY_PATH='/home/alex/Projekte/rust-esp32-can/target/debug/deps:/home/alex/.rustup/toolchains/esp/lib:/home/alex/.rustup/toolchains/esp/lib' /home/alex/.rustup/toolchains/esp/bin/rustc --crate-name build_script_build --edition=2021 /home/alex/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.2/build/build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --diagnostic-width=282 --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 --cfg 'feature="binstart"' --cfg 'feature="native"' --cfg 'feature="std"' -C metadata=e20b5e4b28e472fe -C extra-filename=-e20b5e4b28e472fe --out-dir /home/alex/Projekte/rust-esp32-can/targe [... cut ...]
[esp-idf-sys 0.33.2] Klone nach '/home/alex/Projekte/rust-esp32-can/.embuild/espressif/esp-idf/v5.1/components/openthread/openthread'...
[esp-idf-sys 0.33.2] Klone nach '/home/alex/Projekte/rust-esp32-can/.embuild/espressif/esp-idf/v5.1/components/protobuf-c/protobuf-c'...
[esp-idf-sys 0.33.2] Klone nach '/home/alex/Projekte/rust-esp32-can/.embuild/espressif/esp-idf/v5.1/components/spiffs/spiffs'...
[esp-idf-sys 0.33.2] Klone nach '/home/alex/Projekte/rust-esp32-can/.embuild/espressif/esp-idf/v5.1/components/unity/unity'...
    Building [=======================> ] 143/147: esp-idf-sys(build)
```