esp-rs/esp-idf-sys #29: Document all the environment variables that configure the PIO & Native builds

ivmarkov, 2021-11-05:
Some of these are available [here](https://github.com/ivmarkov/esp-idf-template/blob/master/cargo/.cargo/config.toml), with some comments, but the list is not exhaustive. We need to document these in the `README.md` page of `esp-idf-sys`.

N3xed, 2021-11-05:
For the native build, I've already documented all environment variables that the build script uses. Please tell me if I'm wrong.

ivmarkov, 2021-11-05:
Yes absolutely. I meant putting those in Readme.md. I'm not sure folks are ever looking into our build code infrastructure. Don't worry, I'll try to put these in Readme.md.

N3xed, 2021-11-05:
I mean they already are in the readme, the last section at the bottom 🤔

ivmarkov, 2021-11-10:
Cleaned up the README.md a bit with https://github.com/esp-rs/esp-idf-sys/commit/46128c86d1419cffbb8e25a02be1f5e9c39a6256

closed 2021-11-10 by ivmarkov