esp-rs/esp-idf-sys #30: Native build does not work with ESP-IDF branches named `release/vX.X`

ivmarkov, 2021-11-05:
1. Generate a Cargo ESP-IDF project using the [esp-idf-template](https://github.com/ivmarkov/esp-idf-template):
```sh
cargo generate --vcs none --git https://github.com/ivmarkov/esp-idf-template cargo
```

2. When asked to select `ESP-IDF native build version (stable = 4.3, upcoming = 4.4, master = 5.0; applicable only with `cargo build --features native`)`, choose `upcoming`

3. Enter the generated project:
```sh
cd <your-project>
```

4. Build:
```sh
cargo build --features native
```

Error:
```
  --- stderr
  Cloning into '/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/esp-idf-release/v4.4'...
Updating files: 100% (9463/9463), done.
  Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
  Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
<SNIP>
  Cloning into '/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/esp-idf-release/v4.4/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'...
  Submodule 'vendor/c_exception' (https://github.com/throwtheswitch/cexception.git) registered for path 'components/cmock/CMock/vendor/c_exception'
  Submodule 'vendor/unity' (https://github.com/throwtheswitch/unity.git) registered for path 'components/cmock/CMock/vendor/unity'
  Cloning into '/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/esp-idf-release/v4.4/components/cmock/CMock/vendor/c_exception'...
  Cloning into '/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/esp-idf-release/v4.4/components/cmock/CMock/vendor/unity'...
  Submodule 'ext/tinydtls' (https://github.com/eclipse/tinydtls.git) registered for path 'components/coap/libcoap/ext/tinydtls'
  Cloning into '/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/esp-idf-release/v4.4/components/coap/libcoap/ext/tinydtls'...
  Submodule 'third-party/mruby' (https://github.com/mruby/mruby) registered for path 'components/nghttp/nghttp2/third-party/mruby'
  Submodule 'third-party/neverbleed' (https://github.com/tatsuhiro-t/neverbleed.git) registered for path 'components/nghttp/nghttp2/third-party/neverbleed'
  Cloning into '/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/esp-idf-release/v4.4/components/nghttp/nghttp2/third-party/mruby'...
  Cloning into '/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/esp-idf-release/v4.4/components/nghttp/nghttp2/third-party/neverbleed'...
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  WARNING: You are using pip version 20.3.3; however, version 21.3.1 is available.
  You should consider upgrading via the '/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/python_env/idf4.4_py3.9_env/bin/python -m pip install --upgrade pip' command.
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  fatal: ref refs/remotes/origin/HEAD is not a symbolic ref
  Error: Command '"git" "--git-dir" "/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/esp-idf-release/v4.4/.git" "--work-tree" "/Users/imarkov/projects/esp32/ttt/error/.embuild/espressif/esp-idf-release/v4.4" "symbolic-ref" "refs/remotes/origin/HEAD"' failed with exit code Some(128)

  Caused by:
      fatal: ref refs/remotes/origin/HEAD is not a symbolic ref
```

ivmarkov, 2021-11-05:
@N3xed Any ideas?

N3xed, 2021-11-05:
Yes, the issue was with the `embuild::git::Repository::get_default_branch_of` function which didn't work when the repository only contained a single cloned branch (this function also already had a "FIXME" because I've noticed that it didn't work sometimes). I've created a fix for this.

I should really rewrite the `embuild::git` module to use the `git2` crate instead of querying git using the CLI.

N3xed, 2021-11-05:
Also, I think its better if you clone the tagged release instead of the branch. First, because `esp-idf-sys` compares the checked out tag with the current supported stable version and generates a warning on mismatch, and second because I think that the branch is in development for the minor versions and could change.

ivmarkov, 2021-11-05:
Is there a tagged version for the not-yet-released 4.4?

N3xed, 2021-11-05:
Oh no, my mistake, I thought you used the branch `release/v4.3` for the stable version too. Never mind what I said.

> Is there a tagged version for the not-yet-released 4.4?

Nope, I meant generally (which for stable is already like this).

N3xed, 2021-11-07:
Should be fixed by ivmarkov/embuild#23.
@ivmarkov Can you confirm this?

ivmarkov, 2021-11-07:
Confirmed, works.

Folks need to do `cargo update` in the binary crate root to get the latest dependencies and then clean and rebuild.

closed 2021-11-07 by ivmarkov