esp-rs/esp-idf-sys #31: Building with latest commit to fix ESP32-S3 DAC error results in failed build

Verequies, 2021-11-06:
Building rust-esp32-std-demo for the ESP32-S3 fails to build.
On commit "b28d5454adc620b57b7021291056eca60315d3ed", the latest at time of writing I get the following error:

```
error[E0412]: cannot find type `dac_channel_t` in this scope
     --> /home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-hal-0.23.2/src/gpio.rs:75:25
      |
75    |     fn dac_channel() -> dac_channel_t;
      |                         ^^^^^^^^^^^^^ help: a type alias with a similar name exists: `adc_channel_t`
      |
     ::: /home/user/rust-esp32-std-demo/target/xtensa-esp32s3-espidf/release/build/esp-idf-sys-d57948dc3e50df5e/out/bindings.rs:28769:1
      |
28769 | pub type adc_channel_t = c_types::c_uint;
      | ----------------------------------------- similarly named type alias `adc_channel_t` defined here

For more information about this error, try `rustc --explain E0412`.
error: could not compile `esp-idf-hal` due to previous error
warning: build failed, waiting for other jobs to finish...
error: build failed
```

On the previous commit "090311018dbbb0edc2efc8578c089edf4d59b10d", I get the following error:
```
  /home/user/.espressif/esp-idf-master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:627:20: warning: section does not match previous declaration [-Wsection]
  /home/user/.espressif/esp-idf-master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:356:20: note: previous attribute is here
  /home/user/.espressif/esp-idf-master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:646:26: warning: section does not match previous declaration [-Wsection]
  /home/user/.espressif/esp-idf-master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:398:26: note: previous attribute is here
  /home/user/.cargo/git/checkouts/esp-idf-sys-d4c2e811b42d99d6/0903110/src/include/esp-idf/bindings.h:65:10: fatal error: 'driver/dac.h' file not found
  /home/user/.espressif/esp-idf-master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:627:20: warning: section does not match previous declaration [-Wsection], err: false
  /home/user/.espressif/esp-idf-master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:646:26: warning: section does not match previous declaration [-Wsection], err: false
  /home/user/.cargo/git/checkouts/esp-idf-sys-d4c2e811b42d99d6/0903110/src/include/esp-idf/bindings.h:65:10: fatal error: 'driver/dac.h' file not found, err: true
```

I realise the latest commit is supposed to fix it, however I cannot build the latest. As a work around I have edited 'src/include/esp-idf/bindings.h', and modified line 65 to say '#include "driver/dac_common.h"' instead of the following as a hacky workaround.
https://github.com/esp-rs/esp-idf-sys/blob/9b49ade5fa985c5f0aa3a31b2f758117078c7fde/src/include/esp-idf/bindings.h#L65

From then on I am able to build and flash the project to my ESP32-S3 Dev Board successfully.

N3xed, 2021-11-07:
esp-rs/esp-idf-hal#12 will fix this.

ivmarkov, 2021-11-07:
Should work fine now, after `cargo update` and then a clean build.

closed 2021-11-07 by ivmarkov