esp-rs/esp-idf-sys #33: Unable to build latest ESP-IDF due to pthread_rwlock

Verequies, 2021-11-11:
It seems that ESP-IDF commit https://github.com/espressif/esp-idf/commit/b117bcd2a0a05b1dd334a5eb52a465ecfe6e01d1, has broken compilation on both ESP-IDF master and release/v4.4 branches, resulting in the below error:
```
  = note: Running ldproxy
          Error: Linker /home/user/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /home/user/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_init':
          /home/user/.espressif/esp-idf-master/components/pthread/pthread_rwlock.c:51: multiple definition of `pthread_rwlock_init'; /home/user/projects/rust-esp32-std-demo/target/xtensa-esp32s3-espidf/release/deps/libesp_idf_sys-033449d0383ac0dd.rlib(esp_idf_sys-033449d0383ac0dd.esp_idf_sys.208449cf-cgu.8.rcgu.o):/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.24.4/src/pthread_rwlock.rs:15: first defined here
          /home/user/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_destroy':
          /home/user/.espressif/esp-idf-master/components/pthread/pthread_rwlock.c:103: multiple definition of `pthread_rwlock_destroy'; /home/user/projects/rust-esp32-std-demo/target/xtensa-esp32s3-espidf/release/deps/libesp_idf_sys-033449d0383ac0dd.rlib(esp_idf_sys-033449d0383ac0dd.esp_idf_sys.208449cf-cgu.8.rcgu.o):/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.24.4/src/pthread_rwlock.rs:54: first defined here
          /home/user/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_rdlock':
          /home/user/.espressif/esp-idf-master/components/pthread/pthread_rwlock.c:162: multiple definition of `pthread_rwlock_rdlock'; /home/user/projects/rust-esp32-std-demo/target/xtensa-esp32s3-espidf/release/deps/libesp_idf_sys-033449d0383ac0dd.rlib(esp_idf_sys-033449d0383ac0dd.esp_idf_sys.208449cf-cgu.8.rcgu.o):/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.24.4/src/pthread_rwlock.rs:24: first defined here
          /home/user/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_wrlock':
          /home/user/.espressif/esp-idf-master/components/pthread/pthread_rwlock.c:195: multiple definition of `pthread_rwlock_wrlock'; /home/user/projects/rust-esp32-std-demo/target/xtensa-esp32s3-espidf/release/deps/libesp_idf_sys-033449d0383ac0dd.rlib(esp_idf_sys-033449d0383ac0dd.esp_idf_sys.208449cf-cgu.8.rcgu.o):/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.24.4/src/pthread_rwlock.rs:36: first defined here
          /home/user/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/pthread/libpthread.a(pthread_rwlock.c.obj): in function `pthread_rwlock_unlock':
          /home/user/.espressif/esp-idf-master/components/pthread/pthread_rwlock.c:223: multiple definition of `pthread_rwlock_unlock'; /home/user/projects/rust-esp32-std-demo/target/xtensa-esp32s3-espidf/release/deps/libesp_idf_sys-033449d0383ac0dd.rlib(esp_idf_sys-033449d0383ac0dd.esp_idf_sys.208449cf-cgu.8.rcgu.o):/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.24.4/src/pthread_rwlock.rs:48: first defined here
          collect2: error: ld returned 1 exit status

error: could not compile `rust-esp32-std-demo` due to previous error
```

Reverting to the previous before the above commit, ie https://github.com/espressif/esp-idf/commit/cd97e95a125c00dd41375d990e99db0a0618a1ab on the release/v4.4 branch, results in a successful build.

closed 2021-11-11 by MabezDev

Verequies, 2021-11-11:
@MabezDev Thanks! Can confirm this fixes the issue.
Had to specify the commit in Cargo.toml though as the version number/crate hasn't been updated.

MabezDev, 2021-11-12:
@ivmarkov mind cutting a release now we have the rwlock impl for 4.4?

MabezDev, 2021-11-12:
@ivmarkov very nice! I hadn't noticed that.

@Verequies `0.24.5` should be released as soon as the github action completes.

Verequies, 2021-11-13:
@MabezDev yep, using `0.24.5` now.

stevefan1999-personal, 2023-08-15:
Maybe we should consider reopening this for https://github.com/esp-rs/esp-idf-hal/issues/289

ivmarkov, 2023-08-15:
> Maybe we should consider reopening this for [esp-rs/esp-idf-hal#289](https://github.com/esp-rs/esp-idf-hal/issues/289)

Rather not. These two issues seem completely unrelated.

PR #28 (2021-11-11): feature: gate mutex rwlock patches for v4.4+
An official rwlock impl has been implemented in esp-idf (4.4 backport
imminent), hence for versions >4.3 we no longer need to link the patched
versions.

The current pthread rwlock implementation does not support attributes, is this a problem for Rust? I did a quick grep of the standard library and didn't spot any usage...

Closes #33