esp-rs/esp-idf-sys #34: Error: 'git remote show' yielded invalid output

imxood, 2021-11-13:
native build failed:

[esp-idf-sys 0.24.5] PATH=/home/maxu/.espressif/tools/cmake/3.20.3/bin:/home/maxu/.espressif/python_env/idf5.0_py3.7_env/bin:/home/maxu/.espressif/esp-idf-master/tools:$PATH
[esp-idf-sys 0.24.5] Error: 'git remote show' yielded invalid output
error: failed to run custom build command for `esp-idf-sys v0.24.5`

N3xed, 2021-11-13:
Please post the complete build output by running `cargo build -vvv`.

This should contain the output of `git remote show` which `embuild::git` somehow doesn't parse correctly.

imxood, 2021-11-14:
@N3xed the test code: https://github.com/imxood/test-esp32c3-native 

[the log file](https://github.com/imxood/test-esp32c3-native/blob/master/build.log) is build output.

use esp-idf-template to create basic code,

change Cargo.toml:
```toml
# [features]
# native = ["esp-idf-sys/native"]

[dependencies]
esp-idf-sys = { version = "0.24.6", features = ["binstart", "std", "native"], default-feature = false }
```

N3xed, 2021-11-14:
Well, unfortunately, I can't reproduce your issue.
Could you run `git remote show` and `git remote show origin` in `.embuild/espressif/esp-idf-master` and post your output?

Additionally, you could try to update git, make a full clean and try again (assuming your git is not up to date).

derchr, 2021-11-15:
Hi, I am also affected by this very issue since a few days.
Here are my logs:
```
warning: unused config key `unstable.configurable-env` in `/home/user/Schreibtisch/neues-projekt/.cargo/config.toml`
warning: unused config key `unstable.extra-link-arg` in `/home/user/Schreibtisch/neues-projekt/.cargo/config.toml`
       Fresh core v0.0.0 (/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
       Fresh cc v1.0.69
       Fresh unicode-xid v0.2.2
       Fresh cc v1.0.72
       Fresh fnv v1.0.7
       Fresh once_cell v1.8.0
       Fresh untrusted v0.7.1
       Fresh spin v0.5.2
       Fresh regex-syntax v0.6.25
       Fresh glob v0.3.0
       Fresh version_check v0.9.3
       Fresh autocfg v1.0.1
       Fresh lazy_static v1.4.0
       Fresh adler v1.0.2
       Fresh bitflags v1.3.2
       Fresh ppv-lite86 v0.2.15
       Fresh same-file v1.0.6
       Fresh unicode-width v0.1.9
       Fresh gimli v0.26.1
       Fresh strsim v0.8.0
       Fresh vec_map v0.8.2
       Fresh ansi_term v0.11.0
       Fresh humantime v2.1.0
       Fresh termcolor v1.1.2
       Fresh rustc-hash v1.1.0
       Fresh peeking_take_while v0.1.2
       Fresh base64 v0.13.0
       Fresh shlex v0.1.1
       Fresh lazycell v1.3.0
       Fresh strsim v0.10.0
       Fresh unicode-segmentation v1.8.0
       Fresh chunked_transfer v1.4.0
       Fresh remove_dir_all v0.5.3
       Fresh zero v0.1.2
       Fresh ident_case v1.0.1
       Fresh shlex v1.1.0
       Fresh paste v1.0.6
       Fresh rustc-std-workspace-core v1.99.0 (/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/rustc-std-workspace-core)
       Fresh libloading v0.7.1
       Fresh thread_local v1.1.3
       Fresh cmake v0.1.46
       Fresh walkdir v2.3.2
       Fresh addr2line v0.17.0
       Fresh textwrap v0.11.0
       Fresh xmas-elf v0.8.0
       Fresh heck v0.3.3
       Fresh compiler_builtins v0.1.49
       Fresh libc v0.2.107
       Fresh libc v0.2.105
       Fresh proc-macro2 v1.0.32
       Fresh memchr v2.4.1
       Fresh crossbeam-utils v0.8.5
       Fresh alloc v0.0.0 (/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/library/alloc)
       Fresh cfg-if v0.1.10
       Fresh rustc-demangle v0.1.21
       Fresh quote v1.0.10
       Fresh getrandom v0.2.3
       Fresh atty v0.2.14
       Fresh dirs-sys v0.3.6
       Fresh which v3.1.1
       Fresh remove_dir_all v0.7.0
       Fresh which v4.2.2
       Fresh aho-corasick v0.7.18
       Fresh ring v0.16.20
       Fresh bstr v0.2.17
[... 117 lines cut ...]
  fatal: Keine Namen gefunden, kann nichts beschreiben.
  Error: 'git remote show' yielded invalid output

```
This is the output of `git remote show` and `git remote show origin`:
```
git remote -v 
origin  https://github.com/espressif/esp-idf.git (fetch)
origin  https://github.com/espressif/esp-idf.git (push)

git remote show origin
* remote origin
  Fetch URL: https://github.com/espressif/esp-idf.git
  Push  URL: https://github.com/espressif/esp-idf.git
  HEAD branch: master
  Remote branch:
    master tracked
  Local branch configured for 'git pull':
    master merges with remote master
  Local ref configured for 'git push':
    master pushes to master (up to date)
```

What is funny is that this issue does not occur if i run cargo like so: `LC_ALL=C cargo build`
I think there is an issue when the output of some tool (git?) is not in english.

N3xed, 2021-11-16:
> What is funny is that this issue does not occur if i run cargo like so: `LC_ALL=C cargo build`
> I think there is an issue when the output of some tool (git?) is not in english.

Good find. I haven't really considered that git could output `remote show origin` in a different language.

I've created a PR (ivmarkov/embuild#24) for this, could you check if this fixes the problem?

derchr, 2021-11-16:
Unfortunately, I can still reproduce the issue with the fix.

I added
```
[build-dependencies]
embuild = {git="https://github.com/ivmarkov/embuild", branch = "git-lang-fix"}
```
to the build dependencies and did a clean build (removed ./target and ./.embuild directories). Did I forgot something?

N3xed, 2021-11-16:
Yup, you have to use [cargo patch](https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html#the-patch-section):

```toml
[patch.crates-io]
embuild = {git="https://github.com/ivmarkov/embuild", branch = "git-lang-fix"}
```

derchr, 2021-11-17:
Now I can confirm that it works.
Thank you so much!

closed 2021-11-17 by N3xed

N3xed, 2021-11-17:
The newest `embuild` version (0.25.3) now includes this fix.