esp-rs/esp-idf-sys #35: Error while building for ESP32C3

1c3t3a, 2021-11-17:
Hi there! When trying to build my project for the ESP32-C3 I get a build error for this dependency, the exact same code worked about a week ago:

```
error: failed to run custom build command for `esp-idf-sys v0.25.0`

Caused by:
  process didn't exit successfully: `/../../target/debug/build/esp-idf-sys-e1a1dd77045a2396/build-script-build` (exit status: 1)
  --- stdout
  Found compatible PlatformIO Core 5.2.3 -> /../.platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/ping_setsockopt_fix.diff
  cargo:rerun-if-changed=/../.././sdkconfig.debug
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32-c3-devkitm-1; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
  PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32-C3-DevKitM-1
  HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
  DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
  PACKAGES: 
   - framework-espidf 3.40300.0 (4.3.0) 
   - tool-cmake 3.16.4 
   - tool-esptoolpy 1.30100.210531 (3.1.0) 
   - tool-ninja 1.9.0 
   - toolchain-riscv-esp 1.80400.0 (8.4.0)
  Reading CMake configuration...

  --- stderr
  -- Found Git: /usr/bin/git (found version "2.30.1 (Apple Git-130)") 
  -- The C compiler identification is GNU 8.4.0
  -- The CXX compiler identification is GNU 8.4.0
  -- The ASM compiler identification is GNU
  -- Found assembler: /../.platformio/packages/toolchain-riscv-esp/bin/riscv32-esp-elf-gcc
  -- Check for working C compiler: /../.platformio/packages/toolchain-riscv-esp/bin/riscv32-esp-elf-gcc
  -- Check for working C compiler: /../.platformio/packages/toolchain-riscv-esp/bin/riscv32-esp-elf-gcc -- works
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Check for working CXX compiler: /../.platformio/packages/toolchain-riscv-esp/bin/riscv32-esp-elf-g++
  -- Check for working CXX compiler: /../.platformio/packages/toolchain-riscv-esp/bin/riscv32-esp-elf-g++ -- works
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
  -- Building ESP-IDF components for target esp32c3
  -- Project sdkconfig file /../../target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-fcab891b3db1348e/out/esp-idf/sdkconfig.debug
  -- Found PythonInterp: /../.platformio/penv/bin/python (found version "3.8.11") 
  -- Found Perl: /usr/bin/perl (found version "5.30.3") 
  -- App "esp-idf" version: 1
  -- Adding linker script /../.platformio/packages/framework-espidf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
  -- Adding linker script /../.platformio/packages/framework-espidf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
  -- Adding linker script /../.platformio/packages/framework-espidf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
  -- Adding linker script /../.platformio/packages/framework-espidf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
  -- Adding linker script /../.platformio/packages/framework-espidf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
  -- Adding linker script /../.platformio/packages/framework-espidf/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
  -- Adding linker script /../../target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-fcab891b3db1348e/out/esp-idf/.pio/build/debug/esp-idf/esp32c3/esp32c3_out.ld
  -- Adding linker script /../.platformio/packages/framework-espidf/components/esp32c3/ld/esp32c3.project.ld.in
  -- Adding linker script /../.platformio/packages/framework-espidf/components/esp32c3/ld/esp32c3.peripherals.ld
  -- Components: __pio_env app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32c3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs src tcp_transport tcpip_adapter tinyusb unity vfs wear_levelling wifi_provisioning wpa_supplicant
  -- Component paths: /../../target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-fcab891b3db1348e/out/esp-idf/.pio/build/debug/__pio_env /../.platformio/packages/framework-espidf/components/app_trace /../.platformio/packages/framework-espidf/components/app_update /../.platformio/packages/framework-espidf/components/asio /../.platformio/packages/framework-espidf/components/bootloader /../.platformio/packages/framework-espidf/components/bootloader_support /../.platformio/packages/framework-espidf/components/bt /../.platformio/packages/framework-espidf/components/cbor /../.platformio/packages/framework-espidf/components/cmock /../.platformio/packages/framework-espidf/components/coap /../.platformio/packages/framework-espidf/components/console /../.platformio/packages/framework-espidf/components/cxx /../.platformio/packages/framework-espidf/components/driver /../.platformio/packages/framework-espidf/components/efuse /../.platformio/packages/framework-espidf/components/esp-tls /../.platformio/packages/framework-espidf/components/esp32c3 /../.platformio/packages/framework-espidf/components/esp_adc_cal /../.platformio/packages/framework-espidf/components/esp_common /../.platformio/packages/framework-espidf/components/esp_eth /../.platformio/packages/framework-espidf/components/esp_event /../.platformio/packages/framework-espidf/components/esp_gdbstub /../.platformio/packages/framework-espidf/components/esp_hid /../.platformio/packages/framework-espidf/components/esp_http_client /../.platformio/packages/framework-espidf/components/esp_http_server /../.platformio/packages/framework-espidf/components/esp_https_ota /../.platformio/packages/framework-espidf/components/esp_https_server /../.platformio/packages/framework-espidf/components/esp_hw_support /../.platformio/packages/framework-espidf/components/esp_ipc /../.platformio/packages/framework-espidf/components/esp_local_ctrl /../.platformio/packages/framework-espidf/components/esp_netif /../.platformio/packages/framework-espidf/components/esp_pm /../.platformio/packages/framework-espidf/components/esp_ringbuf /../.platformio/packages/framework-espidf/components/esp_rom /../.platformio/packages/framework-espidf/components/esp_serial_slave_link /../.platformio/packages/framework-espidf/components/esp_system /../.platformio/packages/framework-espidf/components/esp_timer /../.platformio/packages/framework-espidf/components/esp_websocket_client /../.platformio/packages/framework-espidf/components/esp_wifi /../.platformio/packages/framework-espidf/components/espcoredump /../.platformio/packages/framework-espidf/components/esptool_py /../.platformio/packages/framework-espidf/components/expat /../.platformio/packages/framework-espidf/components/fatfs /../.platformio/packages/framework-espidf/components/freemodbus /../.platformio/packages/framework-espidf/components/freertos /../.platformio/packages/framework-espidf/components/hal /../.platformio/packages/framework-espidf/components/heap /../.platformio/packages/framework-espidf/components/idf_test /../.platformio/packages/framework-espidf/components/jsmn /../.platformio/packages/framework-espidf/components/json /../.platformio/packages/framework-espidf/components/libsodium /../.platformio/packages/framework-espidf/components/log /../.platformio/packages/framework-espidf/components/lwip /../.platformio/packages/framework-espidf/components/mbedtls /../.platformio/packages/framework-espidf/components/mdns /../.platformio/packages/framework-espidf/components/mqtt /.. [... cut ...]
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/panic.rs:133:14
    19: std::rt::lang_start_internal
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/rt.rs:128:20
    20: std::rt::lang_start
               at /rustc/c390d69a615f095208ac94841f3310268521b2ee/library/std/src/rt.rs:145:17
    21: <unknown>
               at ./build/build.rs:52:6
```

This errors with all versions of esp-idf-sys, the latest and the oldest I could find that satisfied the API I need.

N3xed, 2021-11-17:
Probably the same as #5.

1c3t3a, 2021-11-17:
Ah you're right! This weird! But thanks for solving this :)

closed 2021-11-17 by 1c3t3a