esp-rs/esp-idf-sys #40: Support MQTTs api of native esp-idf?

MaoniinoaM, 2021-12-07:
Hi, esp-rs team,
I'm implementing a project with ESP32C3 RUST, and I will apply the MQTTs APIs of native ESP-IDF. I check the code, it seems not support yet. I'm not sure if im right?
If yes, could you give me some advice how to implement MQTTs with esp-idf-sys?
Or there's no choice but to move back to C currently?

MabezDev, 2021-12-07:
Correct, it is not supported in this crate just yet.

There are a couple of options:

- Add support here (include the mqtt header files) and use the raw bindings to get something working.
  - Optionally also add a higher level, safe wrapper around the raw bindings. 
- Use a Rust MQTT client (note that at the TLS will not be supported just yet).

MaoniinoaM, 2021-12-08:
Sorry, I'm a newbee for RUST, so I'm not very clear about adding MQTTs support. You mean:
1.include esp mqtt header files in  esp-idf-sys/src/include/esp-idf/bindings.h 
2. build this esp-idf-sys crate locally, it will supports MQTTs directly.

Am I right?

ivmarkov, 2021-12-08:
* I'll add the esp mqtt header files in `esp-idf-sys
* As for implementing a Rust-safe wrapper around them, you have to wait for that (ETA: a few weeks), but here's the relevant issue: https://github.com/esp-rs/esp-idf-svc/issues/21

ivmarkov, 2021-12-08:
MQTT ESP-IDF APIs should be in the bindings now.

closed 2021-12-08 by ivmarkov