esp-rs/esp-idf-sys #44: Building in workspace, the .embuild folder is placed outside of the workspace dir

ryanc-me, 2021-12-26:
i'm trying to set up a Cargo workspace to build both Xtensa and RISC-V binaries for the ESP32-S2.

the setup is:
 * Workspace located at `/opt/esp32-workspace`
 * Binary crates at `/opt/esp32-workspace/main` and `/opt/esp32-workspace/ulp`

when I build from `/opt/esp32-workspace`, the embuild folder is created at `/opt/.embuild`, e.g. outside of the workspace.

for example building from:
 * `/opt/esp32-workspace`
 * `/opt/.embuild` is created

but when building from:
 * `/opt/esp32-workspace/main` (or any subdir)
 * `/opt/esp32-workspace/.embuild` is created

note also that this only happens when `ESP_IDF_TOOLS_INSTALL_DIR` is set to `workspace` or `custom:./.embuild`. the issue is fixed by using `out`, or an absolute path.

apologies if this is the wrong place to post this bug report!

N3xed, 2021-12-29:
Yeah, a bug like this was destined to pop out.

Currently, we get the workspace directory by popping 6 folders off of `esp-idf-sys`'s out dir in the hopes that this lands us outside of the `target` (i.e. the build) directory.

Could you verify why this doesn't work in your case?

ryanc-me, 2022-01-02:
So, I've done a bit more digging. It looks like `OUT_DIR` was sometimes set like this (note the missing target triple):
```bash
/opt/esp32s2-workspace/target/debug/build/esp-idf-sys-79af92772c653ed1/out/
```

However, I'm fairly sure that's only happening because of a bug with Cargo Workspaces, where the `target` set in `config.toml` is not respected (so it was trying to build for `x86_64-unknown-linux-gnu`, rather than `xtensa-esp32s2-espidf`). 

From my understanding, `OUT_DIR` would only be set like above if `$HOST == $TARGET`. I think it's pretty unlikely that someone's building _on_ an ESP32, right?

For what it's worth, I think there are maybe two options to fix the "problem":
 * Check if `$TARGET == $HOST` in [`workspace_dir()`](https://github.com/ivmarkov/embuild/blob/master/src/cargo.rs#L402), and if so, only `pop(5)`.
 * Loop while doing `pop(1)` until you land in a dir named `target`, whose parent contains a `Cargo.toml`, then jump up one more directory.

Are you accepting PRs for embuild? If so, I'd be happy to make one.

N3xed, 2022-01-02:
> However, I'm fairly sure that's only happening because of a bug with Cargo Workspaces, where the `target` set in `config.toml` is not respected (so it was trying to build for `x86_64-unknown-linux-gnu`, rather than `xtensa-esp32s2-espidf`).

That's strange.

> From my understanding, `OUT_DIR` would only be set like above if `$HOST == $TARGET`.

Ahh so it was like that, there's actually a `FIXME` in the code for this already. I'll fix it for the next release.

> I think it's pretty unlikely that someone's building _on_ an ESP32, right?

Haven't thought about that, but I guess `embuild` could be used for other targets where this could be the case. We were just unlucky with this bug I guess.

> Are you accepting PRs for embuild? If so, I'd be happy to make one.

Sure we do. But this is so minor, I'll fix it right now.

closed 2022-01-02 by N3xed