esp-rs/esp-idf-sys #45: ESP_IDF_SDKCONFIG_DEFAULTS & ESP_IDF_SDKCONFIG should have default values, so to say

ivmarkov, 2021-12-27:
Forcing the user to specify one of these two is maybe a bit too much? It is OK that we have these, but if these are not specified, I think `esp-idf-sys` should still look into the workspace root and fetch (& use) files which match the patterns:
* `sdkconfig.*`
* `sdkconfig.defaults.*`

ivmarkov, 2021-12-27:
@N3xed ^^^

N3xed, 2021-12-29:
Agreed.

I implemented this in <https://github.com/esp-rs/esp-idf-sys/commit/db657cd25112156dc5847a551bba63075a6756e1>.
They now have a default name (`sdkconfig` for `ESP_IDF_SDKCONFIG` and `sdkconfig.defaults` for `ESP_IDF_SDKCONFIG_DEFAULTS`). I've also added a similar search behavior to defaults files that already existed for the normal sdkconfig and documented them in the README.

N3xed, 2021-12-29:
This should also make the [`defaults.<chip>` filter logic](https://github.com/esp-rs/esp-idf-sys/commit/d31bb53df4e58c1122be2212119a217c368200d0#diff-41d39a20c88287b3858622367bf4dca53863d1145fa87e97f3e4093c386ce1beR190) obsolete.

N3xed, 2021-12-30:
@ivmarkov Is this okay for you?
And if it is, do you think this warrants a new minor version or can we just release a new patch version?

ivmarkov, 2021-12-31:
@N3xed Sorry - I'll look into it tomorrow on the 2nd of Jan. I know this and a few other things are pending, but I'm looking into something else ATM (esp-idf-hal).