esp-rs/esp-idf-sys #47: Include esp_tls.h header

derchr, 2021-12-30:
Hello,
would it be possible to include the esp_tls.h header? I think this would be useful to be able to make https requests.

Best regards.

ivmarkov, 2022-01-10:
I guess you have not noticed [this](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L538)?

Anyway, I've included this header, even though as per ^^^ it is not necessary for HTTPS requests using ESP-IDF's HTTP client (wrapper): https://github.com/esp-rs/esp-idf-sys/issues/47

closed 2022-01-10 by ivmarkov

derchr, 2022-01-10:
> I guess you have not noticed [this](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L538)?
> 
> Anyway, I've included this header, even though as per ^^^ it is not necessary for HTTPS requests using ESP-IDF's HTTP client (wrapper): [85561be](https://github.com/esp-rs/esp-idf-sys/commit/85561be3784030ff03a4eb8676e48d2bfb2a4bb2)

Oh, no I haven't noticed it. That makes it easier, thanks!