esp-rs/esp-idf-sys #49: `esp-idf-sys` fails to build

999eagle, 2022-01-02:
I have a project using the Rust ESP toolchain and today compilation of `esp-idf-sys` failed without any apparent changes. I've created an entirely new project from scratch using https://github.com/esp-rs/esp-idf-template and even that fails to compile with the exact same error:

```
    Updating crates.io index
   Compiling compiler_builtins v0.1.49
   Compiling core v0.0.0 (/home/sophie/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.103
   Compiling libc v0.2.112
   Compiling cc v1.0.69
   Compiling proc-macro2 v1.0.36
   Compiling std v0.0.0 (/home/sophie/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling unicode-xid v0.2.2
   Compiling cfg-if v1.0.0
   Compiling syn v1.0.84
   Compiling memchr v2.4.1
   Compiling cc v1.0.72
   Compiling serde_derive v1.0.133
   Compiling fnv v1.0.7
   Compiling log v0.4.14
   Compiling once_cell v1.9.0
   Compiling autocfg v1.0.1
   Compiling regex-syntax v0.6.25
   Compiling untrusted v0.7.1
   Compiling spin v0.5.2
   Compiling serde v1.0.133
   Compiling adler v1.0.2
   Compiling version_check v0.9.4
   Compiling glob v0.3.0
   Compiling tinyvec_macros v0.1.0
   Compiling matches v0.1.9
   Compiling lazy_static v1.4.0
   Compiling crossbeam-utils v0.8.5
   Compiling crc32fast v1.3.0
   Compiling gimli v0.26.1
   Compiling minimal-lexical v0.2.1
   Compiling bitflags v1.3.2
   Compiling unicode-bidi v0.3.7
   Compiling same-file v1.0.6
   Compiling unicode-width v0.1.9
   Compiling percent-encoding v2.1.0
   Compiling ppv-lite86 v0.2.16
   Compiling strsim v0.8.0
   Compiling rustc-demangle v0.1.21
   Compiling serde_json v1.0.74
   Compiling humantime v2.1.0
   Compiling ansi_term v0.12.1
   Compiling termcolor v1.1.2
   Compiling vec_map v0.8.2
   Compiling anyhow v1.0.52
   Compiling bindgen v0.59.2
   Compiling either v1.6.1
   Compiling remove_dir_all v0.5.3
   Compiling ryu v1.0.9
   Compiling rustc-hash v1.1.0
   Compiling strsim v0.10.0
   Compiling peeking_take_while v0.1.2
   Compiling unicode-segmentation v1.8.0
   Compiling zero v0.1.2
   Compiling ident_case v1.0.1
   Compiling chunked_transfer v1.4.0
   Compiling shlex v1.1.0
   Compiling lazycell v1.3.0
   Compiling base64 v0.13.0
   Compiling itoa v1.0.1
   Compiling async-trait v0.1.52
   Compiling paste v1.0.6
   Compiling libloading v0.7.2
   Compiling tinyvec v1.5.1
   Compiling thread_local v1.1.3
   Compiling walkdir v2.3.2
[... 147 lines cut ...]

  --- stderr
  Error: Compatible PlatformIO Core not found.
  Reason: PlatformIO executable not found in `/home/sophie/projects/other/esp-test/.embuild/platformio/penv/bin`
  Error: invalid type: null, expected a string at line 1 column 9390
```

Every subsequent build (without cleaning the `.embuild` directory) fails with:

```
   Compiling esp-idf-sys v0.28.3
error: failed to run custom build command for `esp-idf-sys v0.28.3`

Caused by:
  process didn't exit successfully: `/home/sophie/projects/other/esp-test/target/debug/build/esp-idf-sys-e4ce039c35a62e6f/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  Found compatible PlatformIO Core 5.2.4 -> /home/sophie/projects/other/esp-test/.embuild/platformio/penv/bin/platformio

  --- stderr
  Error: invalid type: null, expected a string at line 1 column 9390
```

999eagle, 2022-01-02:
Update: I've tried to use `cargo-pio`. Running `cargo pio -v upgrade` in the bare bones template project results in the same error:

```
Using built-in get-platformio.py
Running command "python3" "/tmp/.tmp4gkLlO" "check" "core" "--dump-state" "/tmp/.tmpgxvBZ2"
Found compatible PlatformIO Core 5.2.4 -> /home/sophie/.platformio/penv/bin/platformio
Resolving Resolver { pio: Pio { is_develop_core: false, platformio_exe: "/home/sophie/.platformio/penv/bin/platformio", penv_dir: "/home/sophie/.platformio/penv", installer_version: "1.0.2", python_version: "3.10.1", core_version: "5.2.4", system: "linux_x86_64", python_exe: "/home/sophie/.platformio/penv/bin/python", cache_dir: "/home/sophie/.platformio/.cache", penv_bin_dir: "/home/sophie/.platformio/penv/bin", core_dir: "/home/sophie/.platformio", log_level: Verbose }, params: ResolutionParams { board: None, mcu: None, platform: None, frameworks: [], target: None } }
Running PlatformIO command "/home/sophie/.platformio/penv/bin/platformio" "platform" "frameworks" "--json-output"
Error: invalid type: null, expected a string at line 1 column 9390
```

Manually running `platformio platform frameworks --json-output` results in a huge JSON file, here's the relevant snippet:

`{"name": "tirtos", "title": "TI-RTOS", "description": "Stripped Energia framework without compiled core", "url": null, "homepage": "https://platformio.org/frameworks/tirtos", "platforms": ["timsp432"]}`

Apparently `cargo-pio` chokes on the `url` field being `null` for an entirely unrelated platform.

mihai-dinculescu, 2022-01-02:
@999eagle your fix works for me, thanks!

Until it is merged into embuild and released, I'm using it via
```
[patch.crates-io]
embuild = { git="https://github.com/999eagle/embuild" }
```

N3xed, 2022-01-02:
Thanks for opening this issue.
I'll keep this open till a new `embuild` version is released.

ivmarkov, 2022-01-02:
We probably want to release a new embuild asap, as all CI builds currently fail because of this issue?

N3xed, 2022-01-02:
Fixed with new `embuild` version `0.27.2`.

closed 2022-01-02 by N3xed