esp-rs/esp-idf-sys #61: Build script failure when building esp-idf-sys as dependency of previously working project

elidupree, 2022-01-29:
I have [a project](https://github.com/elidupree/eliduprees-emg-mouse/tree/main/emg-server) based on [rust-esp32-std-demo](https://github.com/ivmarkov/rust-esp32-std-demo/) that I've been successfully building for my ESP32 in the past, but today, when I said `cargo build`, it decided to recompile `esp-idf-sys` and the `esp-idf-sys` build failed, without me changing anything except for my own code. (I also get the same (similar?) error after switching to the latest `esp-idf-sys` version.)

I don't know how to tell if this is an `esp-idf-sys` issue, or an `embuild` issue, or something else, so I'm reporting it in both places; feel free to close this if it's in the wrong place.

I'm on Debian, and I installed esp-rs using [rust-build](https://github.com/esp-rs/rust-build) as of Jan 11 (rust-build commit 6b71e690c35778c7c62ba5a229542cdfae01d6c4).

Console output:
```
/n/autobuild/eliduprees-emg-mouse/build/emg-server$ cargo build
   Compiling bindgen v0.59.2
   Compiling embuild v0.27.2
   Compiling esp-idf-sys v0.29.1
   Compiling esp-idf-hal v0.32.2
   Compiling esp-idf-svc v0.36.1
   Compiling emg-server v0.1.0 (/n/autobuild/eliduprees-emg-mouse/build/emg-server)
error: failed to run custom build command for `esp-idf-sys v0.29.1`

Caused by:
  process didn't exit successfully: `/n/autobuild/eliduprees-emg-mouse/build/emg-server/target/debug/build/esp-idf-sys-dce598b034e4c44c/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  Found compatible PlatformIO Core 5.2.4 -> /n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/ping_setsockopt_fix.diff
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/missing_riscv_atomics_fix.diff
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32cam; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
  PLATFORM: Espressif 32 (3.4.0) > AI Thinker ESP32-CAM
  HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
  DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
  PACKAGES: 
   - framework-espidf 3.40301.0 (4.3.1) 
   - tool-cmake 3.16.4 
   - tool-esptoolpy 1.30100.210531 (3.1.0) 
   - tool-ninja 1.7.1 
   - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
   - toolchain-riscv32-esp 8.4.0+2021r1 
   - toolchain-xtensa-esp32 8.4.0+2021r1 
   - toolchain-xtensa-esp32s2 8.4.0+2021r1
  Reading CMake configuration...
  LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
  LDF Modes: Finder ~ chain, Compatibility ~ soft
  Found 0 compatible libraries
  Scanning dependencies...
  No dependencies
  Building in debug mode
  Retrieving maximum program size .pio/build/debug/firmware.elf
  Checking size .pio/build/debug/firmware.elf
  Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
  RAM:   [          ]   3.9% (used 12896 bytes from 327680 bytes)
  Flash: [==        ]  15.7% (used 164497 bytes from 1048576 bytes)
  ========================= [SUCCESS] Took 12.18 seconds =========================

  Environment    Status    Duration
  -------------  --------  ------------
  debug          SUCCESS   00:00:12.181
  ========================= 1 succeeded in 00:00:12.181 =========================

  ********************************************************************************
  There are the new updates for platforms (espressif32)
  Please update them via `platformio platform update` command.

  If you want to manually check for the new versions without updating, please use `platformio platform update --dry-run` command.
  ********************************************************************************

  cargo:rerun-if-changed=/n/HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.29.1/src/include/esp-idf/bindings.h

  --- stderr
  Output: "/n/autobuild/eliduprees-emg-mouse/build/emg-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-94df6151a6a52853/out/bindings.rs"
  Bindgen builder flags: ["/n/HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.29.1/src/include/esp-idf/bindings.h", "--rust-target", "1.47", "--blocklist-function", "strtold", "--blocklist-function", "_strtold_r", "--blocklist-function", "v.*printf", "--blocklist-function", "v.*scanf", "--blocklist-function", "_v.*printf_r", "--blocklist-function", "_v.*scanf_r", "--blocklist-function", "esp_log_writev", "--no-layout-tests", "--with-derive-default", "--ctypes-prefix", "c_types", "--generate", "functions,types,vars,methods,constructors,destructors", "--use-core", "--no-rustfmt-bindings", "--", "-D__bindgen", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/esp_pm/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-94df6151a6a52853/out/esp-idf/.pio/build/debug/config", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/newlib/platform_include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/freertos/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/freertos/port/xtensa/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/esp_hw_support/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/esp_hw_support/port/esp32/.", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/heap/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/log/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/lwip/include/apps", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/lwip/include/apps/sntp", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/lwip/lwip/src/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/lwip/port/esp32/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/lwip/port/esp32/include/arch", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/soc/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/soc/esp32/.", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/soc/esp32/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/hal/esp32/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/hal/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/esp_rom/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/esp_rom/esp32", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/esp_rom/include/esp32", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/esp_common/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/components/esp_system/include", "-I/n/autobuild/eliduprees-emg-mouse/build/emg-server/.embuild/platformio/packages/framework-espidf/c [... cut ...] ENABLED", "-target", "xtensa"]
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /n/HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.59.2/src/ir/context.rs:538:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

ivmarkov, 2022-01-30:
There are lots of changes recently and we are moving a bit too fast and sometimes breaking things a bit. For which I apologize.
My suggestion would be:
* Delete the `.embuild` folder from your binary crate completely.
* Manually update all `embuild`, `esp-idf-sys`, `embedded-svc` and `esp-idf-svc` dependencies to the latest ones you can see here: https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/Cargo.toml

elidupree, 2022-01-30:
Thanks for the reply! Unfortunately, I've already tried deleting `.embuild`. Also, I should've mentioned - I get the same error when building the unmodified, latest version of `rust-esp32-std-demo` (both when I posted, and now, when I've done a `git pull` and retried it). Here's the console output of my attempt to rebuild `rust-esp32-std-demo` just now:

```
/n/esp-rust/rust-build/rust-esp32-std-demo$ cargo build            
    Updating crates.io index
  Downloaded embedded-svc v0.16.5
  Downloaded esp-idf-svc v0.36.6
  Downloaded 2 crates (88.0 KB) in 0.40s
   Compiling uncased v0.9.6
   Compiling esp-idf-hal v0.32.5
   Compiling esp-idf-svc v0.36.6
   Compiling rust-esp32-std-demo v0.22.7 (/n/esp-rust/rust-build/rust-esp32-std-demo)
   Compiling esp-idf-sys v0.30.3
   Compiling embedded-svc v0.16.5
error: failed to run custom build command for `esp-idf-sys v0.30.3`

Caused by:
  process didn't exit successfully: `/n/esp-rust/rust-build/rust-esp32-std-demo/target/debug/build/esp-idf-sys-1b6c7e608263dc0e/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  Found compatible PlatformIO Core 5.2.4 -> /n/esp-rust/rust-build/rust-esp32-std-demo/.embuild/platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/missing_riscv_atomics_fix.diff
  cargo:rerun-if-changed=/n/esp-rust/rust-build/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/n/esp-rust/rust-build/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/n/esp-rust/rust-build/rust-esp32-std-demo/sdkconfig.defaults.esp32s2
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32cam; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
  PLATFORM: Espressif 32 (3.4.0) > AI Thinker ESP32-CAM
  HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
  DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
  PACKAGES: 
   - framework-espidf 3.40301.0 (4.3.1) 
   - tool-cmake 3.16.4 
   - tool-esptoolpy 1.30100.210531 (3.1.0) 
   - tool-ninja 1.7.1 
   - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
   - toolchain-riscv32-esp 8.4.0+2021r1 
   - toolchain-xtensa-esp32 8.4.0+2021r1 
   - toolchain-xtensa-esp32s2 8.4.0+2021r1
  Reading CMake configuration...
  LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
  LDF Modes: Finder ~ chain, Compatibility ~ soft
  Found 0 compatible libraries
  Scanning dependencies...
  No dependencies
  Building in debug mode
  Compiling .pio/build/debug/app_update/esp_app_desc.o
  Archiving .pio/build/debug/esp-idf/app_update/libapp_update.a
  Indexing .pio/build/debug/esp-idf/app_update/libapp_update.a
  Generating project linker script .pio/build/debug/esp32.project.ld
  __action_dump([".pio/build/debug/firmware.elf"], [".pio/build/debug/src/main.o"])
  Linking .pio/build/debug/firmware.elf
  Retrieving maximum program size .pio/build/debug/firmware.elf
  Checking size .pio/build/debug/firmware.elf
  Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
  RAM:   [          ]   3.9% (used 12872 bytes from 327680 bytes)
  Flash: [==        ]  15.7% (used 164233 bytes from 1048576 bytes)
  Building .pio/build/debug/firmware.bin
  esptool.py v3.1
  Merged 2 ELF sections
  ========================= [SUCCESS] Took 16.72 seconds =========================

  Environment    Status    Duration
  -------------  --------  ------------
  debug          SUCCESS   00:00:16.722
  ========================= 1 succeeded in 00:00:16.722 =========================
  cargo:rerun-if-changed=/n/HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.30.3/src/include/esp-idf/bindings.h

  --- stderr
  Output: "/n/esp-rust/rust-build/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-85749cb3de2b25fe/out/bindings.rs"
  Bindgen builder flags: ["/n/HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.30.3/src/include/esp-idf/bindings.h", "--rust-target", "1.47", "--blocklist-function", "strtold", "--blocklist-function", "_strtold_r", "--blocklist-function", "v.*printf", "--blocklist-function", "v.*scanf", "--blocklist-function", "_v.*printf_r", "--blocklist-function", "_v.*scanf_r", "--blocklist-function", "esp_log_writev", "--no-layout-tests", "--with-derive-default", "--ctypes-prefix", "c_types", "--generate", "functions,types,vars,methods,constructors,destructors", "--use-core", "--no-rustfmt-bindings", "--", "-D__bindgen", "-I/n/esp-rust/rust-build/rust-esp32-std-demo/.embuild/platformio/packages/framework-espidf/components/esp_pm/include", "-I/n/esp-rust/rust-build/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-85749cb3de2b25fe/out/esp-idf/.pio/build/debug/config", "-I/n/esp-rust/rust-build/rust-esp32-std-demo/.embuild/platformio/packages/framework-espidf/components/newlib/platform_include", "-I/n/esp-rust/rust-build/rust-esp [... cut ...] ENABLED", "-target", "xtensa"]
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /n/HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.59.2/src/ir/context.rs:538:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

N3xed, 2022-01-30:
Were you always using `esp-idf` `v3.4`?
And what LLVM/libclang are you using?

You could try updating LLVM/libclang.

elidupree, 2022-01-30:
> You could try updating LLVM/libclang.

Oh dear. I just realized that the problem was that I forgot to set my LIBCLANG_PATH environment variable to point to the custom one installed by `esp-rust`. I've got my project working again now.

I wonder if that error message could be improved for this situation? Right now, it reads like "Error: There was an error" - if it had said something like "you used the wrong libclang" then I'd have known what the problem was immediately.

ivmarkov, 2022-01-30:
I would still suggest updating your dependencies, because PlatformIO is using a newer ESP-IDF now, and if you remove the `.embuild` directory of your project, it will likely stop compiling.

As for LLVM/CLang and warnings - feel free to open an issue for that. We can try doing something on that front, but it will be low priority. IMHO bindgen itself needs to improve its error reporting, rather than us working around it - in that particular case, it fails because the target we are passing is not supported by LLVM, so I would expect that it spits out an error message along the lines of "unrecognized target" or suchlike.

N3xed, 2022-01-30:
> Oh dear. I just realized that the problem was that I forgot to set my LIBCLANG_PATH environment variable to point to the custom one installed by `esp-rust`. I've got my project working again now.

Nice, good that it's working again.

> I wonder if that error message could be improved for this situation? Right now, it reads like "Error: There was an error" - if it had said something like "you used the wrong libclang" then I'd have known what the problem was immediately.

Yeah, like @ivmarkov already said, this is a `bindgen` issue. We could try to check for this specific case and then output an error message (feel free to make a PR for this), but it should really be improved in `bindgen` itself. But maybe that is also not so straightforward since they opted to just output a general error message.

With that said, I'll close the issue. You can always reopen if something else is amiss (or even create a new issue).

closed 2022-01-30 by N3xed

JeanBlignaut, 2022-01-31:
I'm having a similar issue but I'm a novice at this.
how where do I set this LIBCLANG_PATH ? Ive tried setting up this toolchain on my Mac natively and via docker container and so far have not been able to compile a simple hello world 😢 and been trying for a week 😅

JeanBlignaut, 2022-01-31:
`root@6c18b5fd0939:~/project# cargo build
   Compiling esp-idf-sys v0.30.1
error: failed to run custom build command for `esp-idf-sys v0.30.1`

Caused by:
  process didn't exit successfully: `/root/project/target/debug/build/esp-idf-sys-47b6e8f5fe49b8d5/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  Found compatible PlatformIO Core 5.2.4 -> /root/project/.embuild/platformio/penv/bin/platformio
  cargo:rerun-if-changed=./patches/pthread_destructor_fix.diff
  cargo:rerun-if-changed=./patches/missing_riscv_atomics_fix.diff
  cargo:rerun-if-changed=./patches/ping_setsockopt_fix.diff
  cargo:rerun-if-changed=./patches/missing_xtensa_atomics_fix.diff
  cargo:rerun-if-changed=/root/project/sdkconfig.defaults
  cargo:rerun-if-changed=/root/project/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/root/project/sdkconfig.defaults.esp32s2
  Warning! Ignore unknown configuration option `patches` in section [env]
  Processing debug (board: esp32cam; platform: espressif32; framework: espidf)
  --------------------------------------------------------------------------------
  Verbose mode can be enabled via `-v, --verbose` option
  os.chdir('/root/project/.embuild/platformio/packages/framework-espidf')
  git init
  Initialized empty Git repository in /root/project/.embuild/platformio/packages/framework-espidf/.git/
  os.chdir('/root/project/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b79db3d3ea192b49/out/esp-idf')
  os.chdir('/root/project/.embuild/platformio/packages/framework-espidf')
  git apply /root/project/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b79db3d3ea192b49/out/esp-idf/patches/ping_setsockopt_fix.diff
  os.chdir('/root/project/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b79db3d3ea192b49/out/esp-idf')

  --- stderr
  error: patch failed: components/lwip/apps/ping/ping_sock.c:274
  error: components/lwip/apps/ping/ping_sock.c: patch does not apply
  *** Error 1
  ========================== [FAILED] Took 1.00 seconds ==========================
  Environment    Status    Duration
  -------------  --------  ------------
  debug          FAILED    00:00:01.001
  ==================== 1 failed, 0 succeeded in 00:00:01.001 ====================
  Error: No such file or directory (os error 2)`

ivmarkov, 2022-01-31:
> I'm having a similar issue but I'm a novice at this. how where do I set this LIBCLANG_PATH ? Ive tried setting up this toolchain on my Mac natively and via docker container and so far have not been able to compile a simple hello world 😢 and been trying for a week 😅

Your issue is different. You are stuck with (a) newer PlatformIO, but (b) older `esp-idf-sys` and friends. Just re-clone `rust-esp32-std-demo` from clean state and rebuild, and the particular problem you are observing should be gone. if not, please reply here, and I'll look deeper into it.

JeanBlignaut, 2022-01-31:
Tried again on new Espressif/idf-rust-examples container:
```root@911e4a3c7cd1:/opt# cd /opt/rust-esp32-std-demo/
root@911e4a3c7cd1:/opt/rust-esp32-std-demo# cargo +esp build --release
warning: unused config key `unstable.extra-link-arg` in `/opt/rust-esp32-std-demo/.cargo/config.toml`
warning: unused config key `unstable.configurable-env` in `/opt/rust-esp32-std-demo/.cargo/config.toml`
   Compiling compiler_builtins v0.1.49
   Compiling core v0.0.0 (/opt/rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.103
   Compiling cc v1.0.69
   Compiling std v0.0.0 (/opt/rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling libc v0.2.109
   Compiling proc-macro2 v1.0.33
   Compiling memchr v2.4.1
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.82
   Compiling cfg-if v1.0.0
   Compiling cc v1.0.72
   Compiling log v0.4.14
   Compiling autocfg v1.0.1
   Compiling serde_derive v1.0.130
   Compiling serde v1.0.130
   Compiling once_cell v1.8.0
   Compiling fnv v1.0.7
   Compiling ryu v1.0.6
   Compiling spin v0.5.2
   Compiling untrusted v0.7.1
   Compiling regex-syntax v0.6.25
   Compiling version_check v0.9.3
   Compiling tinyvec_macros v0.1.0
   Compiling glob v0.3.0
   Compiling serde_json v1.0.72
   Compiling anyhow v1.0.51
   Compiling matches v0.1.9
   Compiling lazy_static v1.4.0
   Compiling crossbeam-utils v0.8.5
   Compiling unicode-bidi v0.3.7
   Compiling gimli v0.26.1
   Compiling same-file v1.0.6
   Compiling unicode-width v0.1.9
   Compiling percent-encoding v2.1.0
   Compiling ppv-lite86 v0.2.15
   Compiling adler v1.0.2
   Compiling bitflags v1.3.2
   Compiling strsim v0.8.0
   Compiling termcolor v1.1.2
   Compiling bindgen v0.57.0
   Compiling humantime v2.1.0
   Compiling rustc-demangle v0.1.21
   Compiling ansi_term v0.12.1
   Compiling vec_map v0.8.2
   Compiling shlex v0.1.1
   Compiling chunked_transfer v1.4.0
   Compiling remove_dir_all v0.5.3
   Compiling lazycell v1.3.0
   Compiling base64 v0.13.0
   Compiling either v1.6.1
   Compiling unicode-segmentation v1.8.0
   Compiling rustc-hash v1.1.0
   Compiling zero v0.1.2
   Compiling peeking_take_while v0.1.2
   Compiling itoa v0.4.8
   Compiling shlex v1.1.0
   Compiling futures-core v0.3.18
   Compiling strsim v0.10.0
   Compiling ident_case v1.0.1
   Compiling az v1.2.0
   Compiling async-trait v0.1.51
   Compiling heapless v0.7.8
   Compiling paste v1.0.6
   Compiling unwind v0.0.0 (/opt/rustup/toolchains/esp/lib/rustlib/src/rust/library/unwind)
[... 195 lines cut ...]
    Using cached sniffio-1.2.0-py3-none-any.whl (10 kB)
  Building wheels for collected packages: platformio
    Building wheel for platformio (setup.py): started
    Building wheel for platformio (setup.py): finished with status 'done'
    Created wheel for platformio: filename=platformio-5.2.4-py3-none-any.whl size=344394 sha256=ebbb129fce05bb3b761a74774eb69e70368ed51e76ee98d753e781241ae03e56
    Stored in directory: /root/.cache/pip/wheels/c7/5f/eb/2b681c5612c9a8ace53548f7febbd3edd52d6522ebfe2aafcc
  Successfully built platformio
  Installing collected packages: tabulate, pyserial, pyelftools, ifaddr, certifi, bottle, zeroconf, urllib3, sniffio, semantic_version, marshmallow, idna, h11, colorama, click, charset-normalizer, asgiref, ajsonrpc, aiofiles, wsproto, uvicorn, requests, anyio, starlette, platformio
  Successfully installed aiofiles-0.8.0 ajsonrpc-1.2.0 anyio-3.5.0 asgiref-3.5.0 bottle-0.12.19 certifi-2021.10.8 charset-normalizer-2.0.11 click-8.0.3 colorama-0.4.4 h11-0.13.0 idna-3.3 ifaddr-0.1.7 marshmallow-3.14.1 platformio-5.2.4 pyelftools-0.27 pyserial-3.5 requests-2.27.1 semantic_version-2.8.5 sniffio-1.2.0 starlette-0.17.1 tabulate-0.8.9 urllib3-1.26.8 uvicorn-0.16.0 wsproto-1.0.0 zeroconf-0.37.0

  PlatformIO Core has been successfully installed into an isolated environment `/root/.platformio/penv`!

  The full path to `platformio.exe` is `/root/.platformio/penv/bin/platformio`

  If you need an access to `platformio.exe` from other applications, please install Shell Commands
  (add PlatformIO Core binary directory `/root/.platformio/penv/bin` to the system environment PATH variable):

  See https://docs.platformio.org/page/installation.html#install-shell-commands

  Found compatible PlatformIO Core 5.2.4 -> /root/.platformio/penv/bin/platformio

  --- stderr
  Error: Compatible PlatformIO Core not found.
  Reason: PlatformIO Core was installed using another platform `Linux-5.11.0-1021-azure-x86_64-with-glibc2.29`. Your current platform: Linux-5.13.0-27-generic-x86_64-with-glibc2.29
  Error: invalid type: null, expected a string at line 1 column 9391```

ivmarkov, 2022-01-31:
```
   Compiling embedded-svc v0.15.3
   Compiling esp-idf-sys v0.27.0
   Compiling esp-idf-hal v0.27.0
   Compiling esp-idf-svc v0.32.3
   Compiling rust-esp32-std-demo v0.20.6 (/opt/rust-esp32-std-demo)
```

These dependencies, and the demo itself are old, and are **incompatible** with recent PlatformIO releases. Unfortunately, we do not have (yet) a way to "freeze" the PlatformIO dependency, so when it gets updated, you are in a situation like this.

Really, how about just forgetting about ready-to-use containers for a moment, and then proceeding as I suggested:
* Clone the `rust-esp32-std-demo` on your own PC
* Download and install the Rust & clang bundle from Espressif, as described [here](https://github.com/esp-rs/rust-build)
* Proceed with the build by following the instruction in the `rust-esp32-std-demo` README file

That should get you in a buildable shape.