esp-rs/esp-idf-sys #64: Cannot find bluedroid functions

magmast, 2022-02-01:
I've created a project with `esp-idf-template`, added 

```
CONFIG_BT_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=y
```

to `sdkconfig.defaults` and stared to follow official `esp-idf` tutorial.

Unfortunately after implementing part of some of the functions were missing. Currently I see 3 of them: `esp_bluedroid_init`, `esp_bluedroid_enable` and `esp_ble_gatts_register_callback`. But when I searched generated `bindings.rs` file for "bluedroid" it found nothing despite some mentions in the comments.

I searched for the c header files in the `embuild` folder. I don't remember exactly were I found them, but every file I looked at was exactly the same as those in the `esp-idf` repository and contained all the functions that `bindings.rs` was missing.

Here is the function I was writing:

```rs
unsafe fn unsafe_init() {
    esp_idf_sys::nvs_flash_init();
    esp_idf_sys::esp_bt_controller_init(&mut Default::default());
    esp_idf_sys::esp_bt_controller_enable(esp_idf_sys::esp_bt_mode_t_ESP_BT_MODE_BLE);
    esp_idf_sys::esp_bluedroid_init();
    esp_idf_sys::esp_bluedroid_enable();
    esp_idf_sys::esp_ble_gatts_register_callback();
}
```

ivmarkov, 2022-02-07:
The issue is likely caused by the fact that not all bluedropid header files are included when `esp-idf-sys` is being built. You can figure out which headers have to be included by looking at the ESP-IDF bluetooth C examples.

It is very easy to fix, we just need to add the missing include files [here](https://github.com/esp-rs/esp-idf-sys/blob/master/src/include/esp-idf/bindings.h#L176).

Now, I can try digging up the functions and their include files for you, but I think it would be much more efficient if you try to do this yourself:
* Clone `esp-idf-sys`, next to your binary crate (as a sibling directory)
* Make sure that you are using latest `esp-idf-sys`, `embedded-svc` and so on. Consult the `rust-esp32-std-demo` project for those
* In the `Cargo.toml` file of your binary crate, add this under section `[patch.crates-io]`:
  * `esp-idf-sys = { path = "../esp-idf-sys" }`

Now, whenever you find that you are missing a header, please include - as per above - [here](https://github.com/esp-rs/esp-idf-sys/blob/master/src/include/esp-idf/bindings.h#L176) and then do `cargo clean` + `cargo build`

Once you have a feeling that all bluedroid files are present in the bindings' header file, I would be very grateful if you contribute a PR that closes this issue.

magmast, 2022-02-08:
I tried to do what you said in your comment before creating this issue, but all the headers that should be needed were already included in the `bindings.h` file and that's why I created the issue.

I tried to do it once again today after reading your comment and it is working in a newly created project without any modifications in `esp-idf-sys`.

And now after doing everything again I see what was the problem.

```
CONFIG_BT_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=y
```

See the second line? It should be without `BT`...

```
CONFIG_BLUEDROID_ENABLED=y
```

So sorry for the false alarm. Thanks for your help and for your work on this project.

closed 2022-02-08 by magmast