esp-rs/esp-idf-sys #65: First 3 bytes of files read from FatFs will be truncated

derchr, 2022-02-02:
Hi,
I have created a repro for this bug: https://github.com/derchr/repro-fs
Just follow the instructions in the readme and the bug should show up.

Description of the bug:
What I noticed is that somehow the first 3 bytes of all files that are being read from the filesystem get truncated.
I'm almost certain that 1-2 weeks ago this bug wasn't there as I remember loading binary files (images) from the fs and this worked flawlessly. Now always 3 bytes are missing and the image does not get parsed.
I'm also not sure if this is even caused by esp-idf-sys.

My toolchain is the esp rust compiler version 1.57.0 targeting the default ESP32.

Thanks!

ivmarkov, 2022-02-07:
Likely related to this issue: https://github.com/esp-rs/rust/issues/96

derchr, 2022-02-09:
Yeah, it seems like this is the same bug. Closing it here in favor of the already existing bug report.

closed 2022-02-09 by derchr

bugadani, 2022-03-04:
@derchr `String::with_capacity` only allocates memory, but does not set buffer length, i.e. you can't get a `&mut [u8]` for the allocated memory. Does your example work if you use `vec![0; some_size]` instead? In my case, I'm able to read into 0 filled vectors.

derchr, 2022-03-04:
> @derchr `String::with_capacity` only allocates memory, but does not set buffer length, i.e. you can't get a `&mut [u8]` for the allocated memory. Does your example work if you use `vec![0; some_size]` instead? In my case, I'm able to read into 0 filled vectors.

I remember also testing `Vec::with_capacity` and this did not work. Pre-filling the Vec and writing to it using `&mut [u8]` might actually work as it's similar to the fixed-array method that did work for me.