esp-rs/esp-idf-sys #70: ESP-NOW support

bernii, 2022-02-14:
Hi!

I was wondering if there are any plans to support [ESP-NOW](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html) which is part of ESP-IDF too.

 I've built a couple of projects on it (on top of Arduino stack) and was very happy with the possibilities of having a p2p broadcast network that did not require an intermediate wifi router while having quite efficient communication between ESP devices.

georgik, 2022-02-14:
Hi @bernii . Thank you for the request. I've added the topic to the next community meeting to discuss it in more details: https://github.com/esp-rs/rust/discussions/105

Right now the support of networking features are little bit limited. The support is improving. HTTP and MQTT should work. Also TLS support was added lately.

bernii, 2022-02-15:
thanks @georgik!  absolutely, there's a ton of great work already - I'm already using HTTP(S) and it works great :) Adding esp-now would give an opportunity to have a mesh network between the devices as a fallback or when routers are not available. During my testing (with ardunio stack) I also found out that espnow seems to have a better throughput (vs using wifi router) when it comes to small packages (for near-real-time applications) - that's why I'm thinking it might be an addition worthwhile considering :)

ivmarkov, 2022-04-17:
Closing - all patches merged.

closed 2022-04-17 by ivmarkov