esp-rs/esp-idf-sys #71: printing Instant::now() blocks thread

andresv, 2022-02-15:
```rust
use esp_idf_sys as _; // If using the `binstart` feature of `esp-idf-sys`, always keep this module imported
use std::time::{Duration, Instant};

fn main() {
    // Temporary. Will disappear once ESP-IDF 4.4 is released, but for now it is necessary to call this function once,
    // or else some patches to the runtime implemented by esp-idf-sys might not link properly.
    esp_idf_sys::link_patches();

    use esp_idf_sys::*;
    if let Err(_e) = esp!(unsafe { esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(0)) }) {
        panic!("cannot stop twdt 0");
    }
    if let Err(_e) = esp!(unsafe { esp_task_wdt_delete(xTaskGetIdleTaskHandleForCPU(1)) }) {
        panic!("cannot stop twdt 1");
    }

    std::thread::Builder::new()
        .name("thread 0".to_string())
        .stack_size(16384)
        .spawn(move || {
            let mut t = Instant::now();
            loop {
                std::thread::sleep(Duration::from_millis(2));
                //unsafe { esp_idf_sys::vTaskDelay(1) };
                if Instant::now() - t > Duration::from_millis(1000) {
                    t = Instant::now();
                    //println!("{:?} thread 0", t);
                    println!("thread 0");
                }
            }
        })
        .unwrap();

    std::thread::Builder::new()
        .name("thread 1".to_string())
        .stack_size(16384)
        .spawn(move || {
            let mut t = Instant::now();
            loop {
                std::thread::sleep(Duration::from_millis(2));
                //unsafe { esp_idf_sys::vTaskDelay(1) };
                if Instant::now() - t > Duration::from_millis(1000) {
                    t = Instant::now();
                    //println!("{:?} thread 1", t);
                    println!("thread 1");
                }
            }
        })
        .unwrap();

    println!("Hello, world!");
}
```

With `println!("thread 0");` etc it prints:

```
thread 0
thread 1
thread 1
thread 0
thread 1
thread 0
```

With `println!("{:?} thread 0", t);` etc it prints:

```
Instant { tv_sec: 1, tv_nsec: 91379000 } thread 0
Instant { tv_sec: 2, tv_nsec: 93091000 } thread 0
Instant { tv_sec: 3, tv_nsec: 94130000 } thread 0
Instant { tv_sec: 4, tv_nsec: 95166000 } thread 0
Instant { tv_sec: 5, tv_nsec: 96238000 } thread 0
Instant { tv_sec: 6, tv_nsec: 97318000 } thread 0
Instant { tv_sec: 7, tv_nsec: 98390000 } thread 0
```

Notice that `thread 1` is completely missing.

ivmarkov, 2022-02-17:
What happens if you let the threads to sleep longer. Say - 50 millis?

ivmarkov, 2022-02-17:
Busylooping is only a thing on Arduino, with their toy examples of the `void loop() { busyloop(); }` style. Here, we have an RTOS underneath, so we should avoid that. That's what I think (with my limited knowledge in embedded stuff).

andresv, 2022-02-17:
I agree that busy looping is bad and it should be avoided if possible. However here I have 2 RTOS threads that have the same priority. Kernel does not preempt threads, freertos uses cooperative scheduling. Which means we must be sure to give other threads chance to run by yielding or sleeping.

I am probably just too stupid about freertos, but I thought that when sleep is called with whatever >0 arg it should suspend the current thread and check if some other thread is in ready state and then run that one.
That is why it is surprising for me why short sleeps seem not to yield.

```rust
std::thread::Builder::new()
    .name("thread 0".to_string())
    .spawn(move || {
        let mut t = Instant::now();
        loop {
            // does not work
            //std::thread::sleep(Duration::from_millis(2));
            // works
            std::thread::yield_now();
            if Instant::now() - t > Duration::from_millis(1000) {
                t = Instant::now();
                //println!("{:?} thread 0", t);
                println!("thread 0");
            }
        }
    })
    .unwrap();

std::thread::Builder::new()
    .name("thread 1".to_string())
    .spawn(move || {
        let mut t = Instant::now();
        loop {
            // does not work
            //std::thread::sleep(Duration::from_millis(2));
            // works
            if Instant::now() - t > Duration::from_millis(1000) {
                t = Instant::now();
                //println!("{:?} thread 1", t);
                println!("thread 1");
            }
        }
    })
    .unwrap();
```

EDIT: your point about `CONFIG_FREERTOS_HZ` is correct. Damn thing runs with 100Hz by default, therefore anything < 10ms do not make sense and are "rounded" to 0 which prevents yielding.
Defining `CONFIG_FREERTOS_HZ=1000` in `sdkconfig.defaults` fixes those busy loops.

closed 2022-02-17 by andresv

ivmarkov, 2022-02-17:
The only thing I miss in your explanations is why you need your threads to sleep so little? That is, if that's not confidential / i.e. commercial project etcetera. See the other thread in `esp-idf-template` for my shortest sleep time (that I plan to fix with GPIO interrupts).

andresv, 2022-02-17:
Original case where I blundered: https://gist.github.com/andresv/cfe1f1f5683aceaa55a0559bcd43372d#file-ubloxmqtt-rs-L108.

I wanted to act more quickly to bytes coming from serial.

ivmarkov, 2022-02-17:
I.c. 

I think we need blocking reads from UART implemented. That might be pretty easy:
* We need extra configuration parameter ('blocking reads" or so)
* [This line](https://github.com/esp-rs/esp-idf-hal/blob/master/src/serial.rs#L589) needs the `0` replaced with `portMAX_DELAY`.

Alternatively, you can access the UART peripheral via the ESP-IDF `VFS` as outlined [in this C example](https://github.com/espressif/esp-idf/blob/c69f0ec3292de2b9df5554405744296333d0feb2/examples/peripherals/uart/uart_select/main/uart_select_example_main.c). This should give you blocking reads _and_ you can use regular Rust STD FS read/writes.

ivmarkov, 2022-02-17:
Regarding the C example, funnily, it demonstrates async reads with `select`, but I think sync reads should be well possible too.

ivmarkov, 2022-02-17:
If it is not clear, blocking reads would allow your thread to sleep, and so you can get rid of the "busy looping".