esp-rs/esp-idf-sys #83: Compile only needed esp-idf components

N3xed, 2022-04-12:
Currently, when compiling the esp-idf *all* components are compiled by cmake, but only for select components the bindings are generated (everything that is included in `src/include/esp-idf/bindings.h`). That is currently around 1k C-files, which @igrr estimated could be brought down to around 600.
This behavior is not that detrimental as recompiling the esp-idf is normally rare (e.g. on first build, after clean). (There is also currently a bug where changing the `sdkconfig`/`sdkconfig.defaults` does not cause an `esp-idf` recompilation.)
However, when frequently changing the `sdkconfig`s or switing MCU, etc. and especially for CI such an improvement would be nice indeed.

To fix this we need to pass a list of components to cmake so that it knows specifically which components to build. As a first approach, we can disable building all components for which no bindings are generated (I don't how many that is, though).

In the long term though, we should make this configurable (maybe with an approach like I suggested in #72) as the demand for more and more esp-idf components grows.

ivmarkov, 2022-04-12:
> In the long term though, we should make this configurable (maybe with an approach like I suggested in #72) as the demand for more and more esp-idf components grows.

The approach suggested in #72 might be a bit of an overkill, as it is addressing the problem of how do we compile and generate `*-sys` bindings for *external* (out-of-tree) components. 

For "in-tree" components, that might be an overkill, especially given that we have like 70% of the solution already:
* Look at [EspIdfComponents](https://github.com/esp-rs/esp-idf-sys/blob/master/build/common.rs#L71) and how it is used [here](https://github.com/esp-rs/esp-idf-sys/blob/master/build/native.rs#L53), which is the CMake-first build.
* All that we need is a similar env-var for `native` build, which is (a) first passed down to CMake and (b) after that is parsed with a very similar code to the above, so that the bindings are generated only for the components which had actually been build. It env var is empty, the code should compile and generate bindings for everything, of course. 

From then on, everything is solved already - `esp-idf-sys`, but also `esp-idf-hal` and `esp-idf-svc` are sprinkled with "#ifdefs" using these component strings (they are passed as `cfg` flags to Rust), so that not just the unsafe wrappers in `esp-idf-sys`, but also the safe wrappers in `esp-idf-svc` and `esp-idf-hal` are conditionally compiled, depending on where the component was enabled or not.

ivmarkov, 2022-05-11:
Just to note that this task became even easier recently, as we no longer have a hard-coded list of esp idf components. I'm instead enumerating the components by doing directory scans instead. This should also make the task for plugging custom components easier.

N3xed, 2022-05-11:
Yup, I noticed it, just haven't gotten around to implementing it.
Thanks for the notice though.

maximeborges, 2022-05-31:
Regarding
> (There is also currently a bug where changing the sdkconfig/sdkconfig.defaults does not cause an esp-idf recompilation.)

is there any solution?

N3xed, 2022-05-31:
> Regarding
> 
> > (There is also currently a bug where changing the sdkconfig/sdkconfig.defaults does not cause an esp-idf recompilation.)
> 
> is there any solution?

This is fixed in master (see commit https://github.com/esp-rs/esp-idf-sys/commit/468b7920194622027af0ea92dc75c2f147a66a2f) but not in a release yet.

PR #115 (2022-08-02): Feature/specify built components
Add `$ESP_IDF_COMPONENTS` and `esp_idf_components` configuration options that allow to specify a list of esp-idf components to build. This list can be used to trim down the esp-idf build and reduce compile time.
Only implemented for the native build driver.

Fixes #83
commit 456f9867 (2022-08-02): Allow to specify which components are built (#115)

Add `$ESP_IDF_COMPONENTS` and `esp_idf_components` configuration option that allow
to specify a list of esp-idf components to build. This list can be used to trim down the esp-idf
build and reduce compile time.
Only implemented for the native build driver.
Fixes #83
closed 2022-08-02 by N3xed