esp-rs/esp-idf-sys #90: Build failure: fatal: No names found, cannot describe anything.

andresv, 2022-05-11:
On macOs 11.6.2.

```
error: failed to run custom build command for `esp-idf-sys v0.31.5`

Caused by:
  process didn't exit successfully: `/Users/andres/Development/yk/esp32r/target/release/build/esp-idf-sys-e6cd395419015806/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  Submodule path 'components/asio/asio': checked out 'f31694c9f1746ba189a4bcae2e34db15135ddb22'
  Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
  Submodule path 'components/bt/controller/lib_esp32': checked out 'b877f7e1fc98dccfcf4dbf31f215c5cb44ec3f0d'
  Submodule path 'components/bt/controller/lib_esp32c3_family': checked out '98dcc9591365b5ac486a9f0b474c36bf8c4ca97b'
  Submodule path 'components/bt/host/nimble/nimble': checked out '1dc1ec6e76b0ab3bf93cc9f1ff7a2a09141e7c61'
  Submodule path 'components/cbor/tinycbor': checked out '7c349dbb6b8d76db39383b226d3ebdf59b8ab37d'
  Submodule path 'components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
  Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
  Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
  Submodule path 'components/coap/libcoap': checked out '3aa11612c143c9734d72022720f33e12506f7a2c'
  Submodule path 'components/coap/libcoap/ext/tinydtls': checked out '59055b8a935bc53bf69d002fc089ad4bd08851b2'
  Submodule path 'components/esp_phy/lib': checked out '97e9e8992ea947fa46bc35545000b2c45d7b3060'
  Submodule path 'components/esp_wifi/lib': checked out '1b0fcf9eeb96ddc4d72ab5850a0bf9e95e289da4'
  Submodule path 'components/esptool_py/esptool': checked out 'aca9d70bd810c25bc953ca2f7587892c4dbaffc4'
  Submodule path 'components/expat/expat': checked out '57c7da69b78e3698e112a6b5da19d5109b8232d1'
  Submodule path 'components/ieee802154/lib': checked out 'f7b5e8059a3bb6f321e79ac3bf2aa4d2a9b93326'
  Submodule path 'components/json/cJSON': checked out 'd348621ca93571343a56862df7de4ff3bc9b5667'
  Submodule path 'components/libsodium/libsodium': checked out '4f5e89fa84ce1d178a6765b8b46f2b6f91216677'
  Submodule path 'components/lwip/lwip': checked out '76303df2386902e0d7873be4217f1d9d1b50f982'
  Submodule path 'components/mbedtls/mbedtls': checked out '8b0e35f2ad477fcc2a267cf434528024b8499085'
  Submodule path 'components/mqtt/esp-mqtt': checked out '985078affa8a2d2b56b87c8e6455252850f895c6'
  Submodule path 'components/nghttp/nghttp2': checked out '8f7b008b158e12de0e58247afd170f127dbb6456'
  Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '7c91efc1ffda769a5f1a872c646c82b00698f1b8'
  Submodule path 'components/nghttp/nghttp2/third-party/neverbleed': checked out 'b967ca054f48a36f82d8fcdd32e54ec5144f2751'
  Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '7c91efc1ffda769a5f1a872c646c82b00698f1b8'
  Submodule path 'components/openthread/lib': checked out '9a8d34d8f698cad2c9468468b473e26a3dda51b9'
  Submodule path 'components/openthread/openthread': checked out 'c36c0e77a2465355bcf13bd7dc718d8c9aa6ff64'
  Submodule path 'components/protobuf-c/protobuf-c': checked out 'dac1a65feac4ad72f612aab99f487056fbcf5c1a'
  Submodule path 'components/spiffs/spiffs': checked out 'f5e26c4e933189593a71c6b82cda381a7b21e41c'
  Submodule path 'components/tinyusb/tinyusb': checked out 'c4badd394eda18199c0196ed0be1e2d635f0a5f6'
  Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
  Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
  Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out '36d0642e66ff5b1c7a291873f24c498ca6ffedef'
  PATH=/Users/andres/Development/yk/esp32r/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  pip 21.2.3 from /Users/andres/.pyenv/versions/3.10.0/lib/python3.10/site-packages/pip (python 3.10)
  created virtual environment CPython3.10.0.final.0-64 in 3264ms
    creator CPython3Posix(dest=/Users/andres/Development/yk/esp32r/.embuild/espressif/python_env/idf4.4_py3.10_env, clear=False, no_vcs_ignore=False, global=False)
    seeder PipInvoke(download=False, pip=bundle, setuptools=bundle, wheel=bundle)
      added seed packages: pip==22.0.4, setuptools==62.1.0, wheel==0.37.1
    activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring None: markers 'sys_platform == "win32"' don't match your environment
  Requirement already satisfied: setuptools>=21 in /Users/andres/Development/yk/esp32r/.embuild/espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from -r /Users/andres/Development/yk/esp32r/.embuild/espressif/esp-idf/release-v4.4/requirements.txt (line 4)) (62.1.0)
  Collecting click>=7.0
    Using cached click-8.1.3-py3-none-any.whl (96 kB)
  Collecting pyserial>=3.3
    Using cached https://dl.espressif.com/pypi/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
  Collecting future>=0.15.2
    Downloading https://dl.espressif.com/pypi/future/future-0.18.2-py3-none-any.whl (491 kB)
       ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 491.1/491.1 KB 2.7 MB/s eta 0:00:00
  Collecting cryptography>=2.1.4
    Using cached cryptography-37.0.2-cp36-abi3-macosx_10_10_x86_64.whl (2.8 MB)
  Collecting pyparsing<2.4.0,>=2.0.3
    Using cached https://dl.espressif.com/pypi/pyparsing/pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
  Collecting pyelftools>=0.22
[... 312 lines cut ...]
  CMAKE = None
  running: "cmake" "/Users/andres/Development/yk/esp32r/target/xtensa-esp32-espidf/release/build/esp-idf-sys-b0c1db3b1613eb63/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/Users/andres/Development/yk/esp32r/.embuild/espressif/esp-idf/release-v4.4/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DCMAKE_INSTALL_PREFIX=/Users/andres/Development/yk/esp32r/target/xtensa-esp32-espidf/release/build/esp-idf-sys-b0c1db3b1613eb63/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections"
  -- Configuring incomplete, errors occurred!
  See also "/Users/andres/Development/yk/esp32r/target/xtensa-esp32-espidf/release/build/esp-idf-sys-b0c1db3b1613eb63/out/build/CMakeFiles/CMakeOutput.log".

  --- stderr
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Branch("release/v4.4") }
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.1 at '/Users/andres/Development/yk/esp32r/.embuild/espressif/esp-idf/release-v4.4'
  fatal: No names found, cannot describe anything.
  CMake Error at CMakeLists.txt:4 (project):
    Running

     '/Users/andres/Development/yk/esp32r/.embuild/espressif/tools/ninja/1.10.0/ninja' '--version'

    failed with:

     No such file or directory

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /Users/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

andresv, 2022-05-11:
`fatal: No names found, cannot describe anything.` seems to happen if git tags are not there: https://stackoverflow.com/questions/4916492/git-describe-fails-with-fatal-no-names-found-cannot-describe-anything

ivmarkov, 2022-05-11:
Isn't the actual problem this o e:
No such file or directory

ivmarkov, 2022-05-11:
Strange. Ninja is supposed to be installed during the build...

ivmarkov, 2022-05-11:
Ok so it seems the build installs ninja 1.10.2, but then tries to run ninja 1.10.0. Let me see...

ivmarkov, 2022-05-11:
Can you please wipe out .embuild and then cargo clean and try again?

andresv, 2022-05-11:
That helped.

closed 2022-05-11 by andresv

stevefan1999-personal, 2022-07-14:
I'm still having the same issue on Windows. Removed embuild, wiped cargo output, still same problem

stevefan1999-personal, 2022-07-14:
<details>
<summary>Click here to expand build log</summary>

```
(base) PS C:\Users\steve\rust\abcd> cargo build 
   Compiling esp-idf-sys v0.31.6
error: failed to run custom build command for `esp-idf-sys v0.31.6`

Caused by:
  process didn't exit successfully: `C:\Users\steve\rust\abcd\target\debug\build\esp-idf-sys-cafd80a349bfdbb2\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  Submodule path 'components/asio/asio': checked out 'f31694c9f1746ba189a4bcae2e34db15135ddb22'
  Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
  Submodule path 'components/bt/controller/lib_esp32': checked out 'b877f7e1fc98dccfcf4dbf31f215c5cb44ec3f0d'
  Submodule path 'components/bt/controller/lib_esp32c3_family': checked out '98dcc9591365b5ac486a9f0b474c36bf8c4ca97b'
  Submodule path 'components/bt/host/nimble/nimble': checked out 'dac3a2a0f54589b10a4cff2a99526804747089df'
  Submodule path 'components/cbor/tinycbor': checked out '7c349dbb6b8d76db39383b226d3ebdf59b8ab37d'
  Submodule path 'components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
  Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
  Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
  Submodule path 'components/coap/libcoap': checked out '3aa11612c143c9734d72022720f33e12506f7a2c'
  Submodule path 'components/coap/libcoap/ext/tinydtls': checked out '59055b8a935bc53bf69d002fc089ad4bd08851b2'
  Submodule path 'components/esp_phy/lib': checked out 'ff0d771b8e33e320e11634567ee53b9cd78e6be1'
  Submodule path 'components/esp_wifi/lib': checked out 'ce03645819a4c6233270a45e53484703fabc320f'
  Submodule path 'components/esptool_py/esptool': checked out 'e53efcf84ceb6fac315ccb1348d6950ac33af309'
  Submodule path 'components/expat/expat': checked out '57c7da69b78e3698e112a6b5da19d5109b8232d1'
  Submodule path 'components/ieee802154/lib': checked out 'f7b5e8059a3bb6f321e79ac3bf2aa4d2a9b93326'
  Submodule path 'components/json/cJSON': checked out 'd348621ca93571343a56862df7de4ff3bc9b5667'
  Submodule path 'components/libsodium/libsodium': checked out '4f5e89fa84ce1d178a6765b8b46f2b6f91216677'
  Submodule path 'components/lwip/lwip': checked out '76303df2386902e0d7873be4217f1d9d1b50f982'
  Submodule path 'components/mbedtls/mbedtls': checked out '8b0e35f2ad477fcc2a267cf434528024b8499085'
  Submodule path 'components/mqtt/esp-mqtt': checked out '27eb4726067465c5c67d4ecdca5ddccd26f02580'
  Submodule path 'components/nghttp/nghttp2': checked out '8f7b008b158e12de0e58247afd170f127dbb6456'
  Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '7c91efc1ffda769a5f1a872c646c82b00698f1b8'
  Submodule path 'components/nghttp/nghttp2/third-party/neverbleed': checked out 'b967ca054f48a36f82d8fcdd32e54ec5144f2751'
  Submodule path 'components/openthread/lib': checked out '9a8d34d8f698cad2c9468468b473e26a3dda51b9'
  Submodule path 'components/openthread/openthread': checked out 'c36c0e77a2465355bcf13bd7dc718d8c9aa6ff64'
  Submodule path 'components/protobuf-c/protobuf-c': checked out 'f224ab2eeb648a818eb20687d7150a285442c907'
  Submodule path 'components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
  Submodule path 'components/tinyusb/tinyusb': checked out 'c4badd394eda18199c0196ed0be1e2d635f0a5f6'
  Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
  Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
  Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out '36d0642e66ff5b1c7a291873f24c498ca6ffedef'
  PATH=C:\Users\steve\rust\abcd\.embuild\espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Users\steve\rust\abcd\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%
  pip 21.2.4 from C:\tools\Anaconda3\lib\site-packages\pip (python 3.9)
  created virtual environment CPython3.9.7.final.0-64 in 8835ms
    creator CPython3Windows(dest=C:\Users\steve\rust\abcd\.embuild\espressif\python_env\idf4.4_py3.9_env, clear=False, no_vcs_ignore=False, global=False)
    seeder PipInvoke(download=False, pip=bundle, setuptools=bundle, wheel=bundle)
      added seed packages: pip==22.1.2, setuptools==62.6.0, wheel==0.37.1
    activators BashActivator,BatchActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Processing c:\users\steve\rust\abcd\.embuild\espressif\esp-idf\release-v4.4\tools\kconfig_new\esp-windows-curses
    Preparing metadata (setup.py): started
    Preparing metadata (setup.py): finished with status 'done'
  Requirement already satisfied: setuptools>=21 in c:\users\steve\rust\abcd\.embuild\espressif\python_env\idf4.4_py3.9_env\lib\site-packages (from -r C:\Users\steve\rust\abcd\.embuild\espressif\esp-idf\release-v4.4\requirements.txt (line 4)) (62.6.0)
  Collecting click>=7.0
[... cut ...]
   * branch            7e8e249990ec491ec15990cf95b6d871a66cf64a -> FETCH_HEAD
  From https://github.com/espressif/esp-cryptoauthlib
   * branch            36d0642e66ff5b1c7a291873f24c498ca6ffedef -> FETCH_HEAD
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.1 at 'C:\Users\steve\rust\abcd\.embuild\espressif\esp-idf\release-v4.4'
  fatal: No names found, cannot describe anything.
  Error: Access is denied. (os error 5)
```

</details>

ivmarkov, 2022-07-14:
> Click here to expand build log

Can you re-run the build but with env var `RUST_BACKTRACE=1` exported and set priot to running, so that we can see where exactly the Rust build code fails for you?

glynnmccabe, 2022-07-16:
I was having a similar issue compiling under windows, seems to be caused by a bug in embuild. PR here: https://github.com/esp-rs/embuild/pull/56

N3xed, 2022-07-16:
Fixed with embuild version `0.29.3`, just run `cargo update`.
Thanks, @glynnmccabe for the fix!