esp-rs/esp-idf-sys #92: `fromenv` issues

MabezDev, 2022-05-11:
I think I misread the summary from https://github.com/esp-rs/esp-idf-sys/pull/42#issuecomment-1095670588 initially, I thought the variable it was checking was the ESP_IDF_VERSION, not ESP_IDF_TOOLS_INSTALL_DIR (my bad). This may break builds for users with esp-idf installed on their system because our templates or demos have never explicitly set ESP_IDF_TOOLS_INSTALL_DIR, meaning upon upgrading esp-idf-sys version it will now select there env esp-idf instead.

Is there a reason it was designed like this? I think we either find a fix here, or update all demo's and templates to explicitly set ESP_IDF_TOOLS_INSTALL_DIR to workspace as a default.

ivmarkov, 2022-05-11:
It will select and use their ESP-IDF environment **only** if the environment was activated. Meaning, they did the `. export.sh` thing and have the Espressif toolset on their path.

Or do you mean selecting & using their **ESP-IDF** (but **not** the activated environment)? This is controlled with another variable, `IDF_PATH`, which is also **the** variable used by ESP-IDF itself. For this particular case, you are right - we'll start silently using the users' ESP-IDF SDK, even if it was not activated, as long as this environment variable is set.

But is this the end of the world? I find this to be the correct behavior actually?

MabezDev, 2022-05-11:
Ah yes, you're right, I just realised I had it set in my env permanently as this used to be the way in esp-idf v3. Sorry for the noise!

closed 2022-05-11 by MabezDev

N3xed, 2022-05-11:
Also note:
If you build with `-vv` (to show build script output), it will actually output which `esp-idf` it uses and where (and what version if available).