esp-rs/esp-idf-sys #95: It will show that python is not installed.

hakohumi, 2022-05-17:
https://github.com/esp-rs/esp-idf-template

I generated a project based on this site and ran `cargo build` and got the following.

```
C:\train-rust-esp32\test>cargo build      
   Compiling esp-idf-sys v0.31.5
error: failed to run custom build command for `esp-idf-sys v0.31.5`

Caused by:
  process didn't exit successfully: `C:\train-rust-esp32\test\target\debug\build\esp-idf-sys-579560c2cbc04aea\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU

  --- stderr
  Using custom user-supplied esp-idf repository at 'C:\Users\<user name>\esp\esp-idf' (detected from env variable `IDF_PATH`)
  Error: Could not install esp-idf

  Caused by:
      0: Failed to locate python. Is python installed and in your $PATH?
      1: command '"python" "--version"' failed to start
      2: program not found
```

Python is already installed.

```
C:\train-rust-esp32\test>where python
C:\Users\<user name>\.pyenv\pyenv-win\shims\python
C:\Users\<user name>\.pyenv\pyenv-win\shims\python.bat
```

MabezDev, 2022-05-17:
> Using custom user-supplied esp-idf repository at 'C:\Users\<user name>\esp\esp-idf' (detected from env variable `IDF_PATH`)

This is the offending line. You need to remove $IDF_PATH from your environment, unless you are using a properly activated environment (running export.sh inside esp-idf).

hakohumi, 2022-05-18:
The .cargo/config.toml looks like this

```
[build]
# Uncomment the relevant target for your chip here (ESP32, ESP32-S2, ESP32-S3 or ESP32-C3)
target = "xtensa-esp32-espidf"
#target = "xtensa-esp32s2-espidf"
#target = "xtensa-esp32s3-espidf"
#target = "riscv32imc-esp-espidf"

[target.xtensa-esp32-espidf]
linker = "ldproxy"

[target.xtensa-esp32s2-espidf]
linker = "ldproxy"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"

# Future - necessary for the experimental "native build" of esp-idf-sys with ESP32C3
# See also https://github.com/ivmarkov/embuild/issues/16
rustflags = ["-C", "default-linker-libraries"]

[unstable]

build-std = ["std", "panic_abort"]
#build-std-features = ["panic_immediate_abort"] # Required for older ESP-IDF versions without a realpath implementation

[env]
# Note: these variables are not used when using pio builder
# Enables the esp-idf-sys "native" build feature (`cargo build --features native`) to build against ESP-IDF stable (v4.4)
# ESP_IDF_VERSION = { value = "branch:release/v4.4" }
# Enables the esp-idf-sys "native" build feature (`cargo build --features native`) to build against ESP-IDF master (mainline)
#ESP_IDF_VERSION = { value = "master" }
IDF_PATH = "C:\\Users\\<user name>\\esp\\esp-idf"
ESP_IDF_TOOLS_INSTALL_DIR = "custom:C:\\Users\\<user name>\\.espressif"
# ESP_IDF_REPOSITORY = "https://github.com/espressif/esp-idf.git"
# ESP_IDF_SDKCONFIG_DEFAULTS = ""
# ESP_IDF_SDKCONFIG = ""
MCU = "esp32"

```

I added the [env] part with reference to the README.md of esp-idf-sys.

Before I added this, I was getting errors like this.

```

C:\train-rust-esp32\test>cargo build
    Updating crates.io index
   Compiling winapi v0.3.9                                                                      
   Compiling compiler_builtins v0.1.70
   Compiling core v0.0.0 (C:\Users\<user name>\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core)
   Compiling cfg-if v1.0.0
   Compiling proc-macro2 v1.0.39
   Compiling autocfg v1.1.0
   Compiling unicode-ident v1.0.0
   Compiling memchr v2.5.0
   Compiling lazy_static v1.4.0
   Compiling syn v1.0.95
   Compiling crossbeam-utils v0.8.8
   Compiling log v0.4.17
   Compiling regex-syntax v0.6.25
   Compiling glob v0.3.0
   Compiling libc v0.2.126
   Compiling scopeguard v1.1.0
[... 254 lines cut ...]
   * branch            dac1a65feac4ad72f612aab99f487056fbcf5c1a -> FETCH_HEAD
  From https://github.com/pellepl/spiffs
   * branch            f5e26c4e933189593a71c6b82cda381a7b21e41c -> FETCH_HEAD
  From https://github.com/espressif/tinyusb
   * branch            c4badd394eda18199c0196ed0be1e2d635f0a5f6 -> FETCH_HEAD
  From https://github.com/ThrowTheSwitch/Unity
   * branch            7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 -> FETCH_HEAD
  From https://github.com/leethomason/tinyxml2
   * branch            7e8e249990ec491ec15990cf95b6d871a66cf64a -> FETCH_HEAD
  From https://github.com/espressif/esp-cryptoauthlib
   * branch            36d0642e66ff5b1c7a291873f24c498ca6ffedef -> FETCH_HEAD
  Error: Could not install esp-idf

  Caused by:
      0: Failed to locate python. Is python installed and in your $PATH?
      1: command '"python" "--version"' failed to start
      2: program not found
```

## python --version

```
C:\train-rust-esp32\test>python --version
Python 3.10.2
```

hakohumi, 2022-05-18:
# Backtrace of cargo build

```

C:\train-rust-esp32\test>cargo +esp build 
   Compiling esp-idf-sys v0.31.5
error: failed to run custom build command for `esp-idf-sys v0.31.5`

Caused by:
  process didn't exit successfully: `C:\train-rust-esp32\test\target\debug\build\esp-idf-sys-579560c2cbc04aea\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU

  --- stderr
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Branch("release/v4.4") }
  Error: Could not install esp-idf

  Caused by:
      0: Failed to locate python. Is python installed and in your $PATH?
      1: command '"python" "--version"' failed to start
      2: program not found

  Stack backtrace:
     0: std::backtrace_rs::backtrace::trace_unsynchronized
     1: std::backtrace::Backtrace::capture
     2: std::backtrace::Backtrace::capture
     3: anyhow::context::ext::impl$0::ext_context<enum$<embuild::utils::CmdError>,str>
               at C:\Users\<user name>\.cargo\registry\src\github.com-1ecc6299db9ec823\anyhow-1.0.57\src\context.rs:27
     4: anyhow::context::impl$0::context::closure$0<alloc::string::String,enum$<embuild::utils::CmdError>,str>
               at C:\Users\<user name>\.cargo\registry\src\github.com-1ecc6299db9ec823\anyhow-1.0.57\src\context.rs:50
     5: enum$<core::result::Result<alloc::string::String,enum$<embuild::utils::CmdError> > >::map_err<alloc::string::String,enum$<embuild::utils::CmdError>,anyhow::Error,anyhow::context::impl$0::context::closure_env$0<alloc::string::String,enum$<embuild::utils::C
               at R:\library\core\src\result.rs:842
     6: anyhow::context::impl$0::context<alloc::string::String,enum$<embuild::utils::CmdError>,str>
               at C:\Users\<user name>\.cargo\registry\src\github.com-1ecc6299db9ec823\anyhow-1.0.57\src\context.rs:50
     7: embuild::python::check_python_at_least
               at C:\Users\<user name>\.cargo\registry\src\github.com-1ecc6299db9ec823\embuild-0.29.1\src\python.rs:12
     8: embuild::espidf::Installer::install
               at C:\Users\<user name>\.cargo\registry\src\github.com-1ecc6299db9ec823\embuild-0.29.1\src\espidf.rs:473
     9: build_script_build::build_driver::build_cargo_first::closure$1
               at .\build\native.rs:187
    10: build_script_build::build_driver::build_cargo_first
               at .\build\native.rs:220
    11: build_script_build::build_driver::build
               at .\build\native.rs:45
    12: build_script_build::main
               at .\build\build.rs:46
    13: core::ops::function::FnOnce::call_once<enum$<core::result::Result<tuple$<>,anyhow::Error>, 1, 18446744073709551615, Err> (*)(),tuple$<> >
               at R:\library\core\src\ops\function.rs:227
    14: std::sys_common::backtrace::__rust_begin_short_backtrace<enum$<core::result::Result<tuple$<>,anyhow::Error>, 1, 18446744073709551615, Err> (*)(),enum$<core::result::Result<tuple$<>,anyhow::Error>, 1, 18446744073709551615, Err> >
               at R:\library\std\src\sys_common\backtrace.rs:122
    15: std::rt::lang_start::closure$0<enum$<core::result::Result<tuple$<>,anyhow::Error>, 1, 18446744073709551615, Err> >
               at R:\library\std\src\rt.rs:145
    16: std::panic::catch_unwind
    17: std::rt::lang_start_internal
    18: std::rt::lang_start<enum$<core::result::Result<tuple$<>,anyhow::Error>, 1, 18446744073709551615, Err> >
               at R:\library\std\src\rt.rs:144
    19: main
    20: invoke_main
               at d:\a01\_work\20\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    21: __scrt_common_main_seh
               at d:\a01\_work\20\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    22: BaseThreadInitThunk
    23: RtlUserThreadStart

C:\train-rust-esp32\test>
```

hakohumi, 2022-05-18:
# I was able to build successfully

## What we did

Installed Python 3.10.4 with the official Python installer.
Select "Add to PATH" during installation.

```bat
C:\train-rust-esp32\test>python --version
Python 3.10.4

C:\train-rust-esp32\test>where python
C:\Users\<user name>\AppData\Local\Programs\Python\Python310\python.exe
C:\Users\<user name>\.pyenv\pyenv-win\shims\python
C:\Users\<user name>\.pyenv\pyenv-win\shims\python.bat
```

## Build log

```bat
C:\train-rust-esp32\test>cargo build
   Compiling bindgen v0.59.2
   Compiling embuild v0.29.1
   Compiling esp-idf-sys v0.31.5
   Compiling test v0.1.0 (C:\train-rust-esp32\test)
    Finished dev [optimized + debuginfo] target(s) in 1m 00s
````

I was using pytnon with pyenv-win, but found that this does not work.

I don't know why I shouldn't have to install a new one, since the path should be through pyenv alone.

Was the problem with embuild in the first place, rather than esp-idf-sys...?

MabezDev, 2022-05-18:
I just looked at the pyenv docs, it seems like it adds a `$PATH` entry to the _user_ `$PATH`, whereas I think the python installer uses the _system_ `$PATH`. That's my best guess, but python on windows is somewhat wonky :sweat_smile:. Glad you managed to get it built, closing for now :).

N3xed, 2022-05-18:
> I was using python with pyenv-win, but found that this does not work.
> 
> I don't know why I shouldn't have to install a new one, since the path should be through pyenv alone.

I can only speculate as I don't know how pyenv-win does its python shims. This issue relates to how we run `python` (for example in the [`embuild::python::check_python_at_least`](https://github.com/esp-rs/embuild/blob/a5b3d7ebc76fbcf82d15954112d8202db8e2dcb8/src/python.rs#L11) function). 

Currently, we use the [`std::process::Command`](https://doc.rust-lang.org/std/process/struct.Command.html) API and it's not really documented how that API deals with file extensions on Windows. But from reading the code and going though some issues it may be that it actually always adds a `.exe` extension if the program has no file extension (in our case we use just `python` as the program path).

So even though there is `python` and `python.bat` in $PATH it actually runs `python.exe` which is not found and fails.

> I just looked at the pyenv docs, it seems like it adds a `$PATH` entry to the _user_ `$PATH`, whereas I think the python installer uses the _system_ `$PATH`.

That shouldn't be an issue, at least this [SuperUser answer](https://superuser.com/questions/867728/user-vs-system-environment-variables-do-system-variables-override-user-variabl) says that the system $PATH is prepended to the user $PATH.