esp-rs/esp-idf-sys #97: Move esp-idf bootloader and partition table to known location after build

MabezDev, 2022-05-19:
Currently, they are tucked away in the build folder of esp-idf-sys, in which a unique fingerprint is appended to the folder name. E.g

`target/xtensa-esp32-espidf/debug/build/esp-idf-sys-1d1212f75c9bfd7a/out/build/bootloader/bootloader.bin`

Because of the fingerprint, it's not possible for tooling (like espflash) to grab the bootloader and partition table automatically. If we moved it to a known location in the target directory we could allow tooling to automatically pick up bootload and partition table changes.

I think moving them to the `$TARGET/$PROFILE/` makes sense. E.g `target/xtensa-esp32-espidf/debug/build/esp-idf-sys-1d1212f75c9bfd7a/out/build/bootloader/bootloader.bin` would get copied to `target/xtensa-esp32-espidf/debug/bootloader.bin`.

N3xed, 2022-05-19:
You can actually build the crate with `cargo build --message-format json` and extract these messages with the [cargo_metadata](https://docs.rs/cargo_metadata/latest/cargo_metadata/) crate (using [`cargo_metadata::MessageIter`](https://docs.rs/cargo_metadata/latest/cargo_metadata/struct.MessageIter.html)). One such message is [`BuildScript`](https://docs.rs/cargo_metadata/latest/cargo_metadata/struct.BuildScript.html) which contains the path to the our-dir of that build script.

I actually use that approach in my (very long) WIP `cargo-idf` tool [here](https://github.com/N3xed/embuild/blob/cargo-idf/cargo-idf/src/build.rs).  You can get almost all information with that and `cargo metadata`.

MabezDev, 2022-05-19:
That's a good point! Definitely useful for the cases where the tool is invoking the cargo build process. In the case of espflash (note, not cargo-espflash, the wrapper) it is typically used as a cargo runner, and has no sway over cargo's build process.

Hypothetical usage I am picturing.

```toml
[target.xtensa-esp32s2-espidf]
runner = "espflash --bootloader target/bootloader.bin --monitor"
```

I suppose it be possible to write a build script that could find the esp-idf-sys directory inside `target/` and move it to where ever?

PR #264 (2023-11-28): Copy bootloader and partition table to target folder
After the build completes, the `bootloader.bin` and `partition-table.bin` files will be copied to the target folder, next to the build binaries. This allows tools such as espflash to use the bootloader and partition table more easily (see #97).

For example, this will copy `target/xtensa-esp32-espidf/release/build/esp-idf-sys-66049b51093b6161/out/build/bootloader/bootloader.bin` to just `target/xtensa-esp32-espidf/release/bootloader.bin`.

Closes #97
closed 2023-11-28 by ivmarkov