esp-rs/esp-idf-sys #105: Compilation failed because of different git language

Zduniusz, 2022-06-17:
I've tried to build a project using [this template](https://github.com/esp-rs/esp-idf-template)

But can't get past the following error:
```
error: failed to run custom build command for `esp-idf-sys v0.31.6`

Caused by:
  process didn't exit successfully: `/home/zduniusz/Documents/rustESP/mrugacz/target/release/build/esp-idf-sys-d3fd8d76cc7c38f9/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  IDF_PYTHON_ENV_PATH=/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/python_env/idf4.4_py3.10_env
  PATH=/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/tools/cmake/3.20.3/bin:/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/tools/ninja/1.10.2/:/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/python_env/idf4.4_py3.10_env/bin:/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  Current system platform: linux-amd64
  Skipping xtensa-esp32-elf@esp-2021r2-patch3-8.4.0 (already installed)
  Skipping cmake@3.20.3 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
  IDF_PYTHON_ENV_PATH=/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/python_env/idf4.4_py3.10_env
  PATH=/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/tools/cmake/3.20.3/bin:/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/tools/ninja/1.10.2/:/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/python_env/idf4.4_py3.10_env/bin:/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH

  --- stderr
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Branch("release/v4.4") }
  fatal: Nie znaleziono nazw, nie można nic opisać.
  Using esp-idf v4.4.1 at '/home/zduniusz/Documents/rustESP/mrugacz/.embuild/espressif/esp-idf/release-v4.4'
  fatal: Nie znaleziono nazw, nie można nic opisać.
  Error: 'git remote show' yielded invalid output: '* zdalne repozytorium origin
    Adres do pobierania: https://github.com/espressif/esp-idf.git
    Adres do wypychania: https://github.com/espressif/esp-idf.git
    gałąź HEAD: master
    Zdalna gałąź:
      release/v4.4 śledzone
    Lokalna gałąź ustawiona do „git pull”:
      release/v4.4 scala się ze zdalnym repozytorium release/v4.4
    Lokalna referencja ustawiona do „git push”:
      release/v4.4 wypycha na release/v4.4 (aktualne)'

```

It looks like it tries to get some information from git using `git remote show` command but can't understand its response because it's in non-default language?

I've tried to change its language by creating a shell script in w /usr/bin which adds `LANG=en_GB` to every git command call, but that didn't succeed.

N3xed, 2022-06-19:
I think this is related to https://github.com/esp-rs/embuild/pull/50.
Could you patch `embuild` with its latest master, i.e.
```toml
[patch.crates-io]
embuild = { git = "https://github.com/esp-rs/embuild.git"}
```
to verify this?

Zduniusz, 2022-06-19:
Thanks, unfortunately I'm unable to verify this, because I brute forced my problem by removing all locales from my machine, forcing git to use English. Thanks for replying though.