esp-rs/esp-idf-sys #106: mainline build fails

spookyvision, 2022-06-23:
```code

❯ cargo generate --git https://github.com/esp-rs/esp-idf-template cargo
🤷   Project Name : itsok
🔧   Basedir: /private/tmp/aec.22_05_31_23_09_01.vl ...
🔧   Generating template ...
✔ 🤷   ESP-IDF native build version (v4.3.2 = previous stable, v4.4 = stable, mainline = UNSTABLE) · mainline
✔ 🤷   Rust toolchain (beware: nightly works only for esp32c3!) · nightly
✔ 🤷   STD support · true
✔ 🤷   MCU · esp32c3
✔ 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? (beware: Dev Containers not available for esp-idf v4.3.2) · false

❯ cd itsok

❯ cargo build
```

(time passes)

the full error log is too long for a single issue, I've put it in [a pastebin](https://dpaste.org/s7DJO).
Excerpt:

```code
error: failed to run custom build command for `esp-idf-sys v0.31.6`

Caused by:
  process didn't exit successfully: `/private/tmp/aec.22_05_31_23_09_01.vl/itsok/target/debug/build/esp-idf-sys-79e5b3c57cd1e10f/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU

(...)

Using esp-idf v5.0.0 at '/private/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master'
  Output: "/private/tmp/aec.22_05_31_23_09_01.vl/itsok/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-5362e011e750e18b/out/bindings.rs"
  Bindgen builder flags: ["/Users/ace/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.6/src/include/esp-idf/bindings.h", "--rust-target", "1.47", "--blocklist-function", "strtold", "--blocklist-function", "_strtold_r", "--blocklist-function", "v.*printf", "--blocklist-function", "v.*scanf", "--blocklist-function", "_v.*printf_r", "--blocklist-function", "_v.*scanf_r", "--blocklist-function", "esp_log_writev", "--no-layout-tests", "--with-derive-default", "--ctypes-prefix", "c_types", "--generate", "functions,types,vars,methods,constructors,destructors", "--use-core", "--no-rustfmt-bindings", "--", "-D__bindgen", "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"", "-DUNITY_INCLUDE_CONFIG_H", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/riscv/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-5362e011e750e18b/out/build/config", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/newlib/platform_include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/freertos/FreeRTOS-Kernel/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/freertos/esp_additions/include/freertos", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/freertos/FreeRTOS-Kernel/portable/riscv/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/freertos/esp_additions/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_hw_support/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_hw_support/include/soc", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_hw_support/include/soc/esp32c3", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_hw_support/port/esp32c3/.", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_hw_support/port/esp32c3/private_include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/heap/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/log/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/soc/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/soc/esp32c3/.", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/soc/esp32c3/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/hal/esp32c3/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/hal/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/hal/platform_port/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_rom/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_rom/include/esp32c3", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_rom/esp32c3", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_common/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_system/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_system/port/soc", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_system/port/include/riscv", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_system/port/include/private", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/lwip/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/lwip/include/apps", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/lwip/include/apps/sntp", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/lwip/lwip/src/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/lwip/port/esp32/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/lwip/port/esp32/include/arch", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/vfs/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_wifi/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_event/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_netif/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_phy/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_phy/esp32c3/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_ringbuf/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/efuse/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/efuse/esp32c3/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/driver/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/driver/deprecated", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_pm/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/mbedtls/port/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/mbedtls/mbedtls/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/mbedtls/mbedtls/library", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/mbedtls/esp_crt_bundle/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/app_update/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/spi_flash/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/bootloader_support/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/bootloader_support/bootloader_flash/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/pthread/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/esp_timer/include", "-I/tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/app_trace/include", "-I/tmp/aec.22 [... cut ...]
  /tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/hal/include/hal/mcpwm_types.h:18:9: error: unknown type name 'soc_periph_mcpwm_timer_clk_src_t', err: true
  /tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/hal/include/hal/mcpwm_types.h:23:9: error: unknown type name 'soc_periph_mcpwm_capture_clk_src_t', err: true
  /tmp/aec.22_05_31_23_09_01.vl/itsok/.embuild/espressif/esp-idf/master/components/driver/include/driver/mcpwm.h:77:34: error: use of undeclared identifier 'SOC_MCPWM_GROUPS', err: true
```

MabezDev, 2022-06-23:
Looks like there are changes to the MCPWM interface, confirmed by https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/peripherals.html#mcpwm.

MabezDev, 2022-09-11:
I believe this should now be fixed, at least a fresh clone from mainline works for me now.

closed 2022-09-11 by MabezDev