esp-rs/esp-idf-sys #112: Overriding compile config for IDF components

tcbennun, 2022-07-22:
I want to add a compile definition into the compilation of the `bt` component in ESP-IDF, in order to change a piece of NimBLE config that isn't exposed by ESP-IDF as a Kconfig entry.

In an ESP-IDF C project, I would achieve this by adding this line to my project's CMakeLists.txt:

```cmake
target_compile_definitions(__idf_bt PRIVATE "MYNEWT_VAL_BLE_GATT_MAX_PROCS=16")
```

If I add this to esp-idf-sys's `resources/cmake_project/CMakeLists.txt` (at any point in the file), compilation fails with:

```
 CMake Error at CMakeLists.txt:9 (target_compile_definitions):
    target_compile_definitions may only set INTERFACE properties on INTERFACE
    targets
```

I don't understand what makes this build process different to `idf.py`'s, but it's very annoying that I can't set this. Is there another way?

N3xed, 2022-07-25:
> In an ESP-IDF C project, I would achieve this by adding this line to my project's CMakeLists.txt:
> 
> ```cmake
> target_compile_definitions(__idf_bt PRIVATE "MYNEWT_VAL_BLE_GATT_MAX_PROCS=16")
> ```

Works as expected when
```
CONFIG_CT_ENABLED=y
```
is set in the sdkconfig (at least on my end).

tcbennun, 2022-07-26:
Interesting. I already have that in my sdkconfig as I'm working on a BLE application! Whereabouts are you putting the `target_compile_definitions` line in the CMakeLists? And which ESP-IDF? I'm using v4.4.1.

N3xed, 2022-07-26:
Just at the end of the file. I'm using branch `release/v4.4` (commit 5e6cffbb14fa78e6e8475550c1606b29ec1aa7f0), haven't updated in a while, so I don't know if this is the HEAD of `release/v4.4`.

N3xed, 2022-08-02:
@tcbennun Were you able to make it work?
If not, a repository where this issue is reproducible would be appreciated.

tcbennun, 2022-08-08:
Yup, I just got this to work! I don't know what I was doing differently before.

(Slightly annoyingly, and this is specific to this example, the change won't be picked up in the generated bindings (if I add the def to the `libespidf.elf` target then the constant will disappear due to how `esp_nimble_cfg.h` works), but I can live with that.)

It would be cool to be able to customise the build at this level per project. Perhaps an env var that overrides the path to CMakeLists.txt? Or the base CMakeLists.txt could contain a conditional `include()` of a user-supplied file?

N3xed, 2022-08-26:
> It would be cool to be able to customise the build at this level per project. Perhaps an env var that overrides the path to CMakeLists.txt?

This is currently possible by using [globbing](https://github.com/esp-rs/esp-idf-sys#esp_idf_glob_xxx_base-and-esp_idf_glob_xxx_yyy) to copy files to the dummy cmake project, for example:

(In `.cargo/config.toml`)
```toml
[env]
# src folder in parent directory
ESP_IDF_GLOB_CMAKE_BASE = { value = "src", relative = true }
# only copy the file named `CMakeLists.txt`
ESP_IDF_GLOB_CMAKE_CMAKELISTS = "CMakeLists.txt"
```

This will replace the `CMakeLists.txt` file with a custom one, which is a bit hacky since one would have to copy the contents of the `resources/CMakeLists.txt` file and update it every time it is changed between esp-idf-sys versions.

You could also supply your own `esp-idf` with [`ESP_IDF_REPOSITORY`](https://github.com/esp-rs/esp-idf-sys#esp_idf_repository-esp_idf_repository-native-builder-only] or [`IDF_PATH`](https://github.com/esp-rs/esp-idf-sys#idf_path-idf_path-native-builder-only).

> Or the base CMakeLists.txt could contain a conditional `include()` of a user-supplied file?

I'm not sure this is worth it. Also, it isn't clear when the custom cmake should be included. I think a better approach would be to provide a path of the `resources/CMakeLists.txt` file once it gets overridden in the dummy project so that the custom file can include it whenever. And this avoids adding another configuration option.