esp-rs/esp-idf-sys #116: add cutome component to idf

farbodpm, 2022-08-14:
hello,
I need to add some components to "sys-idf". how can I do it?

I tested this way.

    clone sys-idf-hal
    add esp-littlfs to components
    add esp-littlefs to binding.h
    build it.

but I think this is not right.

N3xed, 2022-08-14:
You can use [extra components](https://github.com/esp-rs/esp-idf-sys#extra-esp-idf-components) to do this.
See https://github.com/esp-rs/esp-idf-sys/pull/114#issuecomment-1207168854 for an example.

closed 2022-08-14 by N3xed

farbodpm, 2022-08-15:
I face this error. I don't know why.

- Configuring done
  -- Generating done
  -- Build files have been written to: /home/esp/Documents/reader/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-164d1a2c2b47d29d/out/build/bootloader
  [1/1] cd /home/esp/Documents/reader/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-164d1a2c2b47d29d/out/build/bootloader/esp-idf/esptool_py && /home/esp/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/esp/.espressif/esp-idf-5d813d762119857b/release-v4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/esp/Documents/reader/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-164d1a2c2b47d29d/out/build/bootloader/bootloader.bin
  Bootloader binary size 0x51d0 bytes. 0x2e30 bytes (36%) free.
  ninja: build stopped: subcommand failed.

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: Some(
          Global,
      ),
      esp_idf_sdkconfig: Some(
          "sdkconfig",
      ),
      esp_idf_sdkconfig_defaults: Some(
          [
              "sdkconfig.defaults",
          ],
      ),
      mcu: Some(
          "esp32s3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Branch(
                  "release/v4.4",
              ),
          ),
          esp_idf_repository: Some(
              "https://github.com/espressif/esp-idf.git",
          ),
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [
              ExtraComponent {
                  component_dirs: [
                      "c_libraries/esp_littlefs",
                  ],
                  bindings_header: Some(
                      "c_libraries/bindings.h",
                  ),
                  bindings_module: None,
                  manifest_dir: "/home/esp/Documents/reader",
              },
          ],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: EspIdfRemote { repo_url: Some("https://github.com/espressif/esp-idf.git"), git_ref: Branch("release/v4.4") }
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.2 at '/home/esp/.espressif/esp-idf-5d813d762119857b/release-v4.4'
  fatal: No names found, cannot describe anything.
  CMake Warning at /home/esp/.espressif/esp-idf-5d813d762119857b/release-v4.4/tools/cmake/build.cmake:479 (message):
    "idf_component.yml" file was found for components:

        /home/esp/Documents/reader/c_libraries/esp_littlefs

    However, the component manager is not enabled.
  Call Stack (most recent call first):
    CMakeLists.txt:14 (idf_build_process)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /home/esp/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace