esp-rs/esp-idf-sys #120: Problem in adding extra component in my workspace

farbodpm, 2022-08-17:
I have tested the extra component in a single project and it works great. but when I am trying to work on a project that have several library that depends on esp-idf-sys. extra components binding doesn't work.

N3xed, 2022-08-17:
Well, only saying that it doesn't work doesn't really help anyone.

What could be the issue? What is the error (if there is one) that you're getting? What could be a potential fix?

Are you aware of the [`ESP_IDF_SYS_ROOT_CRATE` and `CARGO_WORKSPACE_DIR`](https://github.com/esp-rs/esp-idf-sys#build-configuration) environment variables that exist for edge-cases of virtual workspaces and custom workspace setups?

farbodpm, 2022-08-18:
Hi , 
Thanks for your response. I just found out about ESP_IDF_SYS_ROOT_CRATE and I will check it.

N3xed, 2022-08-26:
I'll close this for now, feel free to reopen if this didn't work.

closed 2022-08-26 by N3xed