esp-rs/esp-idf-sys #122: Unable to set Bluetooth Class of Device

maybeinit, 2022-08-19:
I am attempting to rewrite one of the [Bluetooth Espressif examples](https://github.com/espressif/esp-idf/blob/5b1189570025ba027f2ff6c2d91f6ffff3809cc2/examples/bluetooth/bluedroid/classic_bt/bt_hid_mouse_device/main/main.c) in Rust. My understanding is type-safe wrappers for BLE have been made  ([esp-rs/esp-wifi](https://github.com/esp-rs/esp-wifi)) but not for Bluetooth classic, so I'm calling the esp-idf bindings directly.

The issue I'm having is the struct `esp_bt_cod_t` ([espressif docs](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/bluetooth/esp_gap_bt.html#_CPPv412esp_bt_cod_t)) doesn't seem to be generating correctly.

When I try to create it:
```rust
let cod = esp_idf_sys::esp_bt_cod_t {
    major: esp_idf_sys::esp_bt_cod_major_dev_t::ESP_BT_COD_MAJOR_DEV_PERIPHERAL,
};
```

I get the error:
```
error[E0560]: struct `esp_bt_cod_t` has no field named `major`
 --> src/main.rs:8:13
  |
8 |             major: esp_idf_sys::esp_bt_cod_major_dev_t::ESP_BT_COD_MAJOR_DEV_PERIPHERAL,
  |             ^^^^^ `esp_bt_cod_t` does not have this field
  |
  = note: available fields are: `_bitfield_align_1`, `_bitfield_1`
```

---

I used a freshly generated project for `MCU: esp32` from [esp-rs/esp-idf-template](https://github.com/esp-rs/esp-idf-template) and building within vscode's devcontainer.

I updated these dependency versions compared to the template, just to see if they fixed the issue:

```
esp-idf-sys = { version = "0.31.7", features = ["binstart"] }
embuild = "0.30.2"
```
 
And also enabled bluetooth support by adding these to `sdkconfig.defaults`:

```
CONFIG_BT_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=y
CONFIG_BT_HOST=Bluedroid
CONFIG_BT_CONTROLLER=y
CONFIG_BT_CLASSIC_ENABLED=y
CONFIG_BT_HID_ENABLED=y
CONFIG_BT_HID_DEVICE_ENABLED=y
```

---

I don't have enough experience with FFI to know if `esp_bt_cod_t` is just being generated as a stub, or something completely different. Is there a way to fix this? such as adding a missing header to `bindings.h`? Thanks for the help.

N3xed, 2022-08-19:
I think it should be generating correctly but structs with bitfields are a bit different. Since rust has no language construct for bitfields, bindgen generates setter and getter methods on the struct to set them.

The generated bindings are:
```rust
pub struct esp_bt_cod_t {
    pub _bitfield_align_1: [u16; 0],
    pub _bitfield_1: __BindgenBitfieldUnit<[u8; 4usize]>,
}
impl esp_bt_cod_t {
    #[inline]
    pub fn reserved_2(&self) -> u32 { ... }
    #[inline]
    pub fn set_reserved_2(&mut self, val: u32) { ... }
    #[inline]
    pub fn minor(&self) -> u32 { ... }
    #[inline]
    pub fn set_minor(&mut self, val: u32) { ... }
    #[inline]
    pub fn major(&self) -> u32 { ... }
    #[inline]
    pub fn set_major(&mut self, val: u32) { ... }
    #[inline]
    pub fn service(&self) -> u32 { ... }
    #[inline]
    pub fn set_service(&mut self, val: u32) { ... }
    #[inline]
    pub fn reserved_8(&self) -> u32 { ... }
    #[inline]
    pub fn set_reserved_8(&mut self, val: u32) { ... }
    #[inline]
    pub fn new_bitfield_1(
        reserved_2: u32,
        minor: u32,
        major: u32,
        service: u32,
        reserved_8: u32,
    ) -> __BindgenBitfieldUnit<[u8; 4usize]> { ... }
}
```

So, you can either:
- First create the struct with `esp_bt_cod_t::default()` and then use its setters to set the bitfields (`val.set_minor()`, `val.set_major()`, ...).
- Or create the struct with:
  ```rust
    let val = esp_bt_cod_t { 
        _bitfield_1: esp_bt_cod_t::new_bitfield_1(0, minor, major, service, 0),
        ..Default::default()
    }
  ```

The bindings are generated in the `target/<target-triple>/<profile>/build/esp-idf-sys-<hash>/out/bindings.rs`. Unfortunately, rust analyzers jump to definition doesn't work because we `include!()` this file.

By the way, [esp-wifi](https://github.com/esp-rs/esp-wifi) is for bare-metal and not compatible with the `esp-idf-*` crates.

maybeinit, 2022-08-19:
Yup, that was my issue. Thanks for showing me where the bindings are generated, that will come in handy.

Good to know! Thank you for your help!

closed 2022-08-19 by maybeinit