esp-rs/esp-idf-sys #124: Compiling esp-idf-sys v0.31.7 does not go any further

robotoss, 2022-08-23:
Since recently I cannot make a cargo build. There were no such problems before. I have no idea which way to go. Reinstalled toolchain but it did not help.

```
zoxo@Konstantins-MBP Documents % cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cargo
🤷   Project Name : test-project
🔧   Destination: /Users/zoxo/Documents/test-project ...
🔧   Generating template ...
✔ 🤷   MCU · esp32
✔ 🤷   STD support · true
✔ 🤷   ESP-IDF native build version (v4.3.2 = previous stable, v4.4 = stable, mainline = UNSTABLE) · v4.4
? 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? (beware: Dev Containers not avai✔ 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? (beware: Dev Containers not available for esp-idf v4.3.2) · false
[ 1/10]   Done: .cargo/config.toml
[ 2/10]   Done: .cargo
[ 3/10]   Done: .gitignore
[ 4/10]   Done: .vscode
[ 5/10]   Done: Cargo.toml
[ 6/10]   Done: build.rs
[ 7/10]   Done: rust-toolchain.toml
[ 8/10]   Done: sdkconfig.defaults
[ 9/10]   Done: src/main.rs
[10/10]   Done: src
🔧   Moving generated files into: `/Users/zoxo/Documents/test-project`...
✨   Done! New project created /Users/zoxo/Documents/test-project
zoxo@Konstantins-MBP Documents % cd test-project 
zoxo@Konstantins-MBP test-project % cargo build                                                                     
    Updating crates.io index
   Compiling proc-macro2 v1.0.43
   Compiling compiler_builtins v0.1.73
   Compiling quote v1.0.21
   Compiling unicode-ident v1.0.3
   Compiling syn v1.0.99
   Compiling core v0.0.0 (/Users/zoxo/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.132
   Compiling cfg-if v1.0.0
   Compiling memchr v2.5.0
   Compiling serde_derive v1.0.144
   Compiling serde v1.0.144
   Compiling log v0.4.17
   Compiling rustversion v1.0.9
   Compiling regex-syntax v0.6.27
   Compiling serde_json v1.0.85
   Compiling bitflags v1.3.2
   Compiling glob v0.3.0
   Compiling autocfg v1.1.0
   Compiling once_cell v1.13.1
   Compiling libc v0.2.126
   Compiling heck v0.4.0
   Compiling crossbeam-utils v0.8.11
   Compiling itoa v1.0.3
   Compiling anyhow v1.0.62
   Compiling ryu v1.0.11
   Compiling hashbrown v0.12.3
   Compiling os_str_bytes v6.3.0
   Compiling termcolor v1.1.3
   Compiling lazy_static v1.4.0
   Compiling fnv v1.0.7
   Compiling minimal-lexical v0.2.1
   Compiling same-file v1.0.6
   Compiling shlex v1.1.0
   Compiling camino v1.1.1
   Compiling strsim v0.10.0
   Compiling bindgen v0.60.1
   Compiling humantime v2.1.0
   Compiling either v1.8.0
   Compiling semver v1.0.13
   Compiling textwrap v0.15.0
   Compiling cc v1.0.69
   Compiling rustc-hash v1.1.0
   Compiling peeking_take_while v0.1.2
   Compiling memchr v2.4.1
[... 361 lines cut ...]
  WARNING: Git describe was unsuccessful: b''
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.2 at '/Users/zoxo/Documents/test-project/.embuild/espressif/esp-idf/release-v4.4'
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
  WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/zoxo/Documents/test-project/.embuild/espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz.tmp
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
  WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/zoxo/Documents/test-project/.embuild/espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz.tmp
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
  WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/zoxo/Documents/test-project/.embuild/espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz.tmp
  ERROR: Failed to download, and retry count has expired
  Traceback (most recent call last):
    File "/Users/zoxo/Documents/test-project/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 1860, in <module>
      main(sys.argv[1:])
    File "/Users/zoxo/Documents/test-project/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 1856, in main
      action_func(args)
    File "/Users/zoxo/Documents/test-project/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 1470, in action_install
      tool_obj.download(tool_version)
    File "/Users/zoxo/Documents/test-project/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 722, in download
      raise DownloadError()
  __main__.DownloadError
  Error: Could not install esp-idf

  Caused by:
      command '"/Users/zoxo/Documents/test-project/.embuild/espressif/python_env/idf4.4_py3.10_env/bin/python" "/Users/zoxo/Documents/test-project/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py" "--idf-path" "/Users/zoxo/Documents/test-project/.embuild/espressif/esp-idf/release-v4.4" "install" "xtensa-esp32-elf" "cmake" "ninja" "esp32ulp-elf"' exited with non-zero status code 1
```

MabezDev, 2022-08-23:
Seems to be failing to download the tools? Maybe try again, other than that check your connection, it might be being interrupted.

closed 2022-08-23 by MabezDev

woidbaier, 2022-08-23:
I have a different problem with Compiling esp-idf-sys v0.31.7
I also try to build a projekt generated from the [/esp-rs/esp-idf-template](https://github.com/esp-rs/esp-idf-template) for ESP32 on Windows 10 and get the following error:
```
Compiling esp-idf-sys v0.31.7
error: failed to run custom build command for `esp-idf-sys v0.31.7`

Caused by:
  process didn't exit successfully: `C:\Users\woidbaier\CLionProjects\testprojekt\target\debug\build\esp-idf-sys-f6a9336bfafb12a3\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS