esp-rs/esp-idf-sys #125: Partition table sdkconfig option does not change partition table on device

Brad-Hesson, 2022-08-25:
Adding the line `CONFIG_PARTITION_TABLE_TWO_OTA=y` to the sdkconfig.defaults file does not cause the partition table on the device to be changed from the default single app configuration.

It DOES change the sdkconfig file generated in the out directory for the esp-idf-sys crate.  This is the section in question:
```
#
# Partition Table
#
# CONFIG_PARTITION_TABLE_SINGLE_APP is not set
# CONFIG_PARTITION_TABLE_SINGLE_APP_LARGE is not set
CONFIG_PARTITION_TABLE_TWO_OTA=y
# CONFIG_PARTITION_TABLE_CUSTOM is not set
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions_two_ota.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x8000
CONFIG_PARTITION_TABLE_MD5=y
# end of Partition Table
```

and this is what it was before setting the option:
```
#
# Partition Table
#
CONFIG_PARTITION_TABLE_SINGLE_APP=y
# CONFIG_PARTITION_TABLE_SINGLE_APP_LARGE is not set
# CONFIG_PARTITION_TABLE_TWO_OTA is not set
# CONFIG_PARTITION_TABLE_CUSTOM is not set
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions_singleapp.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x8000
CONFIG_PARTITION_TABLE_MD5=y
# end of Partition Table
```
---
Interestingly, setting the option did cause the compile to fail with a build script error referencing the fact that the two ota partition table requires a larger flash size:

```
FAILED: partition_table/partition-table.bin C:/Users/brad-hesson/Desktop/code/esp-idf/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-02cbf1ca7e2d9dfe/out/build/partition_table/partition-table.bin
  cmd.exe /C "cd /D C:\Users\brad-hesson\Desktop\code\esp-idf\target\riscv32imc-esp-espidf\release\build\esp-idf-sys-02cbf1ca7e2d9dfe\out\build\esp-idf\partition_table && C:\Users\brad-hesson\Desktop\code\esp-idf\.embuild\espressif\python_env\idf4.4_py3.10_env\Scripts\python.EXE C:/Users/brad-hesson/Desktop/code/esp-idf/.embuild/espressif/esp-idf/release-v4.4/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 2MB C:/Users/brad-hesson/Desktop/code/esp-idf/.embuild/espressif/esp-idf/release-v4.4/components/partition_table/partitions_two_ota.csv C:/Users/brad-hesson/Desktop/code/esp-idf/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-02cbf1ca7e2d9dfe/out/build/partition_table/partition-table.bin && C:\Users\brad-hesson\Desktop\code\esp-idf\.embuild\espressif\tools\cmake\3.23.1\bin\cmake.exe -E echo "Partition table binary generated. Contents:" && C:\Users\brad-hesson\Desktop\code\esp-idf\.embuild\espressif\tools\cmake\3.23.1\bin\cmake.exe -E echo ******************************************************************************* && C:\Users\brad-hesson\Desktop\code\esp-idf\.embuild\espressif\python_env\idf4.4_py3.10_env\Scripts\python.EXE C:/Users/brad-hesson/Desktop/code/esp-idf/.embuild/espressif/esp-idf/release-v4.4/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 2MB C:/Users/brad-hesson/Desktop/code/esp-idf/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-02cbf1ca7e2d9dfe/out/build/partition_table/partition-table.bin && C:\Users\brad-hesson\Desktop\code\esp-idf\.embuild\espressif\tools\cmake\3.23.1\bin\cmake.exe -E echo *******************************************************************************"
  Partitions tables occupies 3.1MB of flash (3211264 bytes) which does not fit in configured flash size 2MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu.
```

To fix this, I added the config option `CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y` to the sdkconfig.defaults file.  The compile now succeeds and flashes the device (esp32-c3).  However, during boot the device prints out the partition table and sure enough, it still only has the default single app table:
```
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
```

I tried replacing the contents of the partitions_singleapp.csv file with the contents of the partitions_two_ota.csv file in the components section deep down in the .embuild directory.  I thought this would trick the compiler into using the table I want.  However, after a clean and rebuild there was no change.  In fact, deleting the file altogether doesn't even cause a compiler error.  Very suspicous.

MabezDev, 2022-08-25:
What tool are you flashing with? If you are using `espflash` you need to give it the path of the partition table generated by the esp-idf project, else it will just flash the default.

Brad-Hesson, 2022-08-26:
I am using espflash.  Thanks that explains everything!  I just copied the "partitions_two_ota.csv" file into my crate directory and I use it in my espflash command from there.  Wow I wasted a lot of time on trying to figure that out.

axos88, 2022-09-26:
Or use `cargo-espflash`, that takes care of it automagicaly.

MrElectrify, 2022-11-03:
FWIW similarly strange behavior exists for `CONFIG_ESP32_XTAL_FREQ`, no matter what I did it would not take 26, until I specified `CONFIG_ESP32_XTAL_FREQ_26=y`, at which point it seemed to undefine `CONFIG_ESP32_XTAL_FREQ_40`. Maybe this will save somebody else hours.