esp-rs/esp-idf-sys #130: FreeRTOS pdPASS constant not available

Tuupertunut, 2022-09-05:
[ESP-IDF documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv423xTaskCreatePinnedToCore14TaskFunction_tPCKcK8uint32_tPCv11UBaseType_tPC12TaskHandle_tK10BaseType_t) tells that the FreeRTOS function `xTaskCreatePinnedToCore` returns `pdPASS` if it succeeds.

The `pdPASS` constant is defined [here in ESP-IDF headers](https://github.com/espressif/esp-idf/blob/release/v4.4/components/freertos/include/freertos/projdefs.h).

However I can't access this constant from Rust through esp-idf-sys. It is not listed in [the docs](https://esp-rs.github.io/esp-idf-sys/esp_idf_sys/index.html). Some other constants from the same header file can be found in esp-idf-sys, such as `pdFREERTOS_ERRNO_ENOENT`.

This is especially important because the value of `pdPASS` seems to be 1. The C wisdom of "0 means success" doesn't hold here.

N3xed, 2022-09-07:
This is because of the way those macros are defined. We're using [bindgen] to generate the esp-idf bindings and it doesn't support generating constants for macros with casts and generally for function-like macros (see https://github.com/rust-lang/rust-bindgen/issues/316).

We need to either define them ourselves or add special logic using `bindgen::callbacks::ParseCallbacks::func_macro` to generate them ourselves.

ivmarkov, 2023-05-13:
Closing, because for now the philosophy if the project is to just generate raw bindings to whatever is available in the ESP IDF C headers, with very few - if any - customizations. In fact, the one and only exception is the `EspError` struct and its macros.

closed 2023-05-13 by ivmarkov