esp-rs/esp-idf-sys #132: Using esp-idf-sys without downloading anything

TheNeikos, 2022-09-12:
I am a nixos user and would like to use `esp-idf-sys`. However, it insists on downloading things while building (imo a bad idea, it is very unexpected), which I cannot do.

I am using the ESP32-C3 and do not want to use the xtensa toolchain, and instead just use the normal cargo compiler targeting riscv. Currently, this is not possible, as it requires either 'native' or 'pio'. Neither of which I want to use.

Is there any way I can still use your project?

Things that would be required for my usecase:

- No downloads during builds
- List of required build tools (like Clang/cmake) that need to be in $PATH.

Thanks!

MabezDev, 2022-09-12:
Yes, this is possible, and its what I use daily :). esp-idf _is_ a hard dependency on esp-idf-sys so the options are have it downloaded already or download it, the latter being the default for ease of use.

If you download esp-idf manually, run the tool install script once, then all you have to do is source the idf environment ([steps 3-5 from here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#step-2-get-esp-idf)). Once the environment is sourced, esp-idf-sys will use those sources to build esp-idf, instead of downloading it.

`fromenv` shouldn't really be needed, as the build script will detect the environment and switch anyway, but it may be needed to find the environment tools.

TheNeikos, 2022-09-12:
Sadly this still errors, albeit with a different error:

```
❯ cargo check
   Compiling esp-idf-sys v0.31.5
error: failed to run custom build command for `esp-idf-sys v0.31.5`

Caused by:
  process didn't exit successfully: `/home/neikos/projects/embedded/simple_rgb/espressif-trainings/intro/hardware-check/target/debug/build/esp-idf-sys-cf2a3985b2fa3f8f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-changed=/home/neikos/projects/embedded/simple_rgb/espressif-trainings/intro/hardware-check/sdkconfig.defaults
  CMAKE_PREFIX_PATH_riscv32imc-esp-espidf = None
  CMAKE_PREFIX_PATH_riscv32imc_esp_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = Some("/nix/store/4g5fidyk19akng9n91cql60ylrjq93s6-gcc-wrapper-10.3.0:/nix/store/95b5bf92ck38jp5lyih5kw9f2glx4gkm-binutils-wrapper-2.35.1:/nix/store/w9c6m0qy0c2hc77nggbcy4hq9dys8fw7-rust-default-1.65.0-nightly-2022-09-04:/nix/store/gsl84cnlwz1nkn5bgkm88m3lz884fgib-espflash-1.6.0:/nix/store/zv9i80xc9j2csz6jcv3appp8vxhjvaa6-ldproxy-0.3.2:/nix/store/mf3vixxfdvs9vn9fzz24hbjq3fq5pgpl-python3.9-pip-21.1.3:/nix/store/8dxxjbiyxwkvh53q5kh6nydla2anacgi-python3-3.9.6:/nix/store/jsx4mm56igrd6pmh7hs3asfrgg4kd1q2-python3.9-virtualenv-20.8.1:/nix/store/s17aj48g92cyz17fmbwn7yqnwk1ybbz2-python3.9-backports-entry-points-selectable-1.1.0:/nix/store/8smrlirz30mg91xpxsdvjirvnzzrahid-python3.9-distlib-0.3.3:/nix/store/y8f3i57ndxlsr2kmy396q0vn0dhz8wvr-python3.9-filelock-3.3.0:/nix/store/arkwlnr8pjbzpf60125yyzfgxywa3f7i-python3.9-platformdirs-2.4.0:/nix/store/7ixnxb28f4sg2c08a1pnp3yxdb0h48cs-python3.9-six-1.16.0:/nix/store/k39dr5rzpz6ifwif7zdnf7kf3w309gp1-cargo-edit-0.8.0:/nix/store/9xz9aga97wc35y5dmlkya8nyiky9014r-esp32c3-toolchain-2021r2-patch3:/nix/store/cp482m3fyv64rcbl5p7fzzpmigmly0fz-esp-idf-v4.4.1:/nix/store/njg99sqcyglfsg2lf1p755r2wivrrsww-esptool-3.2:/nix/store/dgp3mvkfnf6v0bz0qbh94cxqhi1hld4i-python3.9-pyserial-3.5:/nix/store/ryhpb1y18qf4gcy2m5r2d2jbf05w415p-python3.9-pyaes-1.6.1:/nix/store/h14y9gs3m48n74m58bi5p4lkswy7yy2a-python3.9-ecdsa-0.17.0:/nix/store/llgs201w4ma372xb574inmzji1phzr01-python3.9-reedsolo-1.5.4:/nix/store/khf8kwnc2bwhb5vxjmxkx7jpvyc7c6iw-python3.9-bitstring-3.1.9:/nix/store/7imvz43yrh3a22ajqkabz667hq2zrrz4-python3.9-cryptography-3.4.8-dev:/nix/store/7n6fj45g1q6ilckx332j6rwv0q1mbcnj-python3.9-packaging-20.9:/nix/store/40r94pxfjm049w38zs1dn5mw937mgdwf-python3.9-pyparsing-2.4.7:/nix/store/4a198098n5wm7wnn4ni6z46y8bnpfjn7-python3.9-cffi-1.14.6-dev:/nix/store/4flpmcvw0p795bx1s9lnam7kyl7qhviv-python3.9-pycparser-2.20:/nix/store/wf8vlq8gibp76vp816yw0fxjlc5ldphs-python3.9-cffi-1.14.6:/nix/store/09gdhd7z0cjrdd4m9czpjjbnx815hdlb-python3.9-cryptography-3.4.8:/nix/store/zhf7xxp7k1605wirn29kylg1x2bl4pq3-git-2.33.1:/nix/store/8k89kjnnns1j72mfm1i2xsz2dvpy9kpl-wget-1.21.2:/nix/store/4xzvmpwx8lhjgiypdkffl4nzg6pb4lcb-gnumake-4.3:/nix/store/k64ipsghczdyf0pirbvqmjk4idkh6860-flex-2.6.4:/nix/store/2ysk871hamhwb8npzrycpbpdp77dkdws-gnum4-1.4.19:/nix/store/5nhdzkfiv2cin7hlhq88i8zjhgbi82hn-bison-3.7.6:/nix/store/sind0shkjaywcffa5zkypf5411z8w6c5-gperf-3.1:/nix/store/a4fd1rmx1fjx8ashzwaqwvmwpfbm450l-pkg-config-wrapper-0.29.2:/nix/store/sg4gfdp6ahmz8341zh8r3f2s1hylmy47-cmake-3.21.2:/nix/store/vvdy4xg9gd2fgh4rb4i8sy09s08gnr8s-ninja-1.10.2:/nix/store/mscxycfi44z0jwvyz71yv11c3fwbv3nn-ncurses-6.2-abi5-compat-dev:/nix/store/5qjm3h2nynq925ksqw3z31ci3jpf0z75-ncurses-6.2-abi5-compat:/nix/store/p4axvsdpc70w7d9g4kkl3d25swh71fgd-patchelf-0.13")
  CMAKE_riscv32imc-esp-espidf = None
  CMAKE_riscv32imc_esp_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/home/neikos/projects/embedded/simple_rgb/espressif-trainings/intro/hardware-check/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-32e02a0960456f2b/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/nix/store/cp482m3fyv64rcbl5p7fzzpmigmly0fz-esp-idf-v4.4.1/tools/cmake/toolchain-esp32c3.cmake" "-DCMAKE_BUILD_TYPE=" "-DCMAKE_INSTALL_PREFIX=/home/neikos/projects/embedded/simple_rgb/espressif-trainings/intro/hardware-check/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-32e02a0960456f2b/out" "-DCMAKE_C_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_CXX_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany"
  -- Checking Python dependencies...
  Python requirements from /nix/store/cp482m3fyv64rcbl5p7fzzpmigmly0fz-esp-idf-v4.4.1/requirements.txt are satisfied.
  -- Project sdkconfig file /home/neikos/projects/embedded/simple_rgb/espressif-trainings/intro/hardware-check/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-32e02a0960456f2b/out/sdkconfig
  Loading defaults file /home/neikos/projects/embedded/simple_rgb/espressif-trainings/intro/hardware-check/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-32e02a0960456f2b/out/gen-sdkconfig.defaults...
  Loading defaults file /home/neikos/projects/embedded/simple_rgb/espressif-trainings/intro/hardware-check/sdkconfig.defaults...
  -- App "libespidf" version: 1
[... cut ...]
	configure_file("${GIT_DIR}/HEAD" "/home/neikos/projects/embedded/simple_rgb/espressif-trainings/intro/hardware-check/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-32e02a0960456f2b/out/build/CMakeFiles/git-data/head-ref" COPYONLY)
endif()

if(NOT HEAD_HASH)
	file(READ "/home/neikos/projects/embedded/simple_rgb/espressif-trainings/intro/hardware-check/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-32e02a0960456f2b/out/build/CMakeFiles/git-data/head-ref" HEAD_HASH LIMIT 1024)
	string(STRIP "${HEAD_HASH}" HEAD_HASH)
endif()
```

This doesn't seem to come from `esp-idf-sys` though :thinking:

thiskappaisgrey, 2023-02-24:
I made a `nix shell` for this: https://github.com/thiskappaisgrey/nixpkgs-esp-dev-rust/blob/rust/shells/esp32s2-idf-rust.nix . `esp-idf-sys` still expects `esp-sys` to be a git repo, so I just let it do what it wants and clone the repo. Using nix, I had to provide the necessary libraries that `esp-idf-sys` needs. I'm just posting this in case anyone (who is using nix) comes  across this in the future. Thanks.

knarkzel, 2023-03-20:
> I made a `nix shell` for this: https://github.com/thiskappaisgrey/nixpkgs-esp-dev-rust/blob/rust/shells/esp32s2-idf-rust.nix . `esp-idf-sys` still expects `esp-sys` to be a git repo, so I just let [...]

Did you anything like this? When I try it with [my repository]() I get following:

```
    Updating crates.io index
   Compiling compiler_builtins v0.1.84
   Compiling proc-macro2 v1.0.52
   Compiling quote v1.0.26
   Compiling libc v0.2.140
   Compiling memchr v2.5.0
   Compiling serde_derive v1.0.157
   Compiling serde v1.0.157
   Compiling log v0.4.17
   Compiling syn v1.0.109
   Compiling libc v0.2.138
   Compiling clang-sys v1.6.0
   Compiling rustversion v1.0.12
error: linking with `clang` failed: exit status: 1
  |
  = note: "clang" "-m64" "/tmp/nix-shell.ickrxM/rustc0sxIWy/symbols.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.0.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.1.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.10.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.11.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.12.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.13.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.14.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.15.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.2.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.3.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.4.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.5.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.6.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.7.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.8.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.build_script_build.e8e42ab9-cgu.9.rcgu.o" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/build/log-c5879a61b5f04c22/build_script_build-c5879a61b5f04c22.4focamokwqw4zbv9.rcgu.o" "-Wl,--as-needed" "-L" "/home/odd/source/nix/temp/esp32-hello-world/target/debug/deps" "-L" "/nix/store/a04qn9lgfsm3k6hsap5cncbjmn53zaqr-rust-xtensa-1.67.0.0/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libstd-9bf22f71a5b69462.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libpanic_unwind-2973da644a9aa9a4.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libobject-28811e4da7b8e05d.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libmemchr-2fcb221e9a6784d2.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libaddr2line-438169fb9a4c1933.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libgimli-c82c22e1c759d420.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/librustc_demangle-51090e4c4262203c.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libstd_detect-e09e07f8305f9ee2.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libhashbrown-21d2b358dfc61dfb.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libminiz_oxide-4e7ce016f3ea8f9b.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/rustlib/x86_64-unknown-linux-gnu/lib/libadler-1eb3d2f297872420.rlib" "/nix/store/xc0cvvbrx8ykwlaaarv5ji607n1gkdf6-rustc-1.67.0.0-x86_64-linux/lib/ru [... cut ...] 0d0ee8986ba586" "-Wl,--gc-sections" "-pie" "-Wl,-zrelro,-znow" "-nodefaultlibs"
  = note: /nix/store/nydwzhllkq0a21dny69zdjczh6v275lb-binutils-2.40/bin/ld: cannot find crtbeginS.o: No such file or directory
          /nix/store/nydwzhllkq0a21dny69zdjczh6v275lb-binutils-2.40/bin/ld: cannot find -lgcc: No such file or directory
          clang-15-unwrapped: error: linker command failed with exit code 1 (use -v to see invocation)
          

error: could not compile `libc` due to previous error
```

Vollbrecht, 2024-06-21:
closing in favor of #184

closed 2024-06-21 by Vollbrecht