esp-rs/esp-idf-sys #134: Not rebuilding esp-idf and esp-idf-sys if using IDF_PATH, and changing the path

axos88, 2022-09-20:
workaround is to use cargo `clean -p esp-idf-sys`, but that will rebuild the entire esp-idf for every build.

axos88, 2022-09-26:
Repro:
  - check out esp-idf in a separate directory
  - add ESP_IDF=<path-to-checked-out-repo>
  - cargo build
  - change something in esp-idf
  - cargo build

Expect to rebuild esp-idf-sys (and esp-idf), but that does not happen.

ivmarkov, 2022-09-30:
This is a rare use case and I don't think there is an easy fix, so it will probably stay open for a while. #135 and #137 are more important.

axos88, 2022-10-01:
No worries. At least it's now documented :)

Vollbrecht, 2024-06-21:
Autodetection would be nice but it would be a lot of effort for a niche usecase. If anybody wants to tackle this feel free to work on it but i will close this issue for now ans not planned.

closed 2024-06-21 by Vollbrecht