esp-rs/esp-idf-sys #137: Build error with mismatched types on ESP_IDF_TIME64_CHECK

Seaward233, 2022-09-29:
## Versions
esp-idf-sys version: 0.31.9
embuild version: 0.30.4
esp-idf version: branch:release/v5.0

## Details
When building without espidf_time64 rustflag, build error with mismatched types on ESP_IDF_TIME64_CHECK and ESP_IDF_TIME64_CHECK_LIBC.
With espidf_time64 rustflag, build error with mismatched types just on ESP_IDF_TIME64_CHECK.
Between each build, ```cargo clean``` is done.

## Console output
```
error[E0308]: mismatched types
  --> /home/ansme/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.9/src/lib.rs:31:62
   |
31 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-sys` due to previous error
```

## config.toml
```
[build]
# Uncomment the relevant target for your chip here (ESP32, ESP32-S2, ESP32-S3 or ESP32-C3)
#target = "xtensa-esp32-espidf"
#target = "xtensa-esp32s2-espidf"
#target = "xtensa-esp32s3-espidf"
target = "riscv32imc-esp-espidf"
# rustflags = ["--cfg", "espidf_time64"]

[target.xtensa-esp32-espidf]
linker = "ldproxy"

[target.xtensa-esp32s2-espidf]
linker = "ldproxy"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"

# Future - necessary for the experimental "native build" of esp-idf-sys with ESP32C3
# See also https://github.com/ivmarkov/embuild/issues/16
# rustflags = ["-C", "default-linker-libraries"]

rustflags = ["-C", "default-linker-libraries", "--cfg", "espidf_time64"]

[unstable]

build-std = ["std", "panic_abort"]
#build-std-features = ["panic_immediate_abort"] # Required for older ESP-IDF versions without a realpath implementation

[env]
# Note: these variables are not used when using pio builder
# Enables the esp-idf-sys "native" build feature (`cargo build --features native`) to build against ESP-IDF stable (v4.4)
# ESP_IDF_VERSION = { value = "branch:release/v4.4" }
# Enables the esp-idf-sys "native" build feature (`cargo build --features native`) to build against ESP-IDF master (mainline)
# ESP_IDF_VERSION = { value = "master" }

ESP_IDF_TOOLS_INSTALL_DIR = { value = "global" }
ESP_IDF_VERSION = { value = "branch:release/v5.0" }
```
## Cargo.toml
```
[package]
name = "first-run"
version = "0.1.0"
authors = ["ansme"]
edition = "2021"
resolver = "2"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
pio = ["esp-idf-sys/pio"]

[dependencies]
esp-idf-sys = { version = "0.31.9", features = ["binstart"] }

[build-dependencies]
embuild = "0.30.4"
anyhow = "1"
```

ivmarkov, 2022-09-30:
Using ESP IDF 5 needs a very small upstream change to Rust STD where STD's `Cargo.toml` file is updated to use the latest version of the Rust libc crate which does have support for the `espidf_time64` flag.  If nobody does it for their own reasons in the next few days, I'll open a PR for this in Rust upstream.

In the meantime, unless you absolutely depend on ESP IDF 5, you can switch to ESP IDF 4.4, which is the "most stable" release anyway. The changes from Rust POV between ESP IDF 5 and 4.4 are negligible.

axos88, 2022-10-01:
Adding `libc = { git = "https://github.com/rust-lang/libc.git" }` to cargo patch.crates-io does not seem to work (on latest nightly, 09-30), but I think it should, since https://github.com/rust-lang/libc/pull/2913 has been merged about two weeks ago?

ivmarkov, 2022-11-03:
The upstream change got merged, closing.

closed 2022-11-03 by ivmarkov

axos88, 2022-11-03:
@ivmarkov , can you link the upstream change that was necessary for this? I'm curious as to what was necessary besides https://github.com/rust-lang/libc/pull/2913

ivmarkov, 2022-11-03:
This: https://github.com/rust-lang/rust/blob/master/library/std/Cargo.toml#L18

Basically, Rust stdlib itself had to be updated to depend on a newer version of the libc crate. Cargo is not really following the package versioning resolution across your own crates *and* STD.