esp-rs/esp-idf-sys #140: bindings not generated for mbedtls/asn1write.h

axos88, 2022-10-13:
Seems it is left out of the build for some reason?

axos88, 2022-10-13:
Adding #include "asn1write.h" to say "x509.h" includes it though. So it seems like whatever is the entry point for bindgen does not include asn1write.h

N3xed, 2022-10-14:
[`bindings.h`](https://github.com/esp-rs/esp-idf-sys/blob/master/src/include/esp-idf/bindings.h) is the entry-point for bindgen. For everything that is included there bindings will be generated.

You can also add your own project-specific header to generate extra bindings to the esp-idf with a header only [`extra component`](https://github.com/esp-rs/esp-idf-sys#extra-esp-idf-components).

axos88, 2022-10-16:
@N3xed , tried this, but it tries to look for the header files in my own project directory, and not in esp-idf:

`Cargo.toml`:

```toml

[[package.metadata.esp-idf-sys.extra_components]]
bindings_header = "mbedtls/oid.h"

[[package.metadata.esp-idf-sys.extra_components]]
bindings_header = "asn1write.h"

[[package.metadata.esp-idf-sys.extra_components]]
bindings_header = "x509csr.h"

```

```
  Error: extra components C header file 'XXX/project-name/mbedtls/oid.h' specified by crate '/XXX/project-name' does not exist

```

N3xed, 2022-10-16:
> @N3xed , tried this, but it tries to look for the header files in my own project directory, and not in esp-idf:
>
> `Cargo.toml`:
>
> ```toml
> [[package.metadata.esp-idf-sys.extra_components]] [...]

That's not how that feature works, make your own header file in your project directory and include these headers there. Then you can use your header in an extra component.

PR #143 (2022-10-18): Clarify how bindings_header works
Fixes #140