esp-rs/esp-idf-sys #142: TOOLS_WORKSPACE_INSTALL_DIR should be configurable

liebman, 2022-10-17:
Currently its hard coded to ".embuild".  It would be nice if it were configurable via an environment variable.

Maybe my use case is not common.  I have the project on a shared filesystem on multiple systems.   Manually managing the the IDF and tools installs across many systems will be problematic when it's time to upgrade IDF versions or tools.

N3xed, 2022-10-18:
I may not be understanding your issue, but the (workspace) install dir is already configurable. If you set [`ESP_IDF_TOOLS_INSTALL_DIR`](https://github.com/esp-rs/esp-idf-sys#esp_idf_tools_install_dir-esp_idf_tools_install_dir) to `custom:<path>` you can configure the name and location of the install dir (if its a relative path, it's relative to the workspace dir).

I.e. just setting it to `custom:.my_special_folder` would install the tools in `<workspace-dir>/.my_special_folder` (instead of `<workspace-dir>/.embuild/<builder-name>`).

liebman, 2022-10-18:
This is perfect! Thank you!

closed 2022-10-18 by liebman