esp-rs/esp-idf-sys #149: Compiling on M1 Max MacBook Pro (via Docker) takes a _really_ long time?

bsodmike, 2022-11-03:
Hi all,

Has anyone else come across this by any chance?

Thanks!

ivmarkov, 2022-11-03:
Is this just the first time, or also second and third time?

bsodmike, 2022-11-03:
I think Docker might be the culprit. With all the IO going on from my
session today I noticed it got slower and slower with copying and
downloading files.

I will try a Docker reboot and see if it helps.

In any case it workflow wasn’t great so I’ve decided to run a Docker
instance in one of my VMs (x86 arch) and run Nvim through tmux so that
rust-analyzer also runs in the VM.

Thanks!

bsodmike, 2022-11-03:
I just tried my x86 VM and a clean release compiled under a minute (24 vcpu/64GB RAM/Dual Xeons Dell r740xd).  Will close this as I don't think non-x86 would be supported.

closed 2022-11-03 by bsodmike