esp-rs/esp-idf-sys #151: esp_idf_sys::esp_log_level_set did not work as expected

elotom, 2022-11-07:
It looks like the log level can't be changed during runtime with esp_log_level_set. I'm using ESP32-S3.
Code that I'm using in main to replicate the problem:

```rust
// Default log level is 3 - Info
    info!("Info level");
    error!("Error level");
    unsafe {
        let level = esp_idf_sys::esp_log_level_get("*".as_ptr() as *const i8);
        dbg!(level);
        esp_idf_sys::esp_log_level_set("*".as_ptr() as *const i8, 1);
        let level = esp_idf_sys::esp_log_level_get("*".as_ptr() as *const i8);
        dbg!(level);
    }
    info!("Info level");
    error!("Error level");
```

Terminal output:
[image]

I was expecting that only the Error level will be printed after setting the log level to 1(Error).

jendakol, 2022-11-07:
FYI I'm struggling the same way, trying to set logging to a `DEBUG`.

ivmarkov, 2022-11-10:
Calling `esp_log_level_set` is necessary, but not sufficient. Check this part of the ESP IDF documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html

closed 2022-11-10 by ivmarkov