esp-rs/esp-idf-sys #154: given the same frequency, pwm period decreases (speeds up) when the resolution decreases

tinoadams, 2022-11-20:
I first tried to clarify my understanding on how PWM should work on StackOverflow via the following post:
https://stackoverflow.com/questions/74474847/given-the-same-frequency-why-does-the-pwm-period-decrease-when-the-resolution-d

I've got an ESP32-C3 and the following Rust code. Retaining the frequency but decreasing the resolution (from 13 to 8 bits), I expected the granularity of the pulse width to decrease but it unfortunately also decreases the period (speeds up). I expected the period to stay the same as I thought it is fully determined by the frequency and not by the resolution.

```rust
let pwm_pin = peripherals.pins.gpio10.into_output().unwrap();
let config = TimerConfig::default().frequency((50.Hz()).into()).resolution(Resolution::Bits13);
let timer = Timer::new(peripherals.ledc.timer0, &config)?;
let mut channel = Channel::new(peripherals.ledc.channel0, &timer, pwm_pin)?;

// set duty cycle to 50%
let percent = 50;
let duty = ((channel.get_max_duty() as f64 / 100.0) * percent as f64) as u32;
channel.set_duty(duty);
```

Is this expected behavior and how would I go about calculating the period from any given frequency and resolution pair?

tinoadams, 2022-11-21:
Interesting. Thank you for you detailed explanation. I kept my question deliberately brief, maybe a little too brief.

I am driving a servo with the code above and at 50hz and 13 bits resolution, 50% duty cycle gives me the expected cycle time of 20ms (1/50) and a 10ms pulse width (50% duty cycle).

[image]

However, keeping the frequency and duty cycle calculation the same but changing the code above to 8 bit resolution the  cycle time changes to 250us and 125us pulse width.

[image]

I will have a look through the technically reference manual as you suggested to try and gain a better understanding on how the PWM signal is generated.

Vollbrecht, 2022-11-21:
Question becomes why do you want to change the resolution on an running pwm? Usually you want to change the Frequency ( changing the upper limit) or the duty cycle ( changing the 2nd limit) or both together. The only time you need it if you want to span a really big frequency range with your pwm. To get back to what you see. A simple reverse calculation: 8bit -> 255 = 250us |  x = 20ms  =>  x = y * 20ms/250us => y = 80 & x = 20400. Because 20400 > 13Bits our initial assumption that 255 = 250us was not true but already a bit lower. but its still a factor of 80 down. the problem with low frequency's compared to high input clocks is you either need a high resolution counter ( eg many bits) or divide down your clock source further otherwise you fill your register to fast. ( counting to 255 is faster than counting to 20400 with the same counting speed) but sometimes you are hardware limited and can not reach all desired output speeds ( you are already dividing down your input clock by max divider). So if it hits a hardware limit and cant translate it it has to for example rise the frequency respectively.

tinoadams, 2022-11-21:
Makes total sense. Thank you for taking the time to spell it out for me.

ivmarkov, 2023-02-01:
Closing as it appears that the issue had been resolved.

closed 2023-02-01 by ivmarkov