esp-rs/esp-idf-sys #161: Build with custom certificate bundle fails due to missing parameter for gen_crt_bundle.py

taunusflieger, 2023-01-08:
Using IDF 5.0 and esp-idf-sys v0.32.1

I'm trying to connect to AWS IoT. AWS is using mutual TLS authentication. I don't want the full certificate bundle as it increases the size of my already large image - need only the Amazon CA. I'm attaching the Amazon Root CA as custom certificate bundle. I've verified that this approach works for C based IDF projects.

sdkconfig.defaults:

```
# Certificate Bundle
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=y
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_NONE=y
CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE=y
CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE_PATH="../esp-aws-iot/cert"
Unfortunately it looks like that this build step fails as the specified path parameter is not provided to gen_crt_[bundle.py](http://bundle.py/)
```
Building the solution creates the following error. The build process doesn't provide the path to the cert directory to the ```gen_crt_bundle.py``` tool.
```
  [1/197] Generating x509_crt_bundle
  FAILED: esp-idf/mbedtls/x509_crt_bundle 
  cd /home/michael/src/embedded-rust/esp-aws-iot/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-f9a37df0dfb0796b/out/build/esp-idf/mbedtls && /home/michael/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/michael/.espressif/esp-idf-ae062fbba3ded0aa/release-v5.0/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input -q
  usage: gen_crt_bundle.py [-h] [--quiet] --input INPUT [INPUT ...]
                           [--filter FILTER]
  gen_crt_bundle.py: error: argument --input/-i: expected at least one argument
  [2/197] Performing build step for 'bootloader'
  [0/1] Re-running CMake...
```

N3xed, 2023-01-10:
Paths in the `sdkconfig`/`sdkconfig.defaults` are a bit special, since the esp-idf is built as a dummy project in a separate folder. It should be relative to the `target/<target-triple>/<build profile>/build/esp-idf-sys-<hash>/out` folder because all settings from the `sdkconfig`/`sdkconfig.defaults` are merged into the `target/<target-triple>/<build profile>/build/esp-idf-sys-<hash>/out/sdkconfig` file.

taunusflieger, 2023-01-11:
Thanks, but I don't think so - the error message doesn't list an invalid path / file not found, but it complains that no argument is given for --input/-i

igrr, 2023-01-11:
That is probably because of
```cmake
        get_filename_component(custom_bundle_path
                        ${CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE_PATH} ABSOLUTE BASE_DIR "${project_dir}")
        list(APPEND crt_paths ${custom_bundle_path})
        
        ...
        
        list(APPEND args --input ${crt_paths} -q)
```

`custom_bundle_path` ends up empty because the path doesn't exist, so `crt_paths` is also empty, and the script is called with just the `--input` argument.

I do agree that this looks like a file-not-found error, although sorry about the bad diagnostics in this case!

taunusflieger, 2023-01-11:
okay that makes sense - thanks @igrr and @N3xed. I will as soon as I'm able to build esp-idf-sys which is currently broken for esp-idf v5 due to a branch name change in esp-idf ...

taunusflieger, 2023-01-11:
Thanks that was the solution - path relative to target/...

closed 2023-01-11 by taunusflieger