esp-rs/esp-idf-sys #163: Build error "Failed to list cmake-file-api reply directory"

hawkw, 2023-01-11:
Attempting to build the `esp-idf-template` hello world project for the `esp32c3` fails for me with the following message:

```
  [1009/1016] Linking C static library esp-idf/libsodium/liblibsodium.a
  [1010/1016] Building CXX object esp-idf/asio/CMakeFiles/__idf_asio.dir/asio/asio/src/asio.cpp.obj
  [1011/1016] Linking C static library esp-idf/asio/libasio.a
  [1012/1016] Generating ld/sections.ld
  [1013/1016] Building C object CMakeFiles/libespidf.elf.dir/main.c.obj
  [1014/1016] Linking C executable libespidf.elf
  [1015/1016] Generating binary image from built executable
  esptool.py v3.3.2
  Creating esp32c3 image...
  Merged 1 ELF section
  Successfully created esp32c3 image.
  Generated /home/esp/eclss/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-c333f47f20093269/out/build/libespidf.bin
  [1016/1016] cd /home/esp/eclss/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-c333f47f20093269/out/build/esp-idf/esptool_py && /home/esp/.espressif/python_env/idf4.4_py3.9_env/bin/python /home/esp/.espressif/frameworks/esp-idf-release-v4.4/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/esp/eclss/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-c333f47f20093269/out/build/partition_table/partition-table.bin /home/esp/eclss/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-c333f47f20093269/out/build/libespidf.bin
  libespidf.bin binary size 0x22c40 bytes. Smallest app partition is 0x100000 bytes. 0xdd3c0 bytes (86%) free.
  cargo:root=/home/esp/eclss/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-c333f47f20093269/out

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Branch(
                  "release/v4.4",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/home/esp/.espressif/frameworks/esp-idf-release-v4.4",
          ),
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using activated esp-idf v4.4.3 environment at '/home/esp/.espressif/frameworks/esp-idf-release-v4.4'
  CMake Warning at /home/esp/.espressif/frameworks/esp-idf-release-v4.4/tools/cmake/build.cmake:477 (message):
    "idf_component.yml" file was found for components:

    
        /home/esp/.espressif/frameworks/esp-idf-release-v4.4/components/freemodbus

    However, the component manager is not enabled.
  Call Stack (most recent call first):
    CMakeLists.txt:16 (idf_build_process)

  Error: Failed to list cmake-file-api reply directory

  Caused by:
      No such file or directory (os error 2)
```

I'm using the devcontainer distributed with `esp-idf-template` (docker image `idf-rust:esp32c3_v4.4_1.64.0.0`), so the build environment is all contained in that docker image version. This error should (hopefully) be reproducible using the devcontainer and building for `esp32c3`. Let me know if there's any additional information I can provide!

MabezDev, 2023-01-11:
I can't reproduce this on my machine, but I think switching to the released (tagged) version on esp-idf 4.4 _should_ fix this.

```diff
diff --git a/.cargo/config.toml b/.cargo/config.toml
index 4d039c5..d194237 100644
--- a/.cargo/config.toml
+++ b/.cargo/config.toml
@@ -35,6 +35,6 @@ build-std = ["std", "panic_abort"]
 [env]
 # Note: these variables are not used when using pio builder (`cargo build --features pio`)
 # Builds against ESP-IDF stable (v4.4)
-ESP_IDF_VERSION = "release/v4.4"
+ESP_IDF_VERSION = "tag:v4.4.3"
 # Builds against ESP-IDF master (mainline)
 #ESP_IDF_VERSION = "master"
```

You can read more about esp-idf-sys configuration here: https://github.com/esp-rs/esp-idf-sys#build-configuration.

I imagine some broken commits made it onto the release/v4.4 branch just as you happened to clone it :D. Historically we needed "early" access to esp-idf changes as we were porting the Rust standard library to it, but now its fairly stable. I'll also PR a fix to the repo to use tags by default.

hawkw, 2023-01-11:
Hmm, I still get the same error after switching to use the tag. It's possible there's something else I've messed up, though.

MabezDev, 2023-01-11:
Hmm, maybe try removing the `.embuild` directory, running `cargo clean` and build again.

MabezDev, 2023-01-12:
I setup docker on my other machine this morning to try this again, still working for me. One thing I thought about is that you're not using the devcontainer feature, just the image directly. To fully activate the environment you have to `source /home/esp/esp-export.sh`, though this should be handled by the image anyway ([last command in the docker file](https://hub.docker.com/layers/espressif/idf-rust/esp32c3_v4.4_latest/images/sha256-6d2d0cd9e95a1875b3b8b203cb29f94a7232a7037ec1baabe28b51ec38470e8b?context=explore)), but could be worth a shot.

SergioGasquez, 2023-01-12:
Just gave it a try and it also works fine for me. Here its what I did:
1. `cargo generate https://github.com/esp-rs/esp-idf-template cargo`
  - `esp32c3`, devcontainer support, std support,v4.4
2. Edited `.devcontainer/devcontainer.json`:
```json
  "image": "docker.io/espressif/idf-rust:esp32c3_v4.4_1.64.0.0",
  // "build": {
  //   "dockerfile": "Dockerfile",
  //   "args": {
  //     "CONTAINER_USER": "esp",
  //     "CONTAINER_GROUP": "esp",
  //     "ESP_IDF_VERSION":"release/v4.4",
  //     "ESP_BOARD": "esp32c3"
  //   }
  // },
```
3. Launched the devcontainer and I was able to build the project with no issue.

If [we change the esp-idf version to use tag `v4.4.3`](https://github.com/esp-rs/esp-idf-sys/issues/163#issuecomment-1379554289), then we would need to use the `build` property instead of the `image`, as the dockerhub image uses `release/v4.4'  branch:
```json
  // "image": "docker.io/espressif/idf-rust:esp32c3_v4.4_1.64.0.0",
  "build": {
    "dockerfile": "Dockerfile",
    "args": {
      "CONTAINER_USER": "esp",
      "CONTAINER_GROUP": "esp",
      "ESP_IDF_VERSION":"v4.4.3",
      "ESP_BOARD": "esp32c3"
    }
  },
```

Maybe try deleting your docker `idf-rust:esp32c3_v4.4_1.64.0.0` image so its downloaded again with the current status of the `release/v4.4` branch.

hawkw, 2023-01-12:
Thanks, I'll try again with a fresh template project, and delete all the cached Docker images, and report back!

InRiPa, 2023-01-13:
I believe, I'm having the same or similar issue. 
 I used:
`cargo generate https://github.com/esp-rs/esp-idf-template cargo`
chose `esp32` and using `std` and activated `devcontainer`

Within the devcontainer I can successfully build the program. But using the host system, I believe I get the same error reagrding *modbus*.

Log from container [here](https://privatebin.net/?1e13f71822cb32f2#D782bphvuZcHc4PoZgb28o6N8k9GMqWX7uLt4XAJY439) 

log from host [here](https://privatebin.net/?19680717ff5dc102#AP9PJ1ejaKPU7AMpDuDnFF21NpdWjHsA2qdT3FbubUyT)

The log from the host is quite more verbose. 
I'm using arch linux with (I believe) the most recent versions.

In the container:
rustup 1.25.1 (bb60b1e89 2022-07-12)
cargo 1.64.0-nightly (387270bc7 2022-09-16)

host: 
rustup 1.25.1 (2022-11-01)
cargo 1.66.0-nightly (d65d197ad 2022-11-15)

The following might be the relevant part. 

```
  [1045/1046] Generating binary image from built executable
  esptool.py v3.3.2
  Creating esp32 image...
  Merged 2 ELF sections
  Successfully created esp32 image.
  Generated /home/user/project/Xtensa/blink-esp-idf-template/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f3c44b8bdc76f804/out/build/libespidf.bin
  [1046/1046] cd /home/user/project/Xtensa/blink-esp-idf-template/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f3c44b8bdc76f804/out/build/esp-idf/esptool_py && /home/user/project/Xtensa/blink-esp-idf-template/.embuild/espressif/python_env/idf4.4_py3.10_env/bin/python /home/user/project/Xtensa/blink-esp-idf-template/.embuild/espressif/esp-idf/release-v4.4/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/user/project/Xtensa/blink-esp-idf-template/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f3c44b8bdc76f804/out/build/partition_table/partition-table.bin /home/user/project/Xtensa/blink-esp-idf-template/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f3c44b8bdc76f804/out/build/libespidf.bin
  libespidf.bin binary size 0x271a0 bytes. Smallest app partition is 0x100000 bytes. 0xd8e60 bytes (85%) free.
  cargo:root=/home/user/project/Xtensa/blink-esp-idf-template/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f3c44b8bdc76f804/out
  cargo:rerun-if-changed=/home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.12/src/include/esp-idf/bindings.h
  cargo:rustc-env=EMBUILD_GENERATED_BINDINGS_FILE=/home/user/project/Xtensa/blink-esp-idf-template/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f3c44b8bdc76f804/out/bindings.rs

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Branch(
                  "release/v4.4",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Branch("release/v4.4") }
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.3 at '/home/user/project/Xtensa/blink-esp-idf-template/.embuild/espressif/esp-idf/release-v4.4'
  fatal: No names found, cannot describe anything.
  CMake Warning at /home/user/project/Xtensa/blink-esp-idf-template/.embuild/espressif/esp-idf/release-v4.4/tools/cmake/build.cmake:477 (message):
    "idf_component.yml" file was found for components:

    
    	/home/user/project/Xtensa/blink-esp-idf-template/.embuild/espressif/esp-idf/release-v4.4/components/freemodbus

    However, the component manager is not enabled.
  Call Stack (most recent call first):
    CMakeLists.txt:16 (idf_build_process)

  Built components: esp_ringbuf, efuse, esp_ipc, driver, esp_pm, mbedtls, bootloader, esptool_py, partition_table, app_update, bootloader_support, spi_flash, nvs_flash, pthread, esp_gdbstub, espcoredump, esp_phy, esp_system, esp_rom, hal, vfs, esp_eth, tcpip_adapter, esp_netif, esp_event, wpa_supplicant, esp_wifi, ieee802154, console, openthread, lwip, log, heap, soc, esp_hw_support, xtensa, esp32, esp_common, esp_timer, freertos, newlib, cxx, app_trace, asio, bt, cbor, unity, cmock, coap, nghttp, esp-tls, esp_adc_cal, esp_hid, tcp_transport, esp_http_client, esp_http_server, esp_https_ota, esp_https_server, esp_lcd, protobuf-c, protocomm, mdns, esp_local_ctrl, sdmmc, esp_serial_slave_link, esp_websocket_client, expat, wear_levelling, fatfs, freemodbus, idf_test, jsmn, json, libsodium, mqtt, openssl, perfmon, spiffs, usb, tinyusb, ulp, wifi_provisioning
  error: unknown target triple 'xtensa', please use -triple or -arch
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.60.1/src/ir/context.rs:529:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```

hawkw, 2023-01-13:
Quick update: after switching everything to point at `tag:v4.4` rather than `release/v4.4` and starting with a totally clean repo, everything _does_ appear to work in the Docker devcontainer, for me. Thanks for all the help!

SergioGasquez, 2023-01-13:
> everything does appear to work in the Docker devcontainer

Are you using a Dockerhub image or building it from the `Dockerfile`?

hawkw, 2023-01-13:
> > everything does appear to work in the Docker devcontainer
> 
> Are you using a Dockerhub image or building it from the `Dockerfile`?

I switched to building the image from the dockerfile, as you suggested in https://github.com/esp-rs/esp-idf-sys/issues/163#issuecomment-1380242552.

SergioGasquez, 2023-01-23:
I'll close the issue, feel free to ask further questions on our matrix channel: https://matrix.to/#/#esp-rs:matrix.org

closed 2023-01-23 by SergioGasquez