esp-rs/esp-idf-sys #165: Cargo audit reports a security vulnerability in the `time` crate dependency

reinhardbluelab, 2023-01-13:
Running `cargo audit` on my project results in the following report:

    Fetching advisory database from `https://github.com/RustSec/advisory-db.git`
      Loaded 481 security advisories (from /Users/r/.cargo/advisory-db)
    Updating crates.io index
    Scanning Cargo.lock for vulnerabilities (196 crate dependencies)
Crate:     time
Version:   0.1.44
Title:     Potential segfault in the time crate
Date:      2020-11-18
ID:        RUSTSEC-2020-0071
URL:       https://rustsec.org/advisories/RUSTSEC-2020-0071
Solution:  Upgrade to >=0.2.23
Dependency tree:
time 0.1.44
└── chrono 0.4.22
    └── build-time 0.1.2
        └── esp-idf-sys 0.32.1
            ├── esp-idf-svc 0.45.0
...
error: 1 vulnerability found!

ivmarkov, 2023-02-01:
Is this still a problem now that `time 0.1.45` had been released?

reinhardbluelab, 2023-02-02:
Not sure - according to the output it's only resolved from >=0.2.23.

closed 2024-06-21 by Vollbrecht