esp-rs/esp-idf-sys #166: How do I set multiple output pins at once?

defunctzombie, 2023-01-29:
I am trying to find the correct way to set multiple output pins at once. I've found a `set_out_w1tc` function available on `pub struct gpio_dev_s__bindgen_ty_3__bindgen_ty_1` which seems like it would let me do that. However, the name of this function does not make me think I am on the right path.

There is also a struct `gpio_dev_s` with a field `out_w1ts`. Not sure how to correctly use that but it also looked promosing.

I'm looking for the equivalent of the following if I was trying to do the same thing in c++ with the esp32 sdk.

```
GPIO.out_w1ts = ...;
```

defunctzombie, 2023-01-29:
Ok - I seem to have been able to use `esp_idf_sys::GPIO.out_w1tc = 0;` in an unsafe block. Is that the correct approach for setting that field?

MabezDev, 2023-01-29:
You can use any functions in the crate to interact with esp-idf directly (read: unsafely). If you just want to get something working that's  already been tested and has a safe Rust wrapper, please use the [esp-idf-hal](https://github.com/esp-rs/esp-idf-hal) crate instead.

closed 2023-01-29 by MabezDev

defunctzombie, 2023-01-30:
I did not find a way to set multiple pins at once with `esp-idf-hal`. I suspect that is because embedded-hal has an open issue (https://github.com/rust-embedded/embedded-hal/issues/30) about it so the approach is not yet decided for the hal abstraction.

So I am left figuring out the appropriate fallback using a lower layer than `esp-idf-hal`. I am new to rust and the wrapper library and wanted to confirm that my reasoning on how to set the field in the GPIO struct was correct.